Sloeber/arduino-eclipse-plugin #1155: Platform change includes old platform

mikep01, 2020-03-10:
**Describe the bug**
Changing the platform of a project does not clean up all references to old project and the project will not build.
Creating a new project and choosing the new platform builds properly.

**To Reproduce**

Create project.
Select platform, in my case esp8266/2.4.2.
Build project.
Change platform to esp8266/2.6.3
Build project - fails with compile failure due to referencing old library:
"...
-I.../arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/include"
...
-I"...\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266"
...
 -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"

**Expected behavior**
All references to 2.4.2 should have been removed when updating the platform for an existing project.
Works when starting with a new project.

**Desktop (please complete the following information):**
 - Win10

jantje, 2020-03-10:
Was the release folder deleted?

mikep01, 2020-03-10:
Yes (just re-verified).

jantje, 2020-03-10:
As there are about 700 boards working now ... can you provide the json you use.
Can you provide the sloeber version number?

mikep01, 2020-03-11:
I'm using the "Generic ESP8266 Module".
I'm on 4.3.1.201810101819.  I tried upgrading to 4.3.3.202003100521, but I found that it did not solve the problem and I believe it broke the "upload sketch" function - the command was failing as it was improperly formed.  I've since reverted back to the market version.
Which json can I provide and where would I find it?

jantje, 2020-03-11:
As I'm not going to fix the issue in 4.3.1 It may be interesting to elaborate on the upload sketch issue.

mikep01, 2020-03-11:
Same error was happening on the 4.3.3 version.  It takes me a long time to back out an upgrade (I needed to uninstall, reinstall from the market, then open each of my projects to re-save the board preferences and then clean and rebuild all).  So I'd prefer to wait to upgrade to a version that solves the board version problem, then hopefully I'll only have the "upload sketch" issue if it exists - I was playing with the C++/Arduino package at the time to work around this problem.  If it does I'll open a new issue with details at that time.

Working from memory if it helps, the upload sketch issue looked to be badly formed commands.  There was an extra "/" in the path for the ESP 2.4.2 version esptool launch.  In the 2.6.3 it complained the command line was empty and so invalid.

jantje, 2020-03-11:
usng https://arduino.esp8266.com/stable/package_esp8266com_index.json

Tried the changing 2.6.3 -> 4.3.3 ->2.6.3 version and there was a hickup in the last build. Cleaned the release folder and all was ok again.
But the upload fails in both versions.
I'll look at that one first.

As to testing. Install the product from http://eclipse.baeyens.it/  or https://github.com/Sloeber/arduino-eclipse-plugin/releases and you can have multiple versions of Sloeber at the same time.

mikep01, 2020-03-11:
Yes, that's the json that I'm using.

I assume you mean 2.6.3 to 2.3.3, and that's good to know.  Unfortunately my projects are at 2.4.2 and I'm trying to move up to 2.6.3 which seems to be broken.  If I create a project that starts at 2.6.3 it builds fine - I haven't tried downgrading as it's not something I need.

mikep01, 2020-03-11:
Well, I've no choice but to move up to 4.3.3 now :).  Looks like something was upgraded (CDT and arduino?) so I can't make any new projects on 4.3.1 - it crashes during creation complaining of a missing method.
On 4.3.3 I am able to take a NEW project that's started at platform 2.6.3 back to 2.4.2 and then return to 2.6.3, it builds in all 3 cases, and from 2.4.2 to 2.63.
The problem seems to be projects that started as 2.4.2 in the older plugin version - seems to keep something referencing 2.4.2 in the metadata for the project  after changing to 2.6.3.
Upload doesn't work for any project, I'll open that as a new issue.

uzi18, 2020-03-13:
So need to test upload, never used that =)

uzi18, 2020-03-13:
ok it works for me on avr, so platform depenedent?

jantje, 2020-03-13:
I did already a fix for the upload issue

mikep01, 2020-03-14:
Got it, installed and it's working for projects created on 4.3.3.  Projects that were created before the last update (mine were on 4.3.1) still can't be uploaded and if the platform is changed they won't build:

java.io.IOException: Cannot run program "/" (in directory "...mydirectory..."): CreateProcess error=5, Access is denied
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
	at io.sloeber.core.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:218)
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:75)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:209)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.io.IOException: CreateProcess error=5, Access is denied
	at java.base/java.lang.ProcessImpl.create(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:420)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:151)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
	... 5 more

jantje, 2020-03-14:
@mikep01 I guess with 4.3.3 you mean the nightly.

mikep01, 2020-03-14:
Hmm, I'm not sure, I didn't think so - I updated when I received notice of an update available in Eclipse: Sloeber, the Eclipse Arduino IDE (plugin version) 4.3.3.202003130521 ...
which from the date (0313?) I guessed that it had this fix?

jantje, 2020-03-14:
2020 03 13 => That is yesterday so yes nightly :-)

mikep01, 2020-03-14:
Ok.  It's intended that nightly builds are be released through the eclipse update mechanism?  This is only the 2nd update I've been offered through eclipse, they haven';t bee arriving daily...  not complaining - just wanted confirm this is the intent.

jantje, 2020-03-14:
I guess you have something installed that reports updates. Once you update because of the report (not from Sloeber) Sloeber gets updated as well.
If you have the nightly repository available you will update to the nightly.
FI On my eclipse development installation I get update available reports. If I decide to update based on this report The update will update taking into account all available repositories.

mikep01, 2020-03-20:
I was wondering if this will be fixed or do I need to find a workaround for projects created before 4.3.3?

jantje, 2020-03-20:
I think the nightly should no longer have this.
To fix wrong project 
Project properties->arduino->aply and close
Delete the release folder

mikep01, 2020-03-20:
I may not know how to install the nightly build - is there a link somewhere with instructions?
As I mentioned, "Sloeber, the Eclipse Arduino IDE (plugin version) 4.3.3.202003130521" was automatically found for me and that's what I'm running now.  When I check the nightly build update site using "check for updates" it shows this as the current version.
With this version I still get the error - the project will build on 2.4.2, but if I update the properties to use 2.6.3 (and save), delete the release folder, clean, and build, I get a number of errors starting with (and probably caused by) a failure in this file (notice the version number 2.4.2 still in the path):
...\eclipse\java-latest-released\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:256:0
If I create a new project I can change the ESP8266 version and it will build correctly - this is only an issue with projects that created on the previous version of the IDE.

mikep01, 2020-04-03:
Just got an update notice and installed Sloeber 4.3.3.202004010421.  I tried updating one of my projects to esp8266/2.6.3 but unfortunately the problem is still there and it still wouldn't build.
I don't mind waiting, but if this isn't going to be fixed please let me know so I can find a way to update my projects manually.

jantje, 2020-04-04:
I just tried it here. with the version of januari
[image]

First I build with version 2.4.2
```
12:32:32 **** Incremental Build of configuration Release for project issue1155 ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Start de C++ bouw'
"C:\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:/workspaces/embedded/issue1155/Release/core" -c -Wall -Wextra  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_GENERIC\"" -DLED_BUILTIN=11  -DESP8266   -I"C:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" -I"C:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\variants\generic" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
```
Then I added version 2.6.3 and changed the properties to 2.6.3 and it builds just fine
```
12:38:23 **** Incremental Build of configuration Release for project issue1155 ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Start de C++ bouw'
"C:\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/include" "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/lwip2/include" "-IC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:/workspaces/embedded/issue1155/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fexceptions -DBEARSSL_SSL_BASIC -DNONOSDK22x_191105=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_GENERIC\"" -DLED_BUILTIN=11 -DFLASHMODE_DOUT  -DESP8266   -I"C:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266" -I"C:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.6.3\variants\generic" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
```

I think you have a specific problem so please check following things.
Is the core/core pointing to the right location?
[image]

Are any includes pointing to the old platform?
[image]

is the old platform pointed at in the include project setiings
[image]

mikep01, 2020-04-12:
Shoot, I thought I'd responded to this but it looks like it was lost.
Still broken, I checked the things above.  It looks like it's the includes that get broken - when I change platforms it picks up includes referencing "core" from another project on the old version.  Perhaps coincidence, but that other project happens to be the first one in my workspace (and is not in any way referenced by the one I'm changing).

I'm now completely stuck using the last nightly build (see https://github.com/Sloeber/arduino-eclipse-plugin/issues/1170).  I can't create any new projects as I can't set them up properly, which prevents me from working around this issue by creating a new project to hold the old version.

jantje, 2020-04-12:
Can you show me the wrong include stuff?
If you delete the references (in C C++ and assembly) do they come back? and if so when?

mikep01, 2020-04-12:
I'm not sure how I would delete the references?  There are none in that project source (the includes reference another project that isn't referenced from the failing one) and they can't be removed from the include list.
What I did was find that incorrectly referenced project - it was at 2.4.2 so I updated it to 2.6.3 as well.  After a save & close, the include references were then removed both projects and new ones added to each, now both were referencing ANOTHER project still at 2.4.2.  So I updated that one, and the includes jumped again.  I continued until all of my projects were at 2.6.3, at which point it seems to have solved the problem for now - they all build.
So I can't capture an image of the weirdness anymore.  Basically the "includes" in the project were referencing an unrelated at 2.4.2 including the core from that unrelated project - which caused the build failure.  Once I moved everything up, the bad includes disappeared.

jantje, 2020-04-12:
There should be no references to other projects and if there re you can delete them
[image]
Please provide a image of a the includes referencing another project

mikep01, 2020-04-12:
The references are in the left side under the folder "Includes".  There's no method I could find for deleting invalid references in that include folder, I think it's generated automatically.  As I mentioned, there are no references there to other projects, whatever was generating that folder is picking up an unreferenced project.
I changed one of my projects back to 2.4.2, and the invalid references come back - now pointing to the same project that was causing the trouble before.  Except that referenced project has been updated to 2.6.3.  It's still not referenced anywhere in the failing project:
[image]

jantje, 2020-04-12:
If they are in the include section of the includes you can use the include browser
https://www.youtube.com/watch?v=5Yi5xeeqlsk
If what you see doesn't make sense rebuild the index

mikep01, 2020-04-12:
I'm not sure, but it may be ok now, rebuilding the index may have done the trick.  
Now when I change the target arduino version back and forth the weird references no longer show in the "Includes" folder.

closed 2020-04-12 by mikep01