Sloeber/arduino-eclipse-plugin #1156: Can't upload to arduino

mikep01, 2020-03-11:
Error message reporting failure when uploading to arduino:
Failed to run the Upload recipe 
Cannot run program "" (in directory "C:\{my eclipse directory}"): CreateProcess error=87, The parameter is incorrect

**To Reproduce**
Create a project using EPS8266 2.6.3
Build
Press "upload"

**Expected behavior**
The upload should complete without error.

**Desktop (please complete the following information):**
Windows 10

Sloeber plugin 4.3.3.202003110521

mikep01, 2020-03-11:
In case it helps - when moving from 4.3.1 to 4.3.3, I must open each existing project to the arduino properties, then hit "apply and close" before the project will build.
If I don't do that, it won't build but it will upload.
If I do update the properties, it'll build but no longer upload. New projects build but do not upload either.

jantje, 2020-03-12:
> In case it helps - when moving from 4.3.1 to 4.3.3, I must open each existing project to the arduino properties, then hit "apply and close" before the project will build.
> If I don't do that, it [...]

That is the normal upgrade process. http://eclipse.baeyens.it/how_to.shtml#/n

Something basic is currently wrong when handling environment variables in CDT. I'm investigating but current thinking is that it is a racing condition which makes it very hard to debug 😕

mikep01, 2020-03-20:
Fixed for new projects, thanks!
Still a problem for projects created before this version.

jantje, 2020-10-23:
As this was a regression in between releases I'm simply closing this issue

closed 2020-10-23 by jantje