Sloeber/arduino-eclipse-plugin #1160: RF24 1.3.4 library cannot be comiled with sloeber but with arduino ide

butch01, 2020-03-20:
**Describe the bug**
I tried to compile getting started example of RF24 library. But same sketch can be compiled with arduino ide. Using DOIT esp32 devkit 1.

Tried to with esp arduino core 1.0.4 from arduino IDE and with 1.0.3 from sloeber. (1.0.4 cannot be installed within sloeber automatically)

**To Reproduce**
Steps to reproduce the behavior:
- create new sketch RF24 library getting started example.
- select DOIT DEVKIT esp32 V1
- add SPI lib if not linked yet
- compile

**Expected behavior**
compiled successfully

**Screenshots**
compile errors:
```

'Building file: D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp'
'Starting C++ compile'
"D:\Arduino\Sloeber_4.3.2\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face" "-ID:/Arduino/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="doitESP32devkitV1"  -DESP32 -DCORE_DEBUG_LEVEL=4    -I"D:\Arduino\packages\esp32\hardware\esp32\1.0.4\cores\esp32" -I"D:\Arduino\packages\esp32\hardware\esp32\1.0.4\variants\doitESP32devkitV1" -I"D:\Arduino\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src" -I"D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\utility" -I"D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4" -MMD -MP -MF"libraries\RF24\RF24.cpp.d" -MT"libraries\RF24\RF24.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp"  -o  "libraries\RF24\RF24.cpp.o"
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp: In member function 'void RF24::printDetails()':
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:587:64: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
     "\r\n"),pgm_read_ptr(&rf24_datarate_e_str_P[getDataRate()]));
                                                                ^
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:587:64: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:590:60: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
     "\r\n"),pgm_read_ptr(&rf24_model_e_str_P[isPVariant()]));
                                                            ^
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:590:60: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:593:66: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
     "\r\n"),pgm_read_ptr(&rf24_crclength_e_str_P[getCRCLength()]));
                                                                  ^
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:593:66: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
D:\Arduino\Sloeber_4.3.2\arduinoPlugin\libraries\RF24\1.3.4\RF24.cpp:596:63: error: format '%s' expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
     "\r\n"),  pgm_read_ptr(&rf24_pa_dbm_e_str_P[getPALevel()]));
                                                               ^
[... cut ...]  expects argument of type 'char*', but argument 2 has type 'void*' [-Werror=format=]
cc1plus.exe: some warnings being treated as errors
libraries\RF24\subdir.mk:18: recipe for target 'libraries\RF24\RF24.cpp.o' failed
make: *** [libraries\RF24\RF24.cpp.o] Error 1
"D:/Arduino/Sloeber_4.3.2/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
```

**Desktop (please complete the following information):**
 - OS: Windows 10
- Sloeber 4.3.2

**Additional context**

executer-uno, 2020-11-06:
Looks like it is more strict rules of compilation of not-clear-C code from the RF24 lib. I already met that behaviour that ArduinoIDE have much more wide Warning borders and less Errors.
On your example it is clear that code is not right is better to ask to fix it RF24 library maintainer.

You can try a trick with disabling errors
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1119#issuecomment-568253821
but better to keep code clear.

Brolensky, 2021-02-13:
Hello,
I am using RF24-master, and i think there is no problems... Maybe with your jobs thanks.

closed 2024-02-20 by jantje