Sloeber/arduino-eclipse-plugin #1161: Failuer to create project

asoretmadolell, 2020-03-22:
**Describe the bug**
Whenever I try to create a new Arduino project, it exits with an error. The project folder is created, though.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project.
2. At the last (Finish) step, the error is thrown.

**Expected behavior**
Create a new project successfully. I have always been able to do so. The only difference now is that I'm on a different computer. If I try the Sloeber bundle, it works. But the problem is that I need to do it on Eclipse.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 1909

**Additional context**
Add any other context about the problem here.
[asd.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/4364588/asd.log)

Thank you very much for your help in advance!

jantje, 2020-03-22:
Please try the product downloadable from https://baeyens.it

asoretmadolell, 2020-03-22:
Hello Jantje!
Like I said, yes, I have tried the full product and it has worked for me.
However, I have some specific requirements that force me to use the plugin only.
Is there anything else I can provide you with?

jantje, 2020-03-23:
It is very hard to get all the version to work together properly. That is why we build the product and do not support other setups.
I advice to try to build the plugin in your environment.

asoretmadolell, 2020-03-23:
I completely understand. And fully appreciate your effort. If there's anything I can help you with, I'd be glad to.
I did try the exact same version of JRE that comes bundled with the product, and the error persisted. So I guess it's not a Java issue.
Other than that, I was able to compile previous projects I had successfully! I wasn't able before because there were other errors derived from uninstalling the plugin and it not completely deleting the "arduinoPlugin" folder, which caused a re-installation to fail
Thank you again for everything!

jantje, 2020-04-21:
Thinking about this. The product comes with java 8. I guess you are not using java 8

jandrassy, 2020-04-28:
I had this problem too with 2019-12 and 4.3.1 after reverting from nightly (4.3.2?). Now I installed 2020-03 and 4.3.3 and it can create a new Arduino project.

davydnorris, 2020-05-04:
I've just moved over from the CDT Arduino to this plugin and am having the same problem - Eclipse 2020-03 and CDT 9.11

I use Eclipse and the CDT all the time to build for the ESP8266 chip directly - I have only rarely used Arduino. I have a test sketch for some new sensors and thought this would be a quick way to check out whether I've got them wired correctly. Two days later and I'm thinking I should have written my own code hahaha

Any advice much appreciated

davydnorris, 2020-05-04:
OK so turns out the update site is only serving up 4.3.1. 
@jantje - can we get latest version served up from "http://eclipse.baeyens.it/update/V4/stable" please? It appears to be pointing to 4.3.1

Or a URL to the 4.3.3 plugin?

jandrassy, 2020-05-04:
I installed 4.3.3 from http://eclipse.baeyens.it/update/V4/nightly

jantje, 2020-05-04:
I forgot to update the stable update site.
Should be fixed now though I can't really test this.

jantje, 2020-05-04:
I'm testing right now and things fall apart. Please hold the line.

jantje, 2020-05-04:
After I added http://download.eclipse.org/nebula/releases/latest/ tot he update sites the install via marketplace works fine now and installs 4.3.3.

uzi18, 2020-05-04:
BTW @davydnorris what was wrong with CDT Arduino ?

davydnorris, 2020-05-04:
After I changed to nightly instead of stable, I got 4.3.3 and all of a sudden things worked exactly as in the brochure. Got my test sketch compiled and sensors working. Many thanks

@uzi18 - Eclipse CDT team has officially deprecated the CDT Arduino plugins and they will be removed in CDT 10. They are pointing everyone to this plugin, so @jantje  you are it. 

Also the plug in never worked well, or in a stable fashion, on the ESP and broke from release to release. As I mentioned, I use the NonOS SDK directly with C and the native CDT - never used Arduino much except for the odd demo and test sketch, and it was always a battle. Last two weeks I had tried to get a simple test sketch working, and it looks like CDT Arduino doesn't support all the Arduino pre and post hooks and ESP uses them. I was fiddling with the platform build rules just to get the .ld file made before it was needed. So this morning I uninstalled that and installed this, and 12 hours later I have what I needed. :-)

jantje, 2020-05-04:
>Eclipse CDT team has officially deprecated the CDT Arduino plugins

IMHO Adding arduino stuff to the cdt core was a big mistake. The plan is to remove the arduino stuff from cdt core in CDT Version 11. That will take out some of the arduino duplicates causing confusion.

>They are pointing everyone to this plugin,

Yes since this month. This is why I'm so concerned with this market place install issue. But I feel honoured :-)

> it looks like CDT Arduino doesn't support all the Arduino pre and post hooks and ESP uses them.

Arduino introduced these after the last update of CDT Arduino. Sloeber does not support all hooks. 
IMHO this whole hook thing is bad. Some Board providers use hooks for stuff that should not need the hooks. Some hook implementations are completely adapted to the arduino IDE way (like moving the source code to the temp folder and compile there). Some hook implementations are for "advanced features" where arduino IDE is focussing on the beginner.

uzi18, 2020-05-04:
@jantje nice to here it
did you tried to implement these hooks in sloeber?

in my opinion, you should prepare a draft/doc how to work with sloeber source code, what is where etc.
(for me) almost always it is hard to find some piece of code to work with to fix/debug something and help you :)

jantje, 2020-05-04:
>did you tried to implement these hooks in sloeber?

I implemented some hooks. The problem is that CDT works with make and make does not support all hooks as foreseen by the arduino framework.
The https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.tests/src/io/sloeber/core/CreateAndCompileDefaultInoOnAllBoardsTest.java tests tests most of the boards with the default ino file so hook issues are caught there.
No hook issues remained at the last test.

>and help you :)

It is always nice to get help but some realism is needed here. There are many things to know before you can understand all Sloeber components.
Things that come to mind are
- eclipse framework
- internal CDT workings
- The arduino framework
- make
- environment variables
The first 3 change over time and I don't feel entitle to do any documentation on the first 2. 
It is very likely I will spend more time on documentation than time won back from pull requests.
For instance the eclipse gui is build with SWT. I simply can't get my head around SWT. So most gui stuff is hacked together or provided by contributors like @wimjongman and @rlogiacco (Thanks guys) Can you imagine how much time I would need to spend to document?

Any-ways. Being here and helping others is making me feel good and is helping. I've seen you do that so thank you :-)

If you'd like to get something fixed. Feel free to ask and I'll happily explain my view and options to fix/improve. We can still see whether you'd like to do it yourself or leave it to someone else.

uzi18, 2020-05-04:
as always when got some free time :-)

was thinking about something more generic like - preparing to build is here, fetching boards is here, board programming is there etc.
maybe you can describe some generic workflow in build process
maybe you can just check if sloeber plugin building process description is actual and add some info where to start digging in code

when trying to fix something were used system/console.log() functions and it is far from perfect 

and yes will ask you :)

jantje, 2020-05-04:
@uzi18 I fail to understand your comment.
As to system/console.log(): Sloeber is written so that when there is a problem a dialog box pops up. If there is a problem and no dialog box-> hard to find issue.

jantje, 2020-05-04:
Assuming that the original issue waqs related to a incomplete install via market place and assuming that my test shows all is fine now ....
I'm closing this issue.

closed 2020-05-04 by jantje

uzi18, 2020-05-04:
@jantje mean it was beginning of my Sloeber plugin debugging that's why used console to dump some data ;)