Sloeber/arduino-eclipse-plugin #1164: Nightly windows version

hreintke, 2020-03-31:
Hi,

I want to renew my sloeber environment with a fresh/clean install.
The download on https://eclipse.baeyens.it/ poins to a 4.3.2 version. But there is already a 4.3.3.

Looking for nightly builds I found https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.3.3-SNAPSHOT/ but that only contains linux build.

Is there a windows 64 bit nightly available ?

jantje, 2020-03-31:
4.3.2 is the latest released version. https://github.com/Sloeber/arduino-eclipse-plugin/releases
4.3.3 is in making. I'm currently doing linux tests.

hreintke, 2020-04-01:
When using "about sloeber" and installation details I have for my current install : 

[image]

And for the just downloaded 4.3.2 version : 

[image]

Are you sure that this has the correct sloeber 4.3.2 included ? 

BTW : If you include the windows build in the nightly, I can (partly) do the testing for you.

jantje, 2020-04-01:
The nightly is platform independent (windows/linux/mac) so you should be able to update to the latest and greatest.
Note however that I moved workarounds at the platform installation time. So if things fail you should deinstall and reinstall the failing platform.
 
I forgot to update the plugin version numbers when releasing 4.3.2 so version 4.3.2 it is tagged as 4.3.1
As I'm preparing the 4.3.3 the current nightly is tagged as version 4.3.3
When looking at the version number you can see the last part is a date YYYYMMDDHHMM. This is done automatically and as sutch not prone to user error.

hreintke, 2020-04-02:
Can't find how to use the nightly on windows so keep it at 4.3.2 at this moment.

Other : 
In the process of new install I wanted to keep my current environment as a backup.
I am running sloerber-ide.exe from the c:\Utilities\sloeber directory.
Renamed that to c:\Utilities\sloeber_431 and tried to use sloeber-ide from there.
That failed because there were still references to the (now absent) c:\Utilities\sloeber directory.
So now have renamed back to old name but when starting I get the message : 
"The sloeber-ide executable launcher was unable to locate its companion shared library" and cannot use it.
Do you know how to solve this ?
Remark : I don't know exactly if and what actions i did between the renaming back from new to old name

jantje, 2020-04-02:
>"The sloeber-ide executable launcher was unable to locate its companion shared library" 

I don't know this one but take a look at sloeber-ide.ini. 

When you move stuff around always do the "workspace update actions" as described here http://eclipse.baeyens.it/how_to.shtml#/n

hreintke, 2020-04-02:
Can't find it right now.
Leave it as is because my  4.3.2 environment is up and running

closed 2020-04-02 by hreintke