Sloeber/arduino-eclipse-plugin #1165: Using hyphens in private hardware path causes wrong compile command

hreintke, 2020-04-02:
When using a hyphen in private hardware path f.e.
"c:\userdata\esp32\arduino-esp32" 
the compile command contains
"-DARDUINO_ARCH_arduino-esp32"

This causes 
1/the warning "<command-line>:0:21: warning: ISO C++11 requires whitespace after the macro name"
2/wrong setting of a build time define

Environment : Windows 8.1, sloeber 4.3.2

jantje, 2020-04-02:
I think you need more than only a hyphen in the hardware path.
The hyphen has to be in the hardware folder
and the platform.txt needs to contain some reference to it. not using "a acceptable" quoting/escaping
My guess is that you would have the same problem when this was in a boardsmanager installed board.

hreintke, 2020-04-02:
You are right on hardware folder.
I was referring to hardware path as that is the terminology in the sloeber 
windows->preferences->arduino->private hardware path.

Using `c:\userdata\esp32\esp-test\arduino_esp32` works ok.

It is used as {build_arch}  in the -DARDUINO_ARCH_{build.arch} (platform.txt).
As fas as I can see, there is no reference to build.arch  boards.txt.
And looks like it takes the folder name as the default value.

jantje, 2020-04-02:
build.arch is the folder name of the boards.txt containing folder unless that folder represent a version in which case it is the parent folder name.

IMHO This is in compliances with the Arduino platform specifications

I'm not sure why you think there is a Sloeber bug here.

hreintke, 2020-04-02:
I did not know that specification. 
Noticed it here and reported it as I expected that there was a safeguard to prevent this behavior.

In compliance -> no bug -> closing

closed 2020-04-02 by hreintke