Sloeber/arduino-eclipse-plugin #1166: Prebuild command doesn't expand correct

hreintke, 2020-04-02:
In ESP32 platform.txt there is this line  : 
`recipe.hooks.prebuild.2.pattern.windows=cmd /c if not exist "{build.path}\partitions.csv" copy "{runtime.platform.path}\tools\partitions\{build.partitions}.csv" "{build.path}\partitions.csv"
`

Which expanded to : 

`cmd /c if not exist "C:\Userdata\ESP8266\Projects\Command_32/Release\partitions.csv" copy "C:/Userdata/ESP32/arduino_esp32\tools\partitionsdefault.csv" "C:\Userdata\ESP8266\Projects\Command_32/Release\partitions.csv"`

There is a missing backslash from : 
`{runtime.platform.path}\tools\partitions\{build.partitions}.csv`
to 
`C:/Userdata/ESP32/arduino_esp32\tools\partitionsdefault.csv`

Environment : Windows 8.1, Sloeber 4.3.2. (also seen in 4.3.1)

jantje, 2020-04-02:
This is a cdt issue that won't be fixed quickly
https://bugs.eclipse.org/bugs/show_bug.cgi?id=560330
maybe you can use / to workaround

hreintke, 2020-04-02:
Using `copy "{runtime.platform.path}\tools\partitions\\{build.partitions}.csv"` in platform.txt makes it work.
Pity that we need a local patch to make it work (=remark to eclipse, not to you).

jantje, 2020-10-07:
This is now fixed in CDT10 !!!
Not yet released but removing this from the known limitations

jantje, 2020-10-12:
This is now fixed in the latest nightly

closed 2020-10-12 by jantje

AungKL, 2021-05-16:
> This is now fixed in the latest nightly

I have the same "partitions.csv" target 'pre-build' failed issue? It's work on 1.0.4, but not on 1.0.5, 1.0.6 and 2.0.0.
I want to update the latest nightly, my current version is   

> Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.3.202004230915
and running on Windows 10 64-bit. 

I'm not a developer, could you please show me the way to update?

Thanks

jantje, 2021-05-16:
All the info is available on the website http://eclipse.baeyens.it
The nightly link http://eclipse.baeyens.it/nightly.php
The upgrade procedure http://eclipse.baeyens.it/how_to.shtml#/n

AungKL, 2021-05-23:
Thank for reply,
I chose "method 2. upgrade current product" from [http://eclipse.baeyens.it/how_to.shtml#/n](url)
and able to update 4.3.3.202005042010
but the problem of "partitions.csv" target 'pre-build' failed issue with ESP32 v2.0.0 still not disappear.

**Current Version**

[image]

**Available Version**

[image]

**Updated Version**

[image]

**ESP32 v2.0.0 Setup**

[image]

**ESP32 - "partitions.csv" target 'pre-build' failed issue**

[image]

Is it correct way to update nightly build?

jantje, 2021-05-23:
>but the problem of "partitions.csv" target 'pre-build' failed issue with ESP32 v2.0.0 still not disappear.

That is because .3.3.202005042010 is not the latest nightly but the latest stable
Did you activate the nightly update site
Note this is a difficult upgrade as you will also have to upgrade java and cdt

AungKL, 2021-05-23:
After activating the nightly update site, and jre to v11, 

[image]

then I check for the update, there is an error "No remedy found"

[image]

Could you please help me to show the direction to nightly update?

jantje, 2021-05-23:
From the link I have send before
[image]

You are hitting the disadvantages