Sloeber/arduino-eclipse-plugin #1168: Project does not work after import from Git

uwekoenig, 2020-04-06:
**Describe the bug**
After importing a project from Git it does not work correctly. E.g. it is not possible to compile the project because it happens nothing when pressing the buttons. In the Project Explorer many entries except the source code are missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project "Arduino Sketch"
2. Right click on the project in Project Explorer > Team > Share Project ...
3. Select the repository to share with
4. File > Switch Workspace
5. Create a new workspace
6. Window > Perspective > Open Perspective
7. Select Git
8. Clone the repository
9. Right click on the repository > Import Project ...
10. Import both folders and check if the project is now open in the Project Explorer
11. The problem is now, that many items in the Project Explorer are missing (see the screenshots)

**Expected behavior**
The project opens like in the other workspace or machine.

**Screenshots**
View in Project Explorer in workspace where the project was created
[image]

View in Project Explorer in another workspace where the project was imported from Git
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 4.3.2

uzi18, 2020-04-06:
please leave also git repo link to test

jantje, 2020-04-06:
you need to open the project properties->arduino->apply and close
Note that it is a good practise to mark .project and .cproject as unchanged

uwekoenig, 2020-04-06:
@uzi18: Here is a repository for testing:
https://github.com/uwekoenig/SloeberGit.git

@jantje: "Aduino" does not exist in Project Properties, after importing both projects from the Git repository.
[image]

With your comments I think I found the problem now. It is necessary to import only one of the shown projects (the one with the right path).
[image]
When doing it in that way everything works like expected. Thank you for your quick support!

jantje, 2020-04-06:
Nice to hear it works

closed 2020-04-06 by jantje