Sloeber/arduino-eclipse-plugin #1170: Arduino actions broken

mikep01, 2020-04-12:
**Describe the bug**
Any action I select from the "Arduino" menu results in "Problem Occurred" with a few different descriptions, the consensus of which appears to be:
"There are no projects selected in the project explorer or all selected projects are not arduino projects."

**To Reproduce**
Steps to reproduce the behavior:
1. Select an arduino project
2. Click on "Arduino" and choose any action
3.  See error:  There are no projects selected in the project explorer or all selected projects are not arduino projects.

**Expected behavior**
execution of the selected function against the project.

**Desktop (please complete the following information):**
Windows 10

**Additional context**
nightly build     Sloeber, the Eclipse Arduino IDE (plugin version)	4.3.3.202004010421

jantje, 2020-04-12:
I just tried it here in Windows 10 latest code. 
Works fine here.
Nothing has changed in this area so I don't see why it wouldn't work anymore.
Are you sure you are in the project explorer?
Are you sure you are in arduino perspective?

mikep01, 2020-04-12:
Ok - it works from the project explorer.  The previous update wiped my explorer view and it no longer showed any projects so I switched to the C/C++ Projects view (I thought I'd make a mistake).  Now suddenly the project explorer view is back working (though oddly everything is listed in "Other Projects"?), I'm not sure it was related to me updating all of my original projects to arduino 2.6.3?

jantje, 2020-04-12:
I have no clue what you are referring to with "other project" can you provide a image?

mikep01, 2020-04-12:
Sure, this is what I see in the "Project Explorer":
[image]

I don't know where these views came from and I'm not sure what I was using before, but this is what I see there now.

jantje, 2020-04-12:
I have never seen this but it looks to me you have workingsets and resources and have these visible in the project explorer

mikep01, 2020-04-12:
Weird, yeah, somehow some filters were enabled in that view and started showing things as nested.  I disabled the weird ones and  kept the C/C++ and it's back to normal.  Thanks!

closed 2020-04-12 by mikep01