Sloeber/arduino-eclipse-plugin #1171: Installation: Failed to parse ... library_index.json

climategadgets, 2020-04-13:
**Describe the bug**
"Problem occured" dialog box pops up, with the following message:
```
Failed to parse json file "${eclipse-installation-location}/arduinoPlugin/library_index.json"
java.io.EOFException: End of input at line 149641 column 19 path $.libraries[6779].checksum
```
Note: the file at the specified path doesn't exist.

**To Reproduce**
Steps to reproduce the behavior:
1. Initiate "Install Sloeber Product" from Ecipse Marketplace
1. See error.

**Expected behavior**
Product installs.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux

jantje, 2020-04-13:
I guess there was a glitch during install that made the download fail.
Sloeber deletes the file and the reference to it in case files fail to download

climategadgets, 2020-04-13:
Are there any actions that can be taken so that this error doesn't happen? It is pretty reproducible, up to wiping out the whole Eclipse installation and starting afresh.

jantje, 2020-04-13:
It doesn't happen here.
The file that Sloeber tries to download is http://downloads.arduino.cc/libraries/library_index.json
Maybe you can try https://downloads.arduino.cc/libraries/library_index.json
If you do not have the file you don't have the Arduino provided libraries.
As far as I understand starting a second time the error should no longer be.

jantje, 2020-04-13:
try to install the product from https://eclipse.baeyens.it/ I'm not sure how the market place handles product  installs

climategadgets, 2020-04-13:
Indeed, manually downloading the file into that location when Eclipse is down and then starting it up makes the "Finishing the installation" get to the next step. Wonder why can't the installer do it on its own, doing it with `wget` takes a split second. Is there a log that installer keeps somewhere so it can be diagnosed what it's doing and where it's failing?

jantje, 2020-04-13:
From your error log you can see the file is truncated
It could be your virus checker

climategadgets, 2020-04-13:
The file doesn't exist (not even the zero length). The platform is Linux, there are no virus checkers.

I've got the issue resolved (manual download), so I'll close it now - but if you feel like getting to the root cause and making your product better - I'm willing to help.

closed 2020-04-13 by climategadgets

jantje, 2020-04-13:
As I said. The file is deleted when it fails to parse. Sloeber does so to avoid the same problem to happen again. It also removes the file from the index files to .. avoid the same thing to happen again.
I do realize this does not help "problem solving"

The parsing is done by a json parser that -in your case- says there is an unexpected end of file. Given the other info: the file is not empty. So it must be a download issue, a local issue, or the file changed recently and it manifests on linux (Last tests on linux are from some days ago) and not on windows (just tested it)

If it is consistent on your system it would be great if you could debug. The easiest is to compile the SDK and start a debugging session.
https://github.com/Sloeber/arduino-eclipse-plugin#developing-improving-the-plugin
The problem is probably here https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/api/LibraryManager.java#L309