Sloeber/arduino-eclipse-plugin #1172: Enable generation of .lss listing file

jwd996, 2020-04-13:
This is more a feature request than a bug.

Could you please add an optional run of 
<platform>-object-dump $LISTOPTIONS $PROJECTNAME.elf >$PROJECTNAME.lss
like it exists in standard CDT.

OK, I'm not a typical Arduino user, but I sometimes like to see, whether I am understood by the Compiler by checking the resulting machine code. Especially for the very restricted platforms like ATtiny, this would be very handy.

I'm using the latest version	4.3.1.201910070607

jantje, 2020-04-13:
Sloeber is not meant for the typical Arduino user. ;-)
I'm not sure what you are asking for. Sloeber is build on top of CDT and simply tells CDT how Arduino wants it stuff compiled.
I don't know how to run your command in CDT. But the run external tools should take you a long way.

jwd996, 2020-04-14:
It's not my command, it's a tool from binutils, like avr-as, avr-ld, avr-size, avr-objcopy.
It's just running avr-objdump, like avr-size, which is (configurable) run as part of the build step.
The projects properties also have a page "C/C++ Build->Settings", where I can find a tab Tools Settings, that is modified from the original CDT tab, where e.g. there is no "size" command to configure. No idea whether this is confiigurable or needs some code. If I would know that, I might have come with a patch file :-)

jantje, 2020-04-15:
I added the following to my Marvin project.
[image]
Which results in this at the end  of the build.
```
' '
C:/sloeber/arduinoPlugin/tools/make/make --no-print-directory post-build
"C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-objdump.exe"  mowrobot.elf >mowrobot.lss
Usage: C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-objdump.exe <option(s)> <file(s)>
 Display information from object <file(s)>.
 At least one of the following switches must be given:
  -a, --archive-headers    Display archive header information
  -f, --file-headers       Display the contents of the overall file header
  -p, --private-headers    Display object format specific file header contents
  -P, --private=OPT,OPT... Display object format specific contents
  -h, --[section-]headers  Display the contents of the section headers
  -x, --all-headers        Display the contents of all headers
  -d, --disassemble        Display assembler contents of executable sections
  -D, --disassemble-all    Display assembler contents of all sections
  -S, --source             Intermix source code with disassembly
  -s, --full-contents      Display the full contents of all sections requested
  -g, --debugging          Display debug information in object file
```

jantje, 2020-04-21:
closing due to no reaction

closed 2020-04-21 by jantje

uzi18, 2020-04-22:
@jantje will check what switches have used usually, some of them don't work with others!