Sloeber/arduino-eclipse-plugin #1175: Test part of the build fails on travis and maven build but not in eclipse

jantje, 2020-04-17:
Looking at the travis build you can see the test part of the build fails.
https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin

The jenkins build on my build server using 
`clean verify -Pall,sloeber_release,NOSDK -Dtest=NightlyJenkins`
works fine but the same command on my windows machine fails

but in eclipse on my windows machine the test runs fine .
I'm confused.

jantje, 2020-04-17:
@wimjongman Can you have a look? I'm out of my depth.
The only thing I can thinkof is that the travis states it is a AMD64  machine.
attached is the log file of my local windows build complaining it cant find io.sloeber.core which it just build.
[1587163122752.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/4495564/1587163122752.log)

jantje, 2020-04-18:
The amd64 is not the issue as other builds running on AMD64 worked all fine

jantje, 2020-04-18:
Going back into history on travis I find this build to be the first failing
https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin/builds/662807609
And these are the changes between the 2
https://github.com/Sloeber/arduino-eclipse-plugin/compare/59cca4f81d44...4e27d3a1e3b1
But the fail is an actual fail (I mean a test is failing showing regression) that is fixed later. 
The next publish fails with an error ( due to a coding error but indicating a problem with the boards.txt file)
https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin/builds/662807609
Here the change is introducing the additional boards.txt and platform.txt files to workaround at installation time.
It looks as if the files are not properly created causing an error in logging that wasn't properly tested resulting in a regex problem.
Fixing the regex problem now hides the problem as the problem is logged as info. But the test still fails.

jantje, 2020-04-18:
The local maven build used java 9 which caused problems.
Now the test still fails but it looks more manageable to me.
It may be related to to long command lines on windows though that would be strange to hit this at this time.

jantje, 2020-04-18:
A local maven build update: 
Running the generated makefile gives a build error and the command is (except for the location of files) exactly the same as when running the test from eclipse.
The error indicates a include location to be missing or faulty. 
I no longer think it is related to the travis failure.

jantje, 2020-04-18:
local maven build succeeded  when making the path smaller.
Only Travis build is remaining a problem.

wimjongman, 2020-04-21:
Did you figure it out?

jantje, 2020-04-21:
@wimjongman Thanks for responding. I made lots of progress but unfortunately the problem is not yet fixed. The build and tests now works on my local systems but not on travis. On Travis the build is fine but all the tests failed. My current thinking is that on Travis the sloeber variants of boards.txt and platform.txt get "corrupted" in one way or another. 
As I don't see how to get to the workspace on travis and there are no errors it is hard to investigate.
I guess I have to talk to the people of travis but I have a more important problem at hand here at home (marvin broke down and grass is growing like crazy)

wimjongman, 2020-04-22:
It looks like this command is too quiet:

    sh -c "if $TRAVIS_SECURE_ENV_VARS; then ./travis_exec_and_print_time.sh 'mvn deploy -Dtest=RegressionTest  --quiet --settings settings.xml'; else ./travis_exec_and_print_time.sh 'mvn verify -Dtest=RegressionTest --quiet --settings settings.xml'; fi"

Please look at 6.19 here, to add some more logging to the build.

Alternatively, this specific test could also put out some more information with sysout.println?

jantje, 2020-04-22:
After having done many tests and mods I'm starting to suspect that the tests are run in parallel on travis and that is causing a issue.
For instance this log line https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin/builds/678297082#L582
is part of test https://travis-ci.org/github/Sloeber/arduino-eclipse-plugin/builds/678297082#L630
And it is dumped long before the test ending line and the test only takes 0.001s.
If things ran one by one it should be just in front.
As to what I read the tests should be run one after the other by default.
I can use some help here.

jantje, 2020-04-22:
@wimjongman 
I found the issue but I don't understand what is going on.
Sloeber makes boards.sloeber.txt files based on boards.txt files
Internally it looks like  sloeber uses boards.txt files but when it tries to load the file the boards.sloeber.txt file is loaded.
I have tested this on linux and y build server is a linux machine.
Now it turns out that -due to a bug- the boards.sloeber.txt file is never made. So the board is not found, so the test fails.
But it worked many times....
https://github.com/Sloeber/arduino-eclipse-plugin/blob/f9fa99801ca1ba4b048be6af1b6fe6ffb97fdba6/io.sloeber.core/src/io/sloeber/core/managers/WorkAround.java#L109-L121

jantje, 2020-04-23:
I have updated the release and the download page so this one is closed.

closed 2020-04-23 by jantje

wimjongman, 2020-04-23:
👍

jantje, 2020-04-28:
Maybe strange that this is fixed in 4.3.3 but I I had originally renamed 4.3.3beta to 4.3.3 and then I made 4.3.3.
So when using 4.3.3 make sure you have 4.3.3 and not 4.3.3 beta