Sloeber/arduino-eclipse-plugin #1177: How do I set default values for Private Hardware and Private Libraries with Linux

Vincent3004, 2020-04-23:
**Describe the bug**
Hello,
I would like to start an Arduino project with Sloeber, but I can not go forward in creation wizzard.
I have set following paths for Private Library path and Private Hardware path : 

[image]

Then into the Sloeber IDE, I do File/New/Arduino Sketch, and after giving a name to the project when I reach page «Provide the Arduino Information» there is no way to select a board ?

[image]

Then I try to open the serial monitor view, and the data serial monitoring through port /dev/ttyACM0 is working fine with an arduino Uno R3 plugged…

How is it possible the serial monitor works, so the arduino board is detected but I am not able to select a board in arduino project creation wizzard ?
Should I try to uninstall Arduino IDE and reinstall it ?

Thank you for your help,

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File/New/Arduino Sketch'
2. Give a name to the new project and on next page Click on 'board'
3. no Scroll down menu
4. See error «the folder «/usr/share/arduino/hardware/tools/avr/lib/ssl/private» is empty in Error log tab.

**Expected behavior**
I would like to get a scroll down menu where I can select ATmega 328P board in the arduino sketch wizzard.

**Desktop (please complete the following information):**
 - OS: Linux lubuntu 18.04 64 bit
 - Browser : Mozilla Firefox

jantje, 2020-04-23:
I guess you are on the latest release.
I F*up this release see #1175 
If you update to the nightly it should work.

jantje, 2020-04-23:
Please download the new V4.3.3

Vincent3004, 2020-04-23:
> Please download the new V4.3.3

Hello Jantje, your new version V4.3.3 works great! I still need to to discoter it but it looks really better than Arduino IDE. Have a nice week-end,

jantje, 2020-04-23:
Great. Thanks for the feedback
You can find some video's and other helpfull stuff in the arduino  menu in sloeber and on baeyens.it.
Most of video's are a bit outdated but still great to get started.

closed 2020-04-23 by jantje