Sloeber/arduino-eclipse-plugin #1179: Issues with IDE

oflanagn, 2020-04-27:
Hello All,

I just started using the IDE and can not get anything to compile. Even basic examples. eg:

void setup() {
  pinMode(13, OUTPUT);    // sets the digital pin 13 as output
}

void loop() {
  digitalWrite(13, HIGH); // sets the digital pin 13 on
  delay(1000);            // waits for a second
  digitalWrite(13, LOW);  // sets the digital pin 13 off
  delay(1000);            // waits for a second
}

It will not recognize any of the Arduino functions even though i can see the library's are visible in the project explorer.

uzi18, 2020-04-28:
we need more informations, like core used, sloeber version, error shown in console etc.

oflanagn, 2020-04-28:
Sorry, I am using the ESP32 Feather, The compiler reports the below during a compile:
06:06:34 **** Incremental Build of configuration Release for project Pin_IO ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\Sloeber\Projects\Pin_IO\partitions.csv" copy /y "C:\Sloeber\Projects\Pin_IO\partitions.csv" "C:\Sloeber\Projects\Pin_IO\Release\partitions.csv"
cmd /c if not exist "C:\Sloeber\Projects\Pin_IO\Release\partitions.csv" copy "C:/Users/Jason O'Flanagan/Documents/Arduino/hardware/espressif/esp32\tools\partitionsno_ota.csv" "C:\Sloeber\Projects\Pin_IO\Release\partitions.csv"
The system cannot find the file specified.
makefile:104: recipe for target 'pre-build' failed
make[1]: *** [pre-build] Error 1
makefile:65: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

06:06:35 Build Failed. 4 errors, 0 warnings. (took 1s.362ms)

jantje, 2020-04-28:
please specify the json file url and the platform version you are using
There are over 700 boards that work with sloeber on windows

oflanagn, 2020-04-28:
Sorry Jantje,
Here is the board: 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
The individual board is "Adafruit ESP32 Feather"

jantje, 2020-04-28:
I tried version 1.0.4-rc1 here without problems
```
18:07:21 **** Build of configuration Release for project issue_1179 ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Start de C++ bouw'
"C:\sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/config" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/app_trace" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/app_update" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/asio" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/bootloader_support" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/bt" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/coap" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/console" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/driver" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp-tls" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp32" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_adc_cal" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_event" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_http_client" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_http_server" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_https_ota" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/esp_ringbuf" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/ethernet" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/expat" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/fatfs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/freemodbus" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/freertos" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/heap" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/idf_test" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/jsmn" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/json" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/libsodium" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/log" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/lwip" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/mbedtls" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/mdns" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/micro-ecc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/mqtt" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/newlib" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/nghttp" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/nvs_flash" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/openssl" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/protobuf-c" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/protocomm" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/pthread" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/sdmmc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/smartconfig_ack" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/soc" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/spi_flash" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/spiffs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/tcp_transport" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/tcpip_adapter" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/ulp" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/vfs" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/wear_levelling" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/wifi_provisioning" "-IC:\sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4-rc1/tools/sdk/include/wpa_sup [... cut ...]
.xt.prop._ZTVN9__gnu_cxx26__concurrence_unlock_errorE             12            0
.xt.prop._ZTVSt9bad_alloc                                         12            0
.xt.prop._ZTVN10__cxxabiv120__si_class_type_infoE                 12            0
.debug_loc                                                    513154            0
Total                                                        4185570

'Finished building target: issue_1179'
' '

18:08:02 Build Finished. 0 errors, 0 warnings. (took 40s.507ms)

```

oflanagn, 2020-04-28:
Thank you all, I rolled back to 1.0.4 and the issue was fixed. 

Great support guys!!

closed 2020-04-28 by jantje