Sloeber/arduino-eclipse-plugin #1180: Settings like "Automatically import libraries" revert to installation defaults

jandrassy, 2020-04-28:
Always after starting Eclipse, the plugin "finishes installation". (Is it normal?) After that the settings from the Arduino->Preferences (stored in io.sloeber.arduino.prefs) revert to defaults. (It doesn't help to make the file read-only.)

On Linux. Eclipse 2020-03 or 2019-12. Plugin 4.3.3 from /nightly.

jantje, 2020-04-28:
>Always after starting Eclipse, the plugin "finishes installation". (Is it normal?) 

It wasn't like this but due to a fix Sloeber started doing so.

> After that the settings from the Arduino->Preferences (stored in io.sloeber.arduino.prefs) revert to defaults.

Strange I'll check that. Shouldn't be like that.

>(It doesn't help to make the file read-only.)

Nice way of trouble shooting/investigation/workaround but apart from that it is not the way to go.

jantje, 2020-04-28:
> After that the settings from the Arduino->Preferences (stored in io.sloeber.arduino.prefs) revert to defaults. 

I upgraded my "production environment" to the latest nightly and I could not reproduce this.

jandrassy, 2020-04-28:
So I updated my Sloeber 'development' environment to test it. As I confirmed the Preferences in debuged Eclipse instance, the Eclipse Preferences Recorder poped up. I confirmed it. And the problem is gone even in 'production' environment. There I had the Preference Recorder disabled. The Eclipse Preference Recorder synchronizes the preferences between workspaces a I think it was the reason. It is possible that the problem started when I created a new Eclipse workspace with 2019-12. 
Sorry to bother you with this and thank you for Sloeber once again.

closed 2020-04-28 by jandrassy

jantje, 2020-04-28:
> Preference Recorder 

First time I heard of it I though "Another stupid idea of some stupid moron that will hurt eclipse seriously"
Then I heard eclipse made it mandatory. It didn't feel good. I still havn't found a way to disable this great feature breaking object oriented encapsulation principle.