Sloeber/arduino-eclipse-plugin #1181: Archiver failed during build

JochenKoehler69, 2020-05-01:
When trying to build a project in Sloeber 4.3 I get this error message when archiver is started:

> 'Starting archiver'
"D:\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar" cru  "D:/Arduino/WifiConfiguratorTemplate/NodeMCU_V10_Debug/arduino.ar"    D:/Arduino/WifiConfiguratorTemplate/NodeMCU_V10_Debug/arduino.ar 
D:\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-ar: unable to rename 'D:/Arduino/WifiConfiguratorTemplate/NodeMCU_V10_Debug/arduino.ar'; reason: File exists
makefile:76: recipe for target 'D:\Arduino\WifiConfiguratorTemplate\NodeMCU_V10_Debug\arduino.ar' failed
make: *** [D:\Arduino\WifiConfiguratorTemplate\NodeMCU_V10_Debug\arduino.ar] Error 1

I found the issue [https://github.com/esp8266/Arduino/issues/4919](https://github.com/esp8266/Arduino/issues/4919) and tried to change the "cru" by "cr" in platform.txt but obviously I'm not able to find the right file because in the error it's still "cru".

Any advice?

When building a non-debug release it works and also on other projects it works ...

jantje, 2020-05-01:
I guess the problem is that the library is there 2 times
`cru "D:/Arduino/WifiConfiguratorTemplate/NodeMCU_V10_Debug/arduino.ar" D:/Arduino/WifiConfiguratorTemplate/NodeMCU_V10_Debug/arduino.ar`

jantje, 2020-05-01:
Can you provide a image  of the project properties->arduino of the failing configuration?
There are to many options to try and test.

jantje, 2020-05-27:
closing due  to no response

closed 2020-05-27 by jantje