Sloeber/arduino-eclipse-plugin #1182: Compiler path when using private hardware path

hreintke, 2020-05-09:
Running on Windows 8, Sloeber 4.3.2

I have a configuration with two esp8266 configurations.

1/ Boardmanager defined (C:\Utilities\Sloeber_432\arduinoPlugin\packages\esp8266\hardware\esp8266\2.6.3)
2/ Defined in private hardware path. (c:\userdata\ArduinoGIT_HR_G7)

What I see is that a project wich has c:\userdata\ArduinoGIT_HR as platform folder is using the compiler from the boardmanager platform.

```
'Starting C++ compile'
"C:\Utilities\Sloeber_432\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Userdata/ArduinoGit_HR_G7/tools/sdk/include" "-IC:/Userdata/ArduinoGit_HR_G7/tools/sdk/lwip2/include" "-IC:/Userdata/ArduinoGit_HR_G7/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:/Userdata/ESP8266/Projects/g7test/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -MMD -ffunction-sections -fdata-sections -fexceptions -DBEARSSL_SSL_BASIC -DNONOSDK22x_191024=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=1   -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1MINILITE -DARDUINO_ARCH_ARDUINOGIT_HR_G7 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINILITE"  -DFLASHMODE_DOUT  -DESP8266   -I"C:\Userdata\ArduinoGit_HR_G7\cores\esp8266" -I"C:\Userdata\ArduinoGit_HR_G7\variants\d1_mini" -MMD -MP -MF"g7test.cpp.d" -MT"g7test.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\g7test.cpp"  -o  "g7test.cpp.o"
'Finished building: ..\g7test.cpp'
```

The environment variables which (I think) make this happen are

A.COMPILER.PATH : ${A.RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH}/bin/
A.RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH C:\Utilities\Sloeber_432\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506

I would expect each platform uses his own toolchain to compile.
Otherwise there is no possibility to use different toolchains for different platforms.

I did see the option "Select toolchain like Arduino IDE" but selecting/unselecting made no difference.

How/where is decided to use which toolchain ?

jantje, 2020-05-09:
First note: lots of the toolchain selection  has changed in 4.3.3. Version 4.3.3 is way more compliant with the arduino ide way than V4.3.2.  Though I don't think this will fix your problem.

Basically what you are trying to do (have high levels of control on the toolchain in private hardware without modifying platform.txt) is not supported by the arduino framework.

>How/where is decided to use which toolchain ?

In the Arduino framework the toolchain is installed and selected based upon the json file used to install the platform. 
As the private hardware is not installed by a json file the toolchain ends up  being some kind of "default/undefined" (read whatever the code does) (and seriously changed between Sloeber V 4.3.2 and 4.3.3)
In V4.3.2 it is the latest version of this tool installed by the boards-manager.
in V4.3.3 I have no clue what will happen.

To work around it one can modify the platform.txt and add the wanted redirections. In your example that would be something like
RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH=[somewhere on your system]

To resolve this problem the private hardware platforms should contain a json file and Sloeber should use that json file instead of the system wide json files. However when there is a new toolchain you would still have to "install that toolchain" or Sloeber won't find it.
4.3.3 took a start on this by copying the json files to the platform installs but Sloeber does not yet use those.

>I would expect each platform uses his own toolchain to compile.
Otherwise there is no possibility to use different toolchains for different platforms.

If both platforms are installed by the boards manager this works fine. The problem is that the arduino framework uses the json files to install and locate the toolchains. When using private hardware Sloeber does not have a json file so Sloeber did not install the toolchain and has to make guesses which one you want.

>I did see the option "Select toolchain like Arduino IDE" but selecting/unselecting made no difference.

This option is to stick stronger to what the json files states. As you do not have a json file for your private hardware this will not influence the toolchain selection in this case.

hreintke, 2020-05-10:
@jantje Thanks for the extensive reply. 

> If both platforms are installed by the boards manager this works fine. The problem is that the arduino framework uses the json files to install and locate the toolchains. When using private hardware [...]

In ESP8266 (&ESP32) the download of tools can also be done with the get.py, provided in git.
No need for specific/additional parsing of json for that.

> To work around it one can modify the platform.txt and add the wanted redirections. In your example that would be something like
RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH=[somewhere on your system]

I had/have been looking at the platform.txt. 
The one from the private hardware path contains :  
`runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf`
and
`compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/`

The one from boardmanager only 
`compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/`
That is conform the documentation at which states : 
`# These will be removed by the packager script when doing a JSON release`

But projects on both platforms get the same `C:\Utilities\Sloeber_432\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506`  for `runtime.tools.xtensa-lx106-elf-gcc.path`
In the boardmanager version I cannot find any definition of `runtime.tools.xtensa-lx106-elf-gcc.path` so it must be set (also ?) somewhere else.

--Do you know where ? 

When I update the `compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/`
to `compiler.path={runtime.platform.path}/tools/xtensa-lx106-elf/bin/` it is indeed using the correct toolchain.

> First note: lots of the toolchain selection has changed in 4.3.3. Version 4.3.3 is way more compliant with the arduino ide way than V4.3.2. Though I don't think this will fix your problem.

Did some first tests with 4.3.3 but could not yet get that working with private hardware path platforms. I can install but get errors creating projects.  
"Failed to create project" errorcode -2

hreintke, 2020-05-13:
@jantje
As far as I know now the `runtime.tools.xtensa-lx106-elf-gcc.path` is set by the Arduino framework, based on information in the package json file.
```
[...]
               "toolsDependencies": [
                  {
                     "packager": "esp8266",
                     "version": "2.5.0-4-b40a506",
                     "name": "xtensa-lx106-elf-gcc"
                  },
[...]
```

In the platform.text I use, this value is overridden by the line : 
`runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf`

But that is not used to substitute in `compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/` later in the platform.txt which still uses the framework provided one.

Questions 
- Is overriding framework variables not implemented in sloeber ? 
- At which moment(s) is the platform.txt parsed ?
   - At adding the private hardware path ?
   - At creating a project ?
   - At each compile ?
   - Other ?
- Can you point me at the sloeber code where parsing takes place ?

jantje, 2020-05-13:
>Is overriding framework variables not implemented in sloeber ?

yes

>At which moment(s) is the platform.txt parsed ? 

when you go to the project properties and select "apply" or "apply and close"

>Can you point me at the sloeber code where parsing takes place ?

yes. Here the entry point. https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L725
Note however 4.3.3 is seriously different from previous versions in regards to processing boards.txt and platform.txt.
**If you want to look at the code please move to 4.3.3.**
Differences that string to mind are:
1) In 4.3.3 the json parsing is in the "system environment variables". preferences->C/C++->build->environment. This is only updated when you add/remove platforms. Previous versions did not use the "system environment variables"

2) In 4.3.3 the project level json, boards.txt and platform.txt parsing only parses the board platform and referenced platforms. Previous versions considered all installed tools which is not compliant with arduino ide behaviour.

3) in 4.3.3 workarounds are applied at install and .txt level. Now Sloeber creates boards.sloeber.txt and platform.sloeber.txt when it parses the boards.txt and platform.txt file. If you modify the boards.txt or platform.txt you need to delete the sloeber variant for your modifications to be taken into account. There is a key in the sloeber variants that is hardcoded to support changes in the workarounds. https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/managers/WorkAround.java#L97

I've written a doc on the order of environment variable creation here https://github.com/Sloeber/arduino-eclipse-plugin/wiki/How-environment-variables-are-set

hreintke, 2020-05-13:
Is not supporting framework variables a Eclipse/CDT limitation or a sloeber design decision ?
 
----------

Did some investigation :

Fresh Sloeber 4.3.3
Install arduino avr by default.

-> system environment variables empty.

Create arduino project

-> system environment variables exist

Add esp8266 platform

-> no update to system environment variables

Create esp project 

-> no update to system environment variables
-> runtime.tools for esp8266 exist in project.
-> no project build variables

Build esp project 

-> no change 

--------------------

I do not see any boards.sloeber.txt or platform.sloeber.txt.
There are pre- and post-processing board and platform processing files.

-----------------------

I added a private hardware path : 

-> Creating a project for this platform or changing an existing project to this platform always gives error -2.
-> No way to use that platform.

--------------------

It there are items/settings/options that you would like me to use & test, please let me know.

jantje, 2020-05-13:
> Did some investigation :

Good 👍 

As this result confused me 
>Create esp project
-> no update to system environment variables

I did the same test here and there is indeed a problem.
It looks like only arduino tools are added to the system environment variables and that will cause problems for private hardware.

jantje, 2020-05-13:
With the fix I  did the investigation should get further.
> Did some investigation :
>
> Fresh Sloeber 4.3.3
> Install arduino avr by default.
>
> -> system environment variables empty.
>
> Create arduino project
>
> -> system environment variables exist
>
> Add esp8266 [...]

The fix should now provide the esp8266 tools

> -> runtime.tools for esp8266 exist in project.

This is normal as it is in the referencing project json

> -> no project build variables

I do notunderstand what you try to say

> 
> Build esp project
> 
> -> no change

I do not understand what you try to say

> 
> I do not see any boards.sloeber.txt or platform.sloeber.txt.

These files should be at the same location as the boards.txt and platform.txt

> There are pre- and post-processing board and platform processing files.

Yes but do not modify them as they are replaced each time you start Sloeber

> 
> I added a private hardware path :
> 
> -> Creating a project for this platform or changing an existing project to this platform always gives error -2.
> -> No way to use that platform.

Lets hope the fix makes this a bit better.

hreintke, 2020-05-14:
Hi,
What I did today : 
- Download latest source from git (zipfile)
- run mvn clead verify -DskipTests=true
- copy sloeber to temp directory (running from build environment has "long path" issue
- start sloeber
Getting errors like 
```
Failed to configer Sloeber
Failed to install Arduino AVR Boards
Failed to download "https://eclipse.baeyens.it/download/make.zip".
sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
```
The PKIX building failed I also noticed on other downloads f.e the esp8266...json

Any hint how to solve this ?

On your comment : 

> -> no project build variables

>I do notunderstand what you try to say

After the step I looked in 
Project->properties->c/c++ build->Build variables
There were no build variables shown

>Build esp project

> -> no change

> I do not understand what you try to say

After this step I looked into environmnet/project/build variables and nothing had changed.

jantje, 2020-05-14:
>Any hint how to solve this ?

looks like your system didn't allow sloeber to access the internet

As explained here https://github.com/Sloeber/arduino-eclipse-plugin/wiki/How-environment-variables-are-set Sloeber does not write to Project->properties->c/c++ build->Build variables

If there are no boards.sloeber.txt or platform.sloeber.txt there is either some issue on you system or you are not on 4.3.3 or later

hreintke, 2020-05-15:
>looks like your system didn't allow sloeber to access the internet

Sloeber does have access to internet.
Is caused by certificate issues, not sure yet what it is exactly.

Checked for differences between current 432 and dev 433 environment.
In 432 there is a jre directory, that is not within 433.
Copying 432 jre into 433 directory solved the PKIX issue.

Situation now :

Latest git, jre from 432, new workspace
Installed Arduino by default
Added private hardware path
create hardware path project -> fails

Latest git, jre from 432, new workspace
Installed Arduino by default.
Created arduino project 
Added esp8266 platform from boardmanager, installs OK
Created esp project ->ok
Added private hardware path
Created hardware path project -> fails

On system/project environment variables
After creating a project there are both system and project environment variables.
I expect that project variables override system variables with the same name -> correct ?

BTW : 
In all my trials today I once had a configuration in that I was able to create HW path projects.
Once I had that and thought to know how I came to that, I cleaned and tried to recreate.
But no luck, Until now I do not know how I managed to do that.

jantje, 2020-05-15:
>Created hardware path project -> fails

can you provide the build log?

>I expect that project variables override system variables with the same name -> correct ?

Yes; as described here https://github.com/Sloeber/arduino-eclipse-plugin/wiki/How-environment-variables-are-set

hreintke, 2020-05-15:
I removed the log file and did a create project.
This is the content of the logfile after that : 
```
!SESSION 2020-05-15 17:00:23.067 -----------------------------------------------
eclipse.buildId=unknown
java.version=1.8.0_131
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

!ENTRY org.eclipse.egit.ui 2 0 2020-05-15 17:00:35.224
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\Herman'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY io.sloeber.arduino.core 1 0 2020-05-15 17:00:50.327
!MESSAGE The folder "C:\Users\Herman\Documents\Arduino\hardware" is empty.

!ENTRY io.sloeber.core.ui 4 0 2020-05-15 17:01:02.270
!MESSAGE Failed to create project
!STACK 0
java.lang.ArrayIndexOutOfBoundsException: -2
	at org.eclipse.core.runtime.Path.segment(Path.java:1053)
	at io.sloeber.core.managers.WorkAround.MakePlatformSloeberTXT(WorkAround.java:217)
	at io.sloeber.core.tools.Helpers.setTheEnvironmentVariables(Helpers.java:759)
	at io.sloeber.core.api.BoardDescriptor.save(BoardDescriptor.java:614)
	at io.sloeber.core.api.BoardDescriptor.createProject(BoardDescriptor.java:554)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.createProjectWrapper(NewSketchWizard.java:127)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard$1.execute(NewSketchWizard.java:94)
	at org.eclipse.ui.actions.WorkspaceModifyOperation.lambda$0(WorkspaceModifyOperation.java:110)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2295)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2322)
	at org.eclipse.ui.actions.WorkspaceModifyOperation.run(WorkspaceModifyOperation.java:131)
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:436)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:352)
	at org.eclipse.jface.wizard.WizardDialog.run(WizardDialog.java:1034)
	at io.sloeber.ui.wizard.newsketch.NewSketchWizard.performFinish(NewSketchWizard.java:99)
	at org.eclipse.jface.wizard.WizardDialog.finishPressed(WizardDialog.java:832)
	at org.eclipse.jface.wizard.WizardDialog.buttonPressed(WizardDialog.java:472)
	at org.eclipse.jface.dialogs.Dialog.lambda$0(Dialog.java:619)
	at org.eclipse.swt.events.SelectionListener$1.widgetSelected(SelectionListener.java:84)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:252)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4173)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1057)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3986)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3585)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:823)
	at org.eclipse.jface.window.Window.open(Window.java:799)
	at io.sloeber.ui.actions.NewSketchHandler.execute(NewSketchHandler.java:29)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:95)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:318)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:252)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:498)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
[... 2 lines cut ...]
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1057)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:3986)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3585)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1160)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
[... cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:660)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:597)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1468)
```

jantje, 2020-05-15:
I think I know what causes the issue and the change I made should cater for this situation.
Can you test?

hreintke, 2020-05-16:
First tests are very positive.
Creating project and compiling from hardware path are working OK.
Even the "original" issue on compiler path is solved -> using the one from hw path as defined in platform.txt

Will do some more testing on different combinations/sequences.

Questions : 
The PKIX issue  : 
In the distribution there is a jre, when building from source not. 
At what time in the process and from where do you add the jre ?

Building sloeber from eclipse environment.
I have eclipse environment with sloeber projects. 
When starting "as eclipse project" I get the error : 

[image]

And also having the PKIX issue.
Did download the JDK11 but can't get eclipse/sloeber to run with that.
hesitant to upgrade my whole system to jdk11 for now.
Do you know how to get around this ?

jantje, 2020-05-16:
>First tests are very positive.

great

>In the distribution there is a jre, when building from source not.
At what time in the process and from where do you add the jre ?

For windows the jre is added after the build on my build server.
For  now you need java 8. CDT will move to 11 in september. New versions of Sloeber will be able to run with java 11 then to (hopefully quickly)
Having a jre folder underneed sloeber will take that jre before any installed version
You can also modify the sloeber.ini to point to the java version you want on your system
Add 
```
-vm
C:/Program Files/Java/jdk-11.0.4/bin/javaw.exe
```
before 
`-vmargs`

It says 11.0.4 because Java version 8 is default on my system

hreintke, 2020-05-16:
Converting to 4.3.3 snapshot now.
What I did was installing sloeber in a different directory
was :  c:\Utilities\sloeber_432
now : c:\utilities\slo433_loc
Workspace is not moved : c:\Userdata\ESP8266|projects

When compiling a project I see still references to the old install location.
```
15:12:39 **** Incremental Build of configuration Release for project Message_UDP_32 ****
"C:\\Utilities\\Sloeber_432\\arduinoPlugin\\tools\\make\\make" all 
Cannot run program "C:\Utilities\Sloeber_432\arduinoPlugin\tools\make\make": Launching failed

Error: Program "C:\Utilities\Sloeber_432\arduinoPlugin\tools\make\make" not found in PATH
PATH=[C:\Utilities\Sloeber_432\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/;;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\system32\Wbem;]

15:12:40 Build Finished. 0 errors, 0 warnings. (took 406ms)
```

Solving this requires each project to move to another platform and back to the correct one.

In project preferences I see where it comes from 
JANTJE.MAKE.LOCATION = c:\Utilities\Sloeber_432\ArduinoPlugin\tools\make
Maybe change to : $(JANTJE.ECLIPSE.LOCATION)\ArduinoPlugin\tools\make would solve this

If so, the same could be applied to the runtime tools : 
A.RUNTIME.TOOLSXTENSA.... = c:\Utilities\Sloeber_432\ArduinoPlugin\packages....
to $(JANTJE.ECLIPSE.LOCATION)\ArduiniPlugin\packages....

That would make it install directory agnostic.

jantje, 2020-05-16:
>When compiling a project I see still references to the old install location.

This may be regression. I'll check.
But you should open the project properties->arduino apply/close
This is needed as each sloeber has its own boardsmanager installation so the platform (containing the used board) may not be on your current sloeber install.

hreintke, 2020-05-16:
Did a quick & dirty patch try for the 
```
JANTJE.MAKE.LOCATION = c:\Utilities\Sloeber_432\ArduinoPlugin\tools\make
Maybe change to : $(JANTJE.ECLIPSE.LOCATION)\ArduinoPlugin\tools\make would solve this
```
and that worked.

If you see it as a valid possibility I will also try the runtime tools ones.
Found the location where they are set.

jantje, 2020-05-16:
This is the location for the tool paths https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L672

JANTJE.MAKE_LOCATION is set at https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L498

Technically I see little benefit to it but I do not object to the change and I look forward to contributions. So go ahead. :-)

hreintke, 2020-05-17:
Working on the tool paths.
Will open issue/PR when I have results.

Closing this issue as being solved by applied patches.

closed 2020-05-17 by hreintke