Sloeber/arduino-eclipse-plugin #1185: Error Compiling ChibiOS for Atmega328p in Sloeber

Avong, 2020-05-12:
Firstly, i apology, i did not see how to report my observation, so i opted to report it as bug. 

I have build errors compiling for Atmega328p and ChibiOS in Sloeber.

ChibiOS is RTOS packaged as arduino library. 
i downloaded, extracted and added the library in C:\MyEclipse\Sloeber\arduinoPlugin\libraries
i then created a project and attached the library by cliking on Arduino Menu-> "add library to the selected project". i checked the chibios folder, whenever i compile, i get errors as shown below

C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:145: undefined reference to `chThdCreateStatic'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:148: undefined reference to `chThdCreateStatic'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:151: undefined reference to `chThdCreateStatic'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:154: undefined reference to `chThdCreateStatic'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `Thread3(void*)':
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:107: undefined reference to `chThdSleep'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `Thread4(void*)':
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:129: undefined reference to `chThdSleep'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:134: undefined reference to `chThdSleep'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:136: undefined reference to `chThdSleep'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `Thread2(void*)':
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:62: undefined reference to `chThdSleep'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o:C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:65: more undefined references to `chThdSleep' follow
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `Thread2(void*)':
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:66: undefined reference to `chSemSignal'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:67: undefined reference to `chThdSleep'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `chThdShouldTerminateX':
C:\MyEclipse\Sloeber\arduinoPlugin\libraries\ChibiOS-B\src/rt/include/chthreads.h:347: undefined reference to `ch'
C:\MyEclipse\Sloeber\arduinoPlugin\libraries\ChibiOS-B\src/rt/include/chthreads.h:347: undefined reference to `ch'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `Thread1(void*)':
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:38: undefined reference to `chSemWait'
C:\MyEclipse\FPNRTOS\Release/..\FPNRTOS.cpp:41: undefined reference to `chThdSleep'
C:\Users\avong\AppData\Local\Temp\ccU7RV4i.ltrans0.ltrans.o: In function `chBegin':
C:\MyEclipse\Sloeber\arduinoPlugin\libraries\ChibiOS-B\src/ChRt.cpp:101: undefined reference to `st_lld_init'
C:\MyEclipse\Sloeber\arduinoPlugin\libraries\ChibiOS-B\src/ChRt.cpp:104: undefined reference to `chSysInit'
collect2.exe: error: ld returned 1 exit status
makefile:81: recipe for target 'FPNRTOS.elf' failed
make: *** [FPNRTOS.elf] Error 1
"C:/MyEclipse/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

but the funny thing, the chibios works fine in arduino ide. 
checking deeply indicates missing path to the chibios library. now dont understand what next to do.

I need helping hand, is there something i am not doing correct?

jantje, 2020-05-12:
>i downloaded, extracted and added the library in C:\MyEclipse\Sloeber\arduinoPlugin\libraries

You should not add content to the arduinoPlugin folder. The correct way to add a "private library" is to add the library location (or the parrent of the library location) to the private libraries folders in preferences->arduino 
[image]

>I need helping hand, is there something i am not doing correct?

If you want people to be able to help/investigate the issue you will have to provide al data needed to reproduce. What comes to mind is: 
1) the location where you downloaded the library
2) the code that uses the library
3) the board that is used (json url and screenshot of project->properties->arduino)

>but the funny thing, the chibios works fine in arduino ide.

Based on the info you provided the library builds fine but there is a link problem. Personally I don't think it is funny.

Avong, 2020-05-12:
> You should not add content to the arduinoPlugin folder. The correct way to add a "private library" is to add the library location (or the parrent of the library location) to the private libraries [...]

Thanks, 
Sorry to bug you, 
I did as you advised but still having compliing issues shown below. 
This same library runs succesfully in arduino ide. 
I am not longer comfortable using arduino ide. the sloeber gives me lots of flexibilities.

see these errors:

C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:397:27: error: expected declaration specifiers before 'attribute'
 void chThdExit(msg_t msg) attribute((used));
                           ^~~~~~~~~
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:398:27: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdExit(msg_t msg) {
                           ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:419:28: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdExitS(msg_t msg) {
                            ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:476:31: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 msg_t chThdWait(thread_t *tp) {
                               ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:512:46: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 msg_t chThdSuspendS(thread_reference_t *trp) {
                                              ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:543:72: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 msg_t chThdSuspendTimeoutS(thread_reference_t *trp, systime_t timeout) {
                                                                        ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:568:55: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdResumeI(thread_reference_t *trp, msg_t msg) {
                                                       ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:592:55: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdResumeS(thread_reference_t *trp, msg_t msg) {
                                                       ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:615:54: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdResume(thread_reference_t *trp, msg_t msg) {
                                                      ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:645:69: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 msg_t chThdEnqueueTimeoutS(threads_queue_t *tqp, systime_t timeout) {
                                                                     ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:665:57: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdDequeueNextI(threads_queue_t *tqp, msg_t msg) {
                                                         ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:680:56: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 void chThdDequeueAllI(threads_queue_t *tqp, msg_t msg) {
                                                        ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:685:1: error: expected '{' at end of input
 }
 ^
C:\MyEclipse\MyLibraries\ChibiOS_AVR\src\utility\chthreads.c:397:22: warning: unused parameter 'msg' [-Wunused-parameter]
 void chThdExit(msg_t msg) attribute((used));
                      ^~~
libraries\ChibiOS_AVR\src\utility\subdir.mk:207: recipe for target 'libraries\ChibiOS_AVR\src\utility\chthreads.c.o' failed
make: *** [libraries\ChibiOS_AVR\src\utility\chthreads.c.o] Error 1
"C:/MyEclipse/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

21:06:14 Build Failed. 15 errors, 1 warnings. (took 706ms)

closed 2020-06-30 by jantje