Sloeber/arduino-eclipse-plugin #1187: Couldn't follow the instructions - not sure how to get started on adding a feature

dajtxx, 2020-05-22:
Hi,

I'll say again, amazing project. So much work put into it!

I've done my best to follow the instructions on building sloeber but am stuck.

The mvn build works fine, and if I do "mvn verify -DskipTests=true" I can use the win32x64.cmd file to run the built version, although I get the warning about long paths because it's built in C:/Users/DAVITAYL/Documents/GitHub/arduino-eclipse-plugin/io.sloeber.product/target/products/io.sloeber.product/win32/win32/x86_64/Sloeber/arduinoPlugin

The command "mvn clean verify -PSDK,latest -DskipTests=true" also works but I don't understand what to do from there.

I assume the zip file I'm looking for is C:\Users\DAVITAYL\Documents\GitHub\arduino-eclipse-plugin\io.sloeber.product.sdk\target\io.sloeber.product.sdk-4.3.3-SNAPSHOT.zip, and it exists.

I unzip that to C:\sloeber.sdk and that dir now has the content listed below.

What do I do now? How do I run this "sdk"? Do I import it into a clean eclipse installation, or one with Java and CDT, or an existing sloeber? I didn't understand the next step in the instructions after the unzipping.

Regards, David.

 Directory of C:\sloeber.sdk

23/05/2020  09:41 AM    <DIR>          .
23/05/2020  09:41 AM    <DIR>          ..
23/05/2020  09:34 AM            74,807 artifacts.jar
23/05/2020  09:35 AM            55,056 artifacts.xml.xz
23/05/2020  09:34 AM    <DIR>          binary
23/05/2020  09:34 AM           388,302 content.jar
23/05/2020  09:35 AM           115,720 content.xml.xz
23/05/2020  09:34 AM    <DIR>          features
23/05/2020  09:35 AM               177 p2.index
23/05/2020  09:34 AM    <DIR>          plugins
               5 File(s)        634,062 bytes

jantje, 2020-05-23:
>I assume the zip file I'm looking for is C:\Users\DAVITAYL\Documents\GitHub\arduino-eclipse-plugin\io.sloeber.product.sdk\target\io.sloeber.product.sdk-4.3.3-SNAPSHOT.zip, and it exists.

That is not the correct file

you need ....\git\arduino-eclipse-plugin\io.sloeber.product.sdk\target\products\sloeber-ide-latest-win32.win32.x86_64.tar.gz
or start ....\git\arduino-eclipse-plugin\io.sloeber.product.sdk\target\products\io.sloeber.product\win32\win32\x86_64\sloeber\sloeber-sdk.exe

dajtxx, 2020-05-23:
Thanks. I extracted that tar.gz to a directory up near the root and it runs.

I had some trouble importing the projects because there are two projects named wifi_dnld and two named wifiHD. I just deselected one of each and got past that.

I have been able to start the core project via run as eclipse app, so it's looking good.

Thanks for your help!

jantje, 2020-05-23:
>I had some trouble importing the projects because there are two projects named wifi_dnld and two named wifiHD. I just deselected one of each and got past that.

I have no clue what you are talking about

>I have been able to start the core project via run as eclipse app, so it's looking good.

great

dajtxx, 2020-05-23:
See the 4 bottom projects. Eclipse gave an error if I had all of them selected. I deselected the 1.2.7 ones and could move on.

[image]

jantje, 2020-05-24:
These are target things and should not be imported.
target is a maven thing. Eclipse does not automagically ignore it. Maybe later.

dajtxx, 2020-05-24:
I was having all sorts of grief using the Eclipse from the tar.gz to run and debug the product. The debug menu items and toolbar were constantly disappearing for one thing. And the target was crashing a lot too.

I downloaded Eclipse 2020-03 java dev, installed the eclipse project sdk and cdt dev tools sdk, and loaded the projects into there. The weirdness has gone.

It's complaining about Description	Resource	Path	Location	Type
Plugin execution not covered by lifecycle configuration: org.eclipse.tycho:tycho-compiler-plugin:1.3.0:compile (execution: default-compile, phase: compile)	pom.xml	/io.sloeber.core	line 4	Maven Project Build Lifecycle Mapping Problem

I assume it means I don't have some maven thing installed, but it's not stopping me modifying and debugging, so I'm ignoring that.

Thanks again for your help.

closed 2020-05-24 by dajtxx