Sloeber/arduino-eclipse-plugin #1188: ld.exe: cannot find -lphy -lpp ...etc

jg1xmv, 2020-05-23:
Hello,

First post on github, I am a beginner.

I am trying to set-up Sloeber to use with Arduino sketches.
I am using Sloeber V4.3.3 downloaded from http://eclipse.baeyens.it/stable.php?OS=Windows.
I am using Win 10 64 bit on my desktop.
The board I am using is an esp8266 D1 Mini Pro.
I created a debug version of a sketch following the instructions on this video: https://www.youtube.com/watch?v=WFzVzgEA8Fo

I think I have gone through most of the unexpected hurdles by reading a lot of the posts under the Sloeber github section but I am now getting stuck with the following error when doing a rebuild:

`14:28:24 **** Incremental Build of configuration espDebug for project 20200522_RootaBoota_01 ****
"C:\\C\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Starting combiner'
"C:\C\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\C\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.1/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\C\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.1/tools/sdk/ld/eagle.app.v6.common.ld"
"C:\C\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,C:\Users\addiv\Documents\sloeber-workspace\20200522_RootaBoota_01\espDebug/20200522_RootaBoota_01.map" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\C\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.1/tools/sdk/lib" "-LC:\C\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.1/tools/sdk/ld" "-LC:\C\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.7.1/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.16m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "C:\Users\addiv\Documents\sloeber-workspace\20200522_RootaBoota_01\espDebug/20200522_RootaBoota_01.elf" -Wl,--start-group    .\sloeber.ino.cpp.o  .\libraries\WiFiManager\WiFiManager.cpp.o  .\libraries\MyLib\MyLib.cpp.o  .\libraries\ESP8266mDNS\src\ESP8266mDNS.cpp.o .\libraries\ESP8266mDNS\src\ESP8266mDNS_Legacy.cpp.o .\libraries\ESP8266mDNS\src\LEAmDNS.cpp.o .\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp.o .\libraries\ESP8266mDNS\src\LEAmDNS_Helpers.cpp.o .\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp.o .\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp.o  .\libraries\ESP8266WiFi\src\BearSSLHelpers.cpp.o .\libraries\ESP8266WiFi\src\CertStoreBearSSL.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGratuitous.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA-WPS.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp.o .\libraries\ESP8266WiFi\src\WiFiClient.cpp.o .\libraries\ESP8266WiFi\src\WiFiClientSecureAxTLS.cpp.o .\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp.o .\libraries\ESP8266WiFi\src\WiFiServer.cpp.o .\libraries\ESP8266WiFi\src\WiFiServerSecureAxTLS.cpp.o .\libraries\ESP8266WiFi\src\WiFiServerSecureBearSSL.cpp.o .\libraries\ESP8266WiFi\src\WiFiUdp.cpp.o  .\libraries\ESP8266WebServer\src\detail\mimetable.cpp.o  .\libraries\DNSServer\src\DNSServer.cpp.o    C:\Users\addiv\Documents\sloeber-workspace\20200522_RootaBoota_01\espDebug/arduino.ar  -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-LC:\Users\addiv\Documents\sloeber-workspace\20200522_RootaBoota_01\espDebug"  
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lphy
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lpp
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lnet80211
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -llwip2
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lwpa
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lcrypto
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lmain
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lwps
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lespnow
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lsmartconfig
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lairkiss
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lwpa2
collect2.exe: error: ld returned 1 exit status
makefile:88: recipe for target '20200522_RootaBoota_01.elf' failed
make: *** [20200522_RootaBoota_01.elf] Error 1
"C:/C/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

14:28:26 Build Failed. 14 errors, 0 warnings. (took 1s.503ms)`

It may be easier on the eye, so here is a screenshot of the Build Console output:
[image]

I can find the xtensa-lx106-elf-ld.exe inside the following Sloeber directory:
C:\C\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\bin

It seems the problem occurs after the combiner starts.
In the settings of the project I found the following relating to the combiner:
[image]

Command: ${A.RECIPE.C.COMBINE.PATTERN}
Command line pattern: ${A.JANTJE.PRE.LINK}${A.RECIPE.C.COMBINE.PATTERN.1} ${FLAGS} ${INPUTS} ${A.RECIPE.C.COMBINE.PATTERN.2} ${A.ARCHIVE_FILE_PATH} ${A.RECIPE.C.COMBINE.PATTERN.3} ${JANTJE.EXTRA.ALL} ${JANTJE.EXTRA.LINK}${A.JANTJE.POST.LINK}
Options are blank.

The only thing I can notice is that what gets found uses backslashes in the path and what does not does not but I am not sure if that is the problem and where to change this sadly.

At my beginner's level, this is as far as I could get, so if anybody has any idea on how to resolve this problem it would be much appreciated!

Thanks in advance!
73,
Arnaud

jg1xmv, 2020-05-23:
Actually, upon double-checking I do not have the following directory:
c:/c/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot find -lphy

Not sure where this 4.8.2 version is coming from.

In my Sloeber directory, I searched for ld.exe and got the following results:
[image]

So it could be that Sloeber is looking inside the wrong directory?

I'll have a go and see if I can find where to change this manually. Or could it be that I need to use version 2.5.0 for this to work since there is such a directory? hmmm

jantje, 2020-05-23:
The jantje hardware esp8266 is currently not building in Sloeber.
It was a specific version for a specific esp8266 platform release.
I havn't been using esp8266 for a long time but I was under the impression there is currently a debug option in esp8266 directly.
See more here https://github.com/jantje/hardware/issues/19

jg1xmv, 2020-05-23:
Ahhhh, okidoki. Thanks for the feedback Jan. I was not aware that we could debug the esp8266 directly now despite some extensive searching on the topic. Do you mean using an Olimex or an ESP-PROG (more for the esp-32) with the esp8266? This approach uses 5 pins on the esp8266 and requires extra hardware which is why I was investigating the potential alternative with Sloeber. If the approach you are thinking about it different, would you mind sharing a link on the approach you are thinking about? I shall do more researching as well on my side in the meantime to see if I find something else. Even though it looks like I will not be using Sloeber for debugging in the end, thanks for all the work you've put into this up until now, much appreciated! Cheers.

jantje, 2020-05-23:
You don't need extra hardware to debug the esp8266 (I don't know about esp32 you should ask in the esp32 forum)
To debug (more or less comfortable) you need debug info in the hex file. Normal build commands do not contain debug info.
Adding `-ggdb -Og` to the link command in the project properties->arduino->compile options should do the trick
The rest of the debugging is still the same like in the video.

jg1xmv, 2020-05-24:
Ok, understood, I will play around with this, thank you for your help. Cheers.

closed 2020-05-24 by jg1xmv