Sloeber/arduino-eclipse-plugin #1193: build in sloeber vs Arduino IDE with ESP8266 plattform >2.5.0

SturmWolfgang, 2020-06-06:
following situation
if I build the following example in Arduino IDE:
https://github.com/4dsystems/GFX4DIoD9/tree/master/Examples/Iod9demo
for a 4D Systems gen4 IoD Range (IOD9TH) 
with ESP8266 platform 2.7.1 installed
everything builds fine - 
on sloeber I get like ~75 errors of different kinds, see below.
only if I switch back to ESP8266 plattform 2.5.0 in sloeber I am able to build the sketch without errors.

any suggestions on how to make this build on current 2.7.1 platform?
due to the amounts of errors I am unsure were to start. Is there a way to thrill the errors down to the root cause? help appreciated.

right now I am have the workaround to switch back to 2.5.0 or use Arduino IDE 

here the console output if build with platform 2.5.2:

I also tried to use different Versions of the SD-Library  - this does not make any difference. 
current Version is 1.2.4

> 10:53:35 **** Incremental Build of configuration Release for project Test001 ****
> "C:\\Programmieren\\S433b\\arduinoPlugin\\tools\\make\\make" all 
> 'Building file: ..\sloeber.ino.cpp'
> 'Starting C++ compile'
> "C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Programmieren\S433b\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include" "-IC:\Programmieren\S433b\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include" "-IC:\Programmieren\S433b\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:\Eigene\Programmieren\Arduino\soeberWS\Test001\Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10812 -DARDUINO_GEN4_IOD -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"GEN4_IOD\""  -DFLASHMODE_DIO  -DESP8266   -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\variants\generic" -I"C:\Programmieren\S433b\arduinoPlugin\libraries\GFX4DIoD9\1.0.0\src" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient\src" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI" -I"C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src" -I"C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
> In file included from C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/FatLib/FatLib.h:27:0,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/SdFat.h:33,
>                  from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:36,
>                  from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src/SD.h:25,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\GFX4DIoD9\1.0.0\src/GFX4dIoD9.h:184,
>                  from ..\sloeber.ino.cpp:9:
> C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/FatLib/ArduinoFiles.h:122:7: error: redefinition of 'class fs::File'
>  class File : public FatFile, public Stream {
>        ^
> In file included from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src/SD.h:24:0,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\GFX4DIoD9\1.0.0\src/GFX4dIoD9.h:184,
>                  from ..\sloeber.ino.cpp:9:
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/FS.h:51:7: error: previous definition of 'class fs::File'
>  class File : public Stream
>        ^
> In file included from C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/FatLib/FatLib.h:28:0,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/SdFat.h:33,
>                  from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:36,
>                  from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src/SD.h:25,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\GFX4DIoD9\1.0.0\src/GFX4dIoD9.h:184,
>                  from ..\sloeber.ino.cpp:9:
> C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/FatLib/FatFileSystem.h: In member function 'fs::File FatFileSystem::open(const char*, oflag_t)':
> C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/FatLib/FatFileSystem.h:95:13: error: 'class fs::File' has no member named 'open'
>      tmpFile.open(vwd(), path, oflag);
>              ^
> In file included from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD\src/SD.h:25:0,
>                  from C:\Programmieren\S433b\arduinoPlugin\libraries\GFX4DIoD9\1.0.0\src/GFX4dIoD9.h:184,
>                  from ..\sloeber.ino.cpp:9:
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: At global scope:
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:190:5: error: 'sdfat' does not name a type
>      sdfat::SdFat* getFs()
>      ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:215:5: error: 'sdfat' does not name a type
>      sdfat::SdFat _fs;
>      ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSImpl::exists(const char*)':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:98:27: error: '_fs' was not declared in this scope
>          return _mounted ? _fs.exists(path) : false;
>                            ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSImpl::rename(const char*, const char*)':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:104:27: error: '_fs' was not declared in this scope
>          return _mounted ? _fs.rename(pathFrom, pathTo) : false;
>                            ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSImpl::info(fs::FSInfo&)':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:113:26: error: '_fs' was not declared in this scope
>          info.blockSize = _fs.vol()->blocksPerCluster() * 512;
[... 480 lines cut ...]
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual size_t sdfs::SDFSFileImpl::position() const':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:278:5: warning: control reaches end of non-void function [-Wreturn-type]
>      }
>      ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual size_t sdfs::SDFSFileImpl::size() const':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:283:5: warning: control reaches end of non-void function [-Wreturn-type]
>      }
>      ^
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSFileImpl::truncate(uint32_t)':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:292:5: warning: control reaches end of non-void function [-Wreturn-type]
>      }
>      ^
[... cut ...]
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSFileImpl::isDirectory() const':
> C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:331:5: warning: control reaches end of non-void function [-Wreturn-type]
>      }
>      ^
> subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
> make: *** [sloeber.ino.cpp.o] Error 1
> "C:/Programmieren/S433b/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
> 
> 10:53:36 Build Failed. 75 errors, 12 warnings. (took 1s.548ms)

jantje, 2020-06-06:
I think this is a duplicate of #1163 
Seems like sloeber added C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4 and I'd be surprised this is the correct one.

I propose to delete the libraries folder and see whether Sloeber finds a better one.

SturmWolfgang, 2020-06-06:
thanks for the fast answer

- Deleting the SD library link in the libraries folder of my project does not help - its coming back - same version - instantly
- Deleting / uninstalling the library does not help either - its coming back during build process

I already assumed that there is something wrong with the regarding SD/FS/File function

jantje, 2020-06-06:
>Deleting the SD library link in the libraries folder of my project does not help - its coming back - same version - instantly

means sloeber is sure this one is needed. Best is to check what libraries Arduino IDE uses or ask esp people what is the difference between the 2 versions.
For testing you can always exclude the SD library from the build

```
from C:\Programmieren\S433b\arduinoPlugin\libraries\SdFat\1.1.4\src/SdFat.h:33,
from C:\Programmieren\S433b\arduinoPlugin\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS\src/SDFS.h:36,
```
You could also have a look at line 36 in SDFS.h to see what is included.

jantje, 2020-06-30:
Is this problem fixed?
Can this issue be closed?

SturmWolfgang, 2020-07-26:
sorry for the late reply - I had to pause sloeber for a while
good news is - today I was able to compile and upload the demo https://github.com/4dsystems/GFX4DIoD9/tree/master/Examples/Iod9demo
here is what I did:
using sloeber 4.3.3,  esp8266 package 2.7.2 on win10
a.) delete all libraries
b.) add libraries manually (I took the same as the arduino IDE uses)
ESP8266HTTPClient v1.2 https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient
ESP8266SdFat v1.1.0 https://github.com/earlephilhower/ESP8266SdFat
ESP8266WiFi v1.0 
GFX4DIoD9  v1.0 https://github.com/4dsystems/GFX4DIoD9
SD 2.0.0 http://www.github.com/esp8266/Arduino
SDFS 0.1.0  https://github.com/esp8266/Arduino
SPI 1.0  http://arduino.cc/en/Reference/SPI
c.) I got an compiler error on "CertStoreBearSSL.cpp" of the ESP8266WiFi library:  'DEBUG_ESP_PORT' was not declared in this scope, so I changed the Debug port from "disable" to "Serial" ; not quite sure why this is necessary - should work with 'disable' as in the arduino IDE it is also 'disabled' 
d.) successfull compilation/upload

topic can be closed

closed 2020-10-12 by jantje