Sloeber/arduino-eclipse-plugin #1195: avr-gcc-ar: Cannot find plugin 'liblto_plugin.so'

speedfixer, 2020-06-14:
The program is basically the BLINK example. It compiles and loads ok with Arduino IDE.
The make file runs fine up to line 73, past all compiling.

I did not change anything in the install: just expand/extract, make a .desktop file and run.
The installation is on a mapped drive, but have since ensured that the client path AND host path are identical. This is a second client system with cleaner (but same) setup.

I find the plugin at:
/mnt/linux/installs/Sloeber/arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/libexec/gcc/avr/7.3.0

The complete error message is:

Starting archiver
"/mnt/linux/installs/Sloeber//arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc-ar" rcs     /rdkl/ardproj/tbb/Release/arduino.ar  ./core/core/CDC.cpp.o ./core/core/HardwareSerial.cpp.o ./core/core/HardwareSerial0.cpp.o ./core/core/HardwareSerial1.cpp.o ./core/core/HardwareSerial2.cpp.o ./core/core/HardwareSerial3.cpp.o ./core/core/IPAddress.cpp.o ./core/core/PluggableUSB.cpp.o ./core/core/Print.cpp.o ./core/core/Stream.cpp.o ./core/core/Tone.cpp.o ./core/core/USBCore.cpp.o ./core/core/WInterrupts.c.o ./core/core/WMath.cpp.o ./core/core/WString.cpp.o ./core/core/abi.cpp.o ./core/core/hooks.c.o ./core/core/main.cpp.o ./core/core/new.cpp.o ./core/core/wiring.c.o ./core/core/wiring_analog.c.o ./core/core/wiring_digital.c.o ./core/core/wiring_pulse.S.o ./core/core/wiring_pulse.c.o ./core/core/wiring_shift.c.o 
/mnt/linux/installs/Sloeber//arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/avr-gcc-ar: Cannot find plugin 'liblto_plugin.so'
makefile:73: recipe for target '/rdkl/ardproj/tbb/Release/arduino.ar' failed
make: *** [/rdkl/ardproj/tbb/Release/arduino.ar] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

I examined the properties/C/C++ Build/Environment section as requested.

at A.COMPILER.AR.COMMAND I added:
--plugin /mnt/linux/.../liblto_plugin.so.0.0.0'

At Verify time, this addition was placed as first entry after 'avr-gcc-ar'. No change: same message.

Perhaps that was not the correct field  to place that option, though I don't think it should be needed.

There is only one #include "tbb.h" include file. Should there be more? (tbb was the name I used for the source.)

I don't know where to look next.

david

speedfixer, 2020-06-14:
While the message is: ' ... cannot find ...', could it be that a wrong version is being found in fact?
Both my systems are not empty or single purpose: there are MANY versions of that plugin for many versions of compilers, and I suspect the same is true on my remote source system as well.

How to ensure that the Sloeber system is trying to use just its own version?
How to discover which path is being used by avr-gcc-ar?
Online searches for this imply that some ar versions may have a fixed relative path, others will pick up a path from the compile environment.

Any idea? Is this a StackExchange question candidate?

david

speedfixer, 2020-06-14:
Discussion of plugin placement and path links, etc. here.

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=84995

jantje, 2020-06-14:
Have you tried copying it to mnt/linux/installs/Sloeber//arduinoPlugin/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin

speedfixer, 2020-06-14:
Did before, tried again just now.
As liblto_plugin.so and as liblto_plugin.so.0.0.0

same result

david

speedfixer, 2020-06-14:
copied liblto_plugin.so and liblto_plugin.so.0.0.0 everywhere reasonable
stop Sloeber; restart; same result

On my local system, there is an appearance of liblto_plugin.so at /usr/lib/bfd-plugins as a link.
That points to  ../gcc/x86_64-linux-gnu/7/liblto_plugin.so  -- clearly not what we want.

I take the bottom of the conversation @ the gcc.gnu address above to mean that:

each toolchain, when run, should place a link into that location (or whatever source location the distro uses) of the desired plugin

Is that a crazy idea?

Let me figure out what the true destination of the current link is so I can restore it, and try again with my own link.

I will report results here.

speedfixer, 2020-06-15:
changed the link in my local system - no change

I logged into that remote system (adjusted links for the source file)
(got about 23:)
`(Sloeber:3860): Gtk-WARNING **: 14:33:13.548: Negative content height -2 (allocation 4, extents 3x3)                                                                                                                      while allocating gadget (node progressbar, owner GtkProgressBar)`

but, ultimately - same result - same error message

jantje, 2020-06-15:
I have no clue what is going on

speedfixer, 2020-06-16:
I'll be back when I find an answer.
Thanks for trying.

david

jantje, 2020-06-16:
Thinking about this
Which version of Sloeber are you on?
I ask because there are several Linux versions for this compiler. Maybe Sloeber takes a wrong one.
You can see all the files downloaded by sloeber in [install location]/arduinoplugin/downloads.
The package.index tells you which file needs to be downloaded and for this package for linux there are several versions. 
Based on the filename you should be able to see which file is downloaded and based on you system specs which one should have been downloaded

```
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-arduino5",
          "systems": [
            {
              "size": "34462042",
              "checksum": "SHA-256:f4acd5531c6b82c715e2edfa0aadb13fb718b4095b3ea1aa1f7fbde680069639",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-arm-linux-gnueabihf.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-arm-linux-gnueabihf.tar.bz2"
            },
            {
              "size": "39381972",
              "checksum": "SHA-256:dd9c70190be370a44fb47dab1514de6d8852b861dfa527964b65c740d8d50c10",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "38492678",
              "checksum": "SHA-256:f48706317f04452544ab90e75bd1bb193f8af2cb1002f53aa702f27202c1b38f",
              "host": "x86_64-apple-darwin14",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2"
            },
            {
              "size": "53727984",
              "checksum": "SHA-256:6d4a5d089a36e5b5252befc73da204555b49e376ce7577ee19ca7f028b295830",
              "host": "i686-mingw32",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-w64-mingw32.zip",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-w64-mingw32.zip"
            },
            {
              "size": "38710087",
              "checksum": "SHA-256:2ff12739d7ed09688d6b3c2c126e8df69b5bda1a07ab558799f0e576571e0e1d",
              "host": "i686-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "39114120",
              "checksum": "SHA-256:3effed8ffa1978b6e4a46f1aa2acc629e440b4d77244f71f9b79a916025206fb",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2"
            }
```

speedfixer, 2020-06-18:
I tried a fresh install just now, local only, no remote links and then was then going to take apart the makefile.

But I have run into your 'only mbed' install issue.
I guess I have to try again with nightly.

jantje, 2020-06-18:
Just go to arduino->preferences->platform and boards and install the boards you want.
Nightly won't fix this

speedfixer, 2020-06-18:
There is only the mbed choice.

I went to ANOTHER fresh system (I'm lucky, I do have a lot of systems at home) - NEW download from website - I get many 'Unable to download https://raw.githubusercontent.com/jantje ...' messages along with constant 'log4j:WARN Please initialize the log4j system properly.' messages in the launch console.

Still - only the ' ... arduino-beta/hardware/mbed' choice - no other choice.
The serial monitor sees my nano connected and properly shows its output (from a previous sketch.)

david

speedfixer, 2020-06-18:
{
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "4.8.1-arduino5"

speedfixer, 2020-06-18:
Checking on the avr-gcc versions:

First entry:
`
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "4.8.1-arduino2"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.0.1-arduino2"`

After finding the correct place, THAT file is not downloaded:
"avr-gcc-4.8.1-arduino2-x86_64-pc-linux-gnu.tar.bz2"

I will d/l and insert manually.
Wonder what else I am missing, and why.
Back here in a bit.
david

jantje, 2020-06-18:
>Still - only the ' ... arduino-beta/hardware/mbed' choice - no other choice.

Can you provide a image?
This is what my current test environment looks like
[image]

speedfixer, 2020-06-18:
I see that screen, all looks OK.
Then, on my file/project start - or after I create the source file from the template - you have to go to the Eclipse properties, not the source file properties to find this, even though it appears to have a platform/board choice in the project files.

Let me play with this a bit. I'll be back.

speedfixer, 2020-06-18:
OK. 
Much better.
now:
`Starting combiner

...
/home/david/Documents/sloeber-workspace/tb1/Release/arduino.ar: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
make: *** [tb1.elf] Error 1`

speedfixer, 2020-06-18:
This issue (can't find plugin) can be closed.
I will chase this other trouble and my remote host problems outside this thread.
Thank you for your patience. Thank you for this more robust Arduino environment.

david

(I WOULD appreciate a hint on why I got that error message above, though.)

closed 2020-06-18 by speedfixer

speedfixer, 2020-06-19:
It seems the 'can't find' problem is a result of doing something other than choosing

project/arduino/arduino sketch/default ino

and not a function of my install location or other software on my systems.
So ... some other fail in the tool chain setup (and may be not any function of the Sloeber system.)

What I WILL be doing is using my own hardware libraries (chips, remote control of Arduino from rPI - hosted remotely. If that works, then I will work on 'arduino cpp program' instead of sketch/.ino program. I have several systems around the house that I use as my develop workstation, and all source on a server somewhere else on my property.

Hopefully, I don't have to visit this problem again.

david

jantje, 2020-06-19:
All that should work fine. But I fear you miss some basic understanding of how Sloeber works and that is making it difficult for you.

speedfixer, 2020-06-20:
I wrote a very long answer to this, with a review of your site. But it was a lot of hot air, really.

Your website: modern home page look; old school simple, honest inside. The 'slick' home page makes me nervous: for me - the inside doesn't match the outside.
 
My problem was simple: I couldn't find enough info on any choices or options that need to be made.

I know the problem sounded very complex, but Eclipse and your software was doing its job. The fault was a simple error in how I was allowed to start a first program file.

You need better docs - and that takes away from your happy time of making the software better. That is every programmers dilemma.

Perhaps you can find a way to host an open wiki and allow some people to help you.

david

jantje, 2020-06-21:
> The fault was a simple error in how I was allowed to start a first program file.

I knew it was something basic you missed.

>Perhaps you can find a way to host an open wiki and allow some people to help you.

The website is here on github

In my experience everybody has his way to find documentation. If the documentation is not found that way the user will have a bad experience.

speedfixer, 2020-06-25:
Actually, it seems to still be a mystery.
The two systems I use the most: both work fine with a local loaded Sloeber, remote private lib and source directories.

But one works with remote source just fine; one does not - with same 'can't find plugin' message.

Maybe some day I will find out what it is. For now, I've got work to do.

Excellent product. Thank you.

david