Sloeber/arduino-eclipse-plugin #1197: !MESSAGE Error: Cannot run program "C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2\bin\avr-g++": Launching failed

Mattpisti, 2020-06-27:
Hi, 
I have used Sloeber to develop a few Arduino programs; I have not been working on Sloeber since May 15 2020.
Today I tried to open Sloeber and it failed; I unistalled and reinstalled but nothing changed.

Since May there have been a few Win10 updates, and I am now on Win10 Pro Version 1909; Java 8 Update 251

Below is what I get when I start Sloeber

Any Idea? Tell me if you need more info and/or  log file 

Thanks

!ENTRY org.eclipse.cdt.core 4 0 2020-06-27 15:39:50.869
!MESSAGE Error: Cannot run program "C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2\bin\avr-g++": Launching failed
!STACK 0
java.io.IOException: Cannot run program "C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2\bin\avr-g++": Launching failed
	at org.eclipse.cdt.utils.spawner.Spawner.exec(Spawner.java:352)
	at org.eclipse.cdt.utils.spawner.Spawner.<init>(Spawner.java:94)
	at org.eclipse.cdt.utils.spawner.ProcessFactory.exec(ProcessFactory.java:91)
	at org.eclipse.cdt.core.CommandLauncher.execute(CommandLauncher.java:211)
	at org.eclipse.cdt.internal.core.BuildRunnerHelper.build(BuildRunnerHelper.java:274)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runProgramForLanguage(AbstractBuiltinSpecsDetector.java:801)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:726)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:591)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:517)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:42)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)

jantje, 2020-06-27:
You no longer have the platform version installed.
You can install the version you created the project with or goto project properties->arduino and select the installed platform version

Mattpisti, 2020-06-27:
Hi and thank you for your reply.

I guess you are right because as soon as I try to open the project
properties I get the following

failed to find the platform
C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\platform.txt\nDid
you deinstall the platform?\nDid you import a project and did not install
the platform?

And checking further, there is no hardware\avr\1.6.23\platform.txt.

How can I reinstall it?

Thanks

Matt

jantje, 2020-06-27:
Why don't you take a look at Arduino->how to Arduino->learn Arduino->instruction video's
Who knows maybe the answer is there

closed 2020-06-27 by jantje

Mattpisti, 2020-06-30:
Maybe it is maybe not; at any rate I reinstalled the platform once more and
when I try to use my old workspace where I have all my previous projects
C:\Users\xxxx\Documents\sloeber-workspace I get this error
C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.23\boards.sloeber.txt
does not exist.
But if I check the directory and file are there....

Switching to the installation created workspace C:\Sloeber-Arduino does not
exibit the error but I had to recreate the projects (copying and pasting
the .ino files etc)

Any clue?

Thanks
Matt