Sloeber/arduino-eclipse-plugin #1200: Can't setup Sloeber properly

hugobpontes, 2020-07-03:
**Describe the bug**
This is more of a request for help than a bug but anyway,
Im a very new linux user (im on ubuntu 20.04) and I installed Eclipse 2019-12 (4.14.0) via apt get install on the terminal, and everything appears to have gone as planned.

Afterwards, I searched for the Sloeber arduino eclipse plugin in the eclipse marketplace and installed it. Initially I get the error
```

Sloeber does not have write access to /snap/eclipse/48 and therefore can not create the folder /snap/eclipse/48/arduinoPlugin

Sloeber might still function but if you get strange results you know where to look.
Do not create an issue if you see this!!!
```
Im aware it says not to report but I wonder if this is what causes the errors that come next after I click continue on the error pop up window.

```
Failed to download "https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_3_2/examples_Arduino_1_8_10.zip".
/snap/eclipse/48/arduinoPlugin/downloads/examples_Arduino_1_8_10.zip

An internal error occurred during: "Finishing the installation ..".
Index 0 out of bounds for length 0

Unable to download https://arduino.esp8266.com/stable/package_esp8266com_index.json
/snap/eclipse/48/arduinoPlugin/package_esp8266com_index.json

Unable to download https://downloads.arduino.cc/libraries/library_index.json
/snap/eclipse/48/arduinoPlugin/library_index.json

Unable to download https://downloads.arduino.cc/packages/package_index.json
/snap/eclipse/48/arduinoPlugin/package_index.json

nable to download https://raw.githubusercontent.com/jantje/ArduinoLibraries/master/library_jantje_index.json
/snap/eclipse/48/arduinoPlugin/library_jantje_index.json

Unable to download https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json
/snap/eclipse/48/arduinoPlugin/package_jantje_index.json

```
If I keep going despite this I eventually get this error when trying to create a sketch

`No boards.txt files found in the arduino hardware folders. I looked in:`

**To Reproduce**
not applicable I think 
**Expected behavior**
I expected that there wouldnt be any errors and that Id be able to use the IDE for arduino programming.

Could anybody tell me what to do? Thanks in advance!

jantje, 2020-07-03:
You could try setting SLOEBER_HOME to a location you have write acces and install using the nightly update site using help->install and using http://eclipse.baeyens.it/update/V4/nightly

jantje, 2020-07-03:
As you are a beginner I strongly advice to install the product. This is download a archive and extract.
http://eclipse.baeyens.it/stable.php?OS=Linux
And read the linux comments here http://eclipse.baeyens.it/installAdvice.shtml

hugobpontes, 2020-07-03:
Thanks for your reply but:

I think the nightly link is broken?

And regarding the producr you suggested, that sounds like a great idea for me but I don't understand how to actually install it. I've read the linux comments but it seems to run on the assumption that I've already installed it. What do I have to run to install it?

jantje, 2020-07-03:
To install: download the archive at http://eclipse.baeyens.it/stable.php?OS=Linux
and unpack it

hugobpontes, 2020-07-04:
> To install: download the archive at http://eclipse.baeyens.it/stable.php?OS=Linux
> and unpack it

Yes but after that how do I run it? Is it through the terminal? Sorry for being such a newbie

hugobpontes, 2020-07-06:
> > To install: download the archive at http://eclipse.baeyens.it/stable.php?OS=Linux
> > and unpack it
> 
> Yes but after that how do I run it? Is it through the terminal? Sorry for being such a newbie

Please! I'm sure it's pretty simple..

jantje, 2020-07-06:
>Please! I'm sure it's pretty simple..

It is. But if you can't figure this one out, I'm sure Sloeber is to far fetched for you.

closed 2020-07-07 by jantje

hugobpontes, 2020-07-07:
$ ./sloeber-ide in the extracted folder.
Like I said I'm very new to Linux and was unsure how exactly this installation works since it differs from Windows. I hadn't realized sloeber-ide was an executable for Linux and was working on other stuff at the moment so I didn't think too much about it, just needing to install fast without much thought, since my colleague told me he'd show me how sloeber works later and I didn't want to bore him with installation details. Eventually I learned about the ls command and that a green filename means its an executable and after that I did that in the extracted folder and then ran it. 
It's really not nice to question my intelligence when I just asked for a quick tip. Would it be so hard to write one line? I assume not since you took the time to write that one snide comment...I'd rather you said nothing really, it would have been just as helpful but less unpleasant. Anyway thanks for the first replies and links which were quite helpful.

erdemuncuoglu, 2020-07-08:
Dear @hugobpontes,

I highly regard your disappointment about not receiving enough support for your issue. I just ask you to consider it that way: 

Sloeber is rather a heavy open source project (without any warranties) which is mainly developed and supported by one man (@jantje) under a bigger stress but for FREE(DOM). Don't compare it with other free (as a beer) open source projects which are backed by corporate giants whom does support the clients more than their freedom. And we (either volunteers or users) should help him more than we ask for his help to keep things running for decades unless you decide to pay for a private consultancy.

Your issue was not related with Sloeber but lack of your Linux knowledge so it was more likely not to get enough support here.

hugobpontes, 2020-07-08:
@erdemuncuoglu 
I understand it completely and am thankful for Sloeber itself and that I got a reply at all, and the reply was actually quite helpful! I was just a bit lost as to how to run it, and I wasn't sure if my inability to run it was something related to Sloeber or simply Linux. I've come to realize it was the latter which makes me realize that I probably could have solved it with a simple google search if I knew the problem was just with my Linux knowledge.
I just found his response quite unpleasant considering how simple it would be to tell me the solution instead of that but it's fine really I was just caught off guard a bit by it and am thankful still like I said :)