Sloeber/arduino-eclipse-plugin #1202: Failed to build project undefined reference to `SystemInit'

hasikp1, 2020-07-09:
I am getting the following errors when new project is started using STM32 core 1.9.0 

```
'Starting combiner'
"C:\STM32Toolchain\eclipse\arduinoPlugin\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3  -mthumb -Os --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0 -Wl,--defsym=LD_MAX_SIZE=65536 -Wl,--defsym=LD_MAX_DATA_SIZE=20480 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common "-Wl,--default-script=C:\STM32Toolchain\eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.9.0\variants\PILL_F103XX/ldscript.ld" "-Wl,--script=C:\STM32Toolchain\eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.9.0/system/ldscript.ld" "-Wl,-Map,D:/Mobitel/eclipseArduino/v5/Release/v5.map"   "-LC:\STM32Toolchain\eclipse\arduinoPlugin\packages\STM32\tools\CMSIS\5.5.1/CMSIS/DSP/Lib/GCC/" -larm_cortexM3l_math -o "D:/Mobitel/eclipseArduino/v5/Release/v5.elf" "-LD:/Mobitel/eclipseArduino/v5/Release" -Wl,--start-group    .\sloeber.ino.cpp.o  .\core\variant\PeripheralPins.c.o .\core\variant\variant.cpp.o     D:/Mobitel/eclipseArduino/v5/Release/arduino.ar  -lc -Wl,--end-group -lm -lgcc -lstdc++
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(startup_stm32yyxx.S.o): in function `LoopFillZerobss':
(.text.Reset_Handler+0x26): undefined reference to `SystemInit'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(main.cpp.o): in function `premain()':
main.cpp:(.text.startup._Z7premainv+0x4): undefined reference to `HAL_NVIC_SetPriorityGrouping'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(board.c.o): in function `init':
board.c:(.text.init+0x0): undefined reference to `hw_config_init'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(HardwareSerial.cpp.o): in function `HardwareSerial::write(unsigned char)':
HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x24): undefined reference to `uart_enable_tx'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x48): undefined reference to `serial_tx_active'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerial5writeEh+0x52): undefined reference to `uart_attach_tx_callback'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(HardwareSerial.cpp.o): in function `HardwareSerial::HardwareSerial(void*, HalfDuplexMode_t)':
HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x52): undefined reference to `pinmap_pin'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x5e): undefined reference to `pinmap_pin'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(HardwareSerial.cpp.o): in function `HardwareSerial::enableHalfDuplexRx()':
HardwareSerial.cpp:(.text._ZN14HardwareSerial18enableHalfDuplexRxEv+0x26): undefined reference to `uart_enable_rx'
c:/stm32toolchain/eclipse/arduinoplugin/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/Mobitel/eclipseArduino/v5/Release/arduino.ar(wiring_time.c.o): in function `millis':
wiring_time.c:(.text.millis+0x0): undefined reference to `getCurrentMillis'
collect2.exe: error: ld returned 1 exit status
makefile:97: recipe for target 'v5.elf' failed
make[1]: *** [v5.elf] Error 1
makefile:76: recipe for target 'all' failed
make: *** [all] Error 2
"C:/STM32Toolchain/eclipse/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

10:23:55 Build Failed. 14 errors, 0 warnings. (took 23s.591ms)
```

Is there a fix for this

jantje, 2020-10-23:
You are probably hitting the max command line length issue in windows #1053

jantje, 2020-11-21:
closing as assuming duplicate of #1053

closed 2020-11-21 by jantje