Sloeber/arduino-eclipse-plugin #1205: 4.3.3 - no variable JANTJE.MAKE_LOCATION

daveharrisuk, 2020-07-17:
Hi
Using Sloeber fine but Win10 pissed me off yet again. Its been on my mind to go Linux, so I became a newbie and 
- formatted my disk
and installed
- single boot Ubuntu 20
- KiCad
- Sloeber 4.3.3 product

Sloeber had Platform and Board PATH issues. Google led me to uninstall Sloeber and then installed arduino 1.8.13 and then re-installed Sloeber.
Arduino and KiCad test fine.
Sloeber better, but Verify of a Blinky, gives...

10:55:43 **** Incremental Build of configuration Release for project test ****
make all
Cannot run program "make": Unknown reason

Error: Program "make" not found in PATH
PATH=[/bin/::/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin]

Project>Properties>C++ Build shows  External Builder and ${JANTJE.MAKE_LOCATION}make
But there is no variable JANTJE.MAKE_LOCATION ?

Dave

jantje, 2020-07-17:
Hi Dave
Linux is supposed to have a make on board. That is why JANTJE.MAKE_LOCATION resolves to nothing on Linux. 
But maybe it is better to read the Linux section on this page http://eclipse.baeyens.it/installAdvice.shtml
That should get you going
I've switched between Linux and Windows a couple of times. They both have their pro's and con's

daveharrisuk, 2020-07-17:
"sudo apt install make" fixed it. Many thanks.

Re: Sloeber had Platform and Board PATH issues until Arduino was in. Is this a deficiency in the instructions or did I miss something? 

I felt glad to kick mIcro$oft out, but Ive kept my other machine Win10... yuk. It has some Win only apps. Dual boot did not 'float my boat'.

jantje, 2020-07-17:
>Re: Sloeber had Platform and Board PATH issues until Arduino was in.
I would not expect this kind of issues. But the link states

> Arduino IDE Linux version comes with an install script that amongst others set udev rules. I have run it on my Linux system.
> I don't know whether running the arduino IDE install script did anything helpfull or harmfull.

daveharrisuk, 2020-07-18:
Hmm... It seems on this combo of versions, Arduino needs to be in first.

jantje, 2020-07-18:
So it seems arduino ide install does helpful stuff
I'll use this ticket to update the documentation

closed 2020-11-21 by jantje