Sloeber/arduino-eclipse-plugin #1206: Error reports Arduino.h functions not recognized - compiles and uploads successfully

slowtoaccept, 2020-07-19:
Get following problem reports when compiling simple sketch:
Function 'delay' could not be resolved	New_Blink.cpp	/New_Blink	line 12	Semantic Error
Function 'delay' could not be resolved	New_Blink.cpp	/New_Blink	line 14	Semantic Error
Function 'digitalWrite' could not be resolved	New_Blink.cpp	/New_Blink	line 11	Semantic Error
Function 'digitalWrite' could not be resolved	New_Blink.cpp	/New_Blink	line 13	Semantic Error
Function 'pinMode' could not be resolved	New_Blink.cpp	/New_Blink	line 6	Semantic Error
Symbol 'HIGH' could not be resolved	New_Blink.cpp	/New_Blink	line 11	Semantic Error
Symbol 'LOW' could not be resolved	New_Blink.cpp	/New_Blink	line 13	Semantic Error
Symbol 'OUTPUT' could not be resolved	New_Blink.cpp	/New_Blink	line 6	Semantic Error

jantje, 2020-07-19:
https://forum.arduino.cc/index.php?topic=439651.msg3289168#msg3289168

closed 2020-07-19 by jantje

slowtoaccept, 2020-07-19:
Context-sensitive help still doesn't like functions and symbols. How do I correct?
[image]