Sloeber/arduino-eclipse-plugin #1208: How to link a static library (.a file) ?

LusterWong, 2020-07-23:
Sloeber v4.3.3 
windows 7 64bit
ESP32
--
I compiled an encryption algorithm in the Arduino IDE, and successfully generated static library files, and tested it in the Arduino IDE.The library directory looks like this (3 files):
LibTest/src/esp32/LibTest.a
LibTest/src/LibTest.h
LibTest/library.properties
There's only one function in LibTest
```
void LibTestP(void)
{
    printf("hello world\n");
}
```
The contents of the LISdsf. A file look like this:
```
name=LibTest
version=1.0.0
author=Test
maintainer=Test
sentence=A Test
paragraph=
category=Other
url=
architectures=*
precompiled=true
ldflags=-Test
```
Sloeber can identify the library and successfully add it to the project. LibTestP() was called in setup() and an error occurred after compilation.
```
undefined reference to `LibTestP(void)'
```

If Eclipse's standard C/C++ project setup has the option to link to a static library, but Sloeber doesn't, can it be added in the next version?Or are there any other solutions at the moment?

jantje, 2020-09-26:
I don't think this will currently work in Sloeber because Sloeber does not support library.properties

As a workaround you could add /LibTest.a to the link command in project properties

jantje, 2021-07-11:
duplicate of #1126

closed 2021-07-11 by jantje