Sloeber/arduino-eclipse-plugin #1209: Can't install V4.3.3 on OS X Catalina

arieavnur, 2020-07-23:
**Describe the bug**
Trying to open the sloeber.app, get "the file is corrupted, move it to trash."
**To Reproduce**
Steps to reproduce the behavior:
1. On OS X 10.15.5
2. Download the 4.3.3 packaged file, or V4.3.3_mac64.2020-04-23_11-36-51.zip 15-58-14-204.zip,
3. Open it with the archive utility as instructed. 
4. Double click on the Sloeber.app file, 
5. Get: “Sloeber.app” is damaged and can’t be opened.
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] )S X 10.15.5
 - Browser [e.g. chrome, safari] Safari
 - Version [e.g. 22]  13.1.1 (15609.2.9.1.2)

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

erdemuncuoglu, 2020-07-24:
Check this out,
[http://eclipse.baeyens.it/stable.php?OS=MacOS](http://eclipse.baeyens.it/stable.php?OS=MacOS)

arieavnur, 2020-07-30:
Same result, see image attachment below.

jantje, 2020-07-31:
The instructions contain another step

> Due to mac new security approach a simple download and extract will not work.
> 
> Follow following steps to install.
> 
> 1)download with mac browser.
> 
> 2)unpacked the sloeber ide zip file with the macOS native Archive Utility. Not in the download folder!!!
> 
> 3)in the folder where you extracted Sloeber run: **sudo xattr -r -d com.apple.quarantine Sloeber.app**
> 
> Read the info on this site at: install->start here-> MacOS comments

arieavnur, 2020-07-31:
But I get from the download a sloeber.app and not sloeber.zip or some other archive...

arieavnur, 2020-07-31:
It worked this time!! Thank you.
But my board—Uno R3 isn’t among the boards list offered by the create project form. 
What do I do? Where do I find documentation?
Thanks,
Arie.

jantje, 2020-08-01:
you are hitting #1196 
Here is the workaround https://github.com/Sloeber/arduino-eclipse-plugin/issues/1196#issuecomment-646379295

closed 2020-09-26 by jantje