Sloeber/arduino-eclipse-plugin #1210: 'Starting combiner' in path string handling has a BUG.

LusterWong, 2020-07-30:
**Using the environment**
Sloeber v4.3.3 (Zip package installation, path: E:\software\sloeber)
win7 64bit
target: ESP32 Arduino 1.0.4

**Step**
1. create workspaces on D:\ names "ESP32_Arduino_Sloeber" .
2. create project names "Build_Test" in the default path (inD:\ESP32_Arduino_Sloeber).
3. Add library to project: BLE、ESP32-BLE-Keyboard、FS、lv_arduino、SPI、SPIFFS、TFT_eSPI。
4. Compile without writing any code.

**BUG**
An error occurred after compiling the file, with the following contents:
```
'Starting combiner'
'Finished building: E:\software\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.4\cores\esp32\wiring_shift.c'
' '
'Starting archiver'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-ar" cru     D:\ESP32_Arduino_Sloeber\Build_Test\Release/arduino.ar  .\core\core\libb64\cdecode.c.o .\core\core\libb64\cencode.c.o  .\core\core\Esp.cpp.o .\core\core\FunctionalInterrupt.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\IPAddress.cpp.o .\core\core\IPv6Address.cpp.o .\core\core\MD5Builder.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\StreamString.cpp.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\base64.cpp.o .\core\core\cbuf.cpp.o .\core\core\esp32-hal-adc.c.o .\core\core\esp32-hal-bt.c.o .\core\core\esp32-hal-cpu.c.o .\core\core\esp32-hal-dac.c.o .\core\core\esp32-hal-gpio.c.o .\core\core\esp32-hal-i2c.c.o .\core\core\esp32-hal-ledc.c.o .\core\core\esp32-hal-matrix.c.o .\core\core\esp32-hal-misc.c.o .\core\core\esp32-hal-psram.c.o .\core\core\esp32-hal-rmt.c.o .\core\core\esp32-hal-sigmadelta.c.o .\core\core\esp32-hal-spi.c.o .\core\core\esp32-hal-time.c.o .\core\core\esp32-hal-timer.c.o .\core\core\esp32-hal-touch.c.o .\core\core\esp32-hal-uart.c.o .\core\core\main.cpp.o .\core\core\stdlib_noniso.c.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o 
'Finished building: D:\ESP32_Arduino_Sloeber\Build_Test\Release\arduino.ar'
' '
'Starting combiner'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group    .\Build_Test.cpp.o  .\libraries\lv_arduino\src\src\lv_widgets\lv_arc.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_bar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btnmatrix.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_calendar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_canvas.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_chart.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_checkbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cont.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cpicker.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_dropdown.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_gauge.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_img.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_imgbtn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_keyboard.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_label.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_led.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_line.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_linemeter.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_list.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_msgbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objmask.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objx_templ.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_page.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_roller.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_slider.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinner.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_switch.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_table.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tabview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_textarea.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tileview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_win.c.o  .\libraries\lv_arduino\src\src\lv_themes\lv_theme.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_empty.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_material.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_mono.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_template.c.o  .\libraries\lv_arduino\src\src\lv_misc\lv_anim.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_area.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_async.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_bidi.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_color.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_debug.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_fs.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_gc.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_ll.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_log.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_math.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_mem.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_printf.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_task.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_templ.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt_ap.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_utils.c.o  .\libraries\lv_arduino\src\src\lv_hal\lv_hal_disp.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_indev.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_tick.c.o  .\libraries\lv_arduino\src\src\lv_gpu\lv_gpu_stm32_dma2d.c.o  .\libraries\lv_arduino\src\src\lv_font\lv_font.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_dejavu_16_persian_hebrew.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_fmt_txt.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12_subpx.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_14.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_16.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_18.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_20.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_22.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_24.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_26.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28_compressed.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_30.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_32.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_34.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_36.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_38.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_40.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_42.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_44.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_46.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_48.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_simsun_16_cjk.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_unscii_8.c.o  .\libraries\lv_arduino\src\src\lv_draw\lv_draw_arc.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_blend.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_img.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_label.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_line.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_mask.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_rect.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_draw_triangle.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_img_buf.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_img_cache.c.o .\libraries\lv_arduino\src\src\lv_draw\lv_img_decoder.c.o  .\libraries\lv_arduino\src\src\lv_core\lv_disp.c.o .\libraries\lv_arduino\src\src\lv_core\lv_group.c.o .\libraries\lv_arduino\src\src\lv_core\lv_indev.c.o .\libr [... cut ...]
14:22:47 Build Failed. 2 errors, 2 warnings. (took 3m:49s.124ms)
```
One of the above words : xtensa-esp32-elf-gcc: error: D:\ESP32_Arduino_Sloeber\uild_Test\Release/arduino.ar
The project name is "Build_Test" but combiner use a path was "\uild_Test", the letter B is missing.
And then I remove all the libraries, and I recompile, and I can compile, which is weird.

**Screenshots**
[image]
Green is project name.
Blue is increase the library
Read is incorrect path string.

How do I fix it, or do I have a problem with it?
Thanks.

LusterWong, 2020-07-30:
‘’‘
'Finished building: D:\NewSloeber\BulidTest\Release\arduino.ar'
' '
'Starting combiner'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group    .\BulidTest.cpp.o  .\libraries\lv_arduino\src\src\lv_widgets\lv_arc.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_bar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btnmatrix.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_calendar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_canvas.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_chart.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_checkbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cont.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cpicker.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_dropdown.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_gauge.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_img.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_imgbtn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_keyboard.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_label.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_led.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_line.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_linemeter.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_list.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_msgbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objmask.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objx_templ.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_page.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_roller.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_slider.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinner.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_switch.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_table.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tabview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_textarea.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tileview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_win.c.o  .\libraries\lv_arduino\src\src\lv_themes\lv_theme.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_empty.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_material.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_mono.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_template.c.o  .\libraries\lv_arduino\src\src\lv_misc\lv_anim.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_area.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_async.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_bidi.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_color.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_debug.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_fs.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_gc.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_ll.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_log.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_math.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_mem.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_printf.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_task.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_templ.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt_ap.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_utils.c.o  .\libraries\lv_arduino\src\src\lv_hal\lv_hal_disp.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_indev.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_tick.c.o  .\libraries\lv_arduino\src\src\lv_gpu\lv_gpu_stm32_dma2d.c.o  .\libraries\lv_arduino\src\src\lv_font\lv_font.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_dejavu_16_persian_hebrew.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_fmt_txt.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12_subpx.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_14.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_16.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_18.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_20.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_22.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_24.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_26.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28_compressed.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_30.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_32.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_34.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_36.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_38.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_40.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_42.c.o .\libraries\lv_arduino\src\ [... cut ...]
makefile:95: recipe for target 'BulidTest.elf' failed
make: *** [BulidTest.elf] Error 1
"E:/software/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

15:10:51 Build Failed. 2 errors, 2 warnings. (took 3m:41s.638ms)
’‘’
I created ws and project again, this time there is no "_" in the name, but the path string also has an error:
“xtensa-esp32-elf-gcc: error: D:\NewSloeber\BulidTest\Reease/arduino.ar”
The correct would be “xtensa-esp32-elf-gcc: error: D:\NewSloeber\BulidTest\Release/arduino.ar”

LusterWong, 2020-07-30:
```
'Starting combiner'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group    .\test.cpp.o  .\libraries\lv_arduino\src\src\lv_widgets\lv_arc.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_bar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btnmatrix.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_calendar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_canvas.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_chart.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_checkbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cont.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cpicker.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_dropdown.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_gauge.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_img.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_imgbtn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_keyboard.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_label.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_led.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_line.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_linemeter.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_list.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_msgbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objmask.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objx_templ.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_page.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_roller.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_slider.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinner.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_switch.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_table.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tabview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_textarea.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tileview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_win.c.o  .\libraries\lv_arduino\src\src\lv_themes\lv_theme.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_empty.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_material.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_mono.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_template.c.o  .\libraries\lv_arduino\src\src\lv_misc\lv_anim.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_area.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_async.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_bidi.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_color.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_debug.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_fs.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_gc.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_ll.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_log.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_math.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_mem.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_printf.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_task.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_templ.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt_ap.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_utils.c.o  .\libraries\lv_arduino\src\src\lv_hal\lv_hal_disp.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_indev.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_tick.c.o  .\libraries\lv_arduino\src\src\lv_gpu\lv_gpu_stm32_dma2d.c.o  .\libraries\lv_arduino\src\src\lv_font\lv_font.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_dejavu_16_persian_hebrew.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_fmt_txt.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12_subpx.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_14.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_16.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_18.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_20.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_22.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_24.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_26.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28_compressed.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_30.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_32.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_34.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_36.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_38.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_40.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_42.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_44.c.o .\libraries\lv_arduino\src\src\lv_f [... cut ...]  -lopenssl -lesp_ringbuf -lfr  -lstdc++ -Wl,--end-group -Wl,-EL -o "D:\sloeber\test\Release/test.elf"
xtensa-esp32-elf-gcc: error: D:\sloeber\test\Release/arduinoar: No such file or directory
makefile:95: recipe for target 'test.elf' failed
make: *** [test.elf] Error 1
"E:/software/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

15:26:13 Build Failed. 2 errors, 2 warnings. (took 3m:50s.201ms)
```
Another case is "arduino.ar" becomes "arduinoar"

jantje, 2020-07-30:
This looks like #1153 which I never could reproduce.
Given the fact that you have this in many instances it looks like something local. Though I do not have a clue what might cause this

LusterWong, 2020-07-31:
> This looks like #1153 which I never could reproduce.
> Given the fact that you have this in many instances it looks like something local. Though I do not have a clue what might cause this
I read the #1153 in detail and it was a bit similar and tried to change the "ESP32-ble-Keyboard" in the example to "ESP32_BLE_Keyboard" for verification, but the same error occurred.

I wondered if there was an occasional problem. After that, I tried 3 computers, all of which installed v4.3.3 sloeber, created workspaces and project, and added the same library, but the same error occurred.

This problem can be 100% repeated. If you can test it, I'm sure you can solve it. Here are the steps.
1. Download these libraries: 
    [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)
    [lv_arduino](https://github.com/lvgl/lv_arduino)
    [ESP32-BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard)
2. Create a new workspaces and arduino project. Board is "ESP32 Wrover Module". Use an cpp  template.
3. Add the library:
    BLE (from the ESP32 Arduino)
    FS (from the ESP32 Arduino)
    SPI (from the ESP32 Arduino)
    SPIFFS (from the ESP32 Arduino)
    TFT_eSPI
    lv_arduino
    ESP32-BLE-Keyboard
4. You don't write any code, compile it, and see the error.

***Screenshots***
[image]

In the above example I tested the error as follows:
"xtensa-esp32-elf-gcc: error: E:\sloeber\test\Release/arduinoar: No such file or directory"
"arduinoar" is the wrong name.......

Thank you for your help.

LusterWong, 2020-07-31:
In addition, #1153 appears to be a problem in the automatically generated Makefile file. I checked the makefile generated by the above example and didn't find any error. "rrduino.ar" is right it doesn't become "arduinoar", which is strange. I really don't know what the problem is.
```
################################################################################
# Automatically-generated file. Do not edit!
################################################################################

-include ..\makefile.init

RM := -del 

# All of the sources participating in the build are defined here
-include sources.mk
-include subdir.mk
-include libraries\lv_arduino\src\src\lv_widgets\subdir.mk
-include libraries\lv_arduino\src\src\lv_themes\subdir.mk
-include libraries\lv_arduino\src\src\lv_misc\subdir.mk
-include libraries\lv_arduino\src\src\lv_hal\subdir.mk
-include libraries\lv_arduino\src\src\lv_gpu\subdir.mk
-include libraries\lv_arduino\src\src\lv_font\subdir.mk
-include libraries\lv_arduino\src\src\lv_draw\subdir.mk
-include libraries\lv_arduino\src\src\lv_core\subdir.mk
-include libraries\lv_arduino\src\porting\subdir.mk
-include libraries\TFT_eSPI\subdir.mk
-include libraries\SPI\src\subdir.mk
-include libraries\SPIFFS\src\subdir.mk
-include libraries\FS\src\subdir.mk
-include libraries\ESP32-BLE-Keyboard\subdir.mk
-include libraries\BLE\src\subdir.mk
-include core\core\libb64\subdir.mk
-include core\core\subdir.mk
-include objects.mk

ifneq ($(MAKECMDGOALS),clean)
ifneq ($(strip $(CC_DEPS)),)
-include $(CC_DEPS)
endif
ifneq ($(strip $(C++_DEPS)),)
-include $(C++_DEPS)
endif
ifneq ($(strip $(C_UPPER_DEPS)),)
-include $(C_UPPER_DEPS)
endif
ifneq ($(strip $(CXX_DEPS)),)
-include $(CXX_DEPS)
endif
ifneq ($(strip $(ASM_DEPS)),)
-include $(ASM_DEPS)
endif
ifneq ($(strip $(INO_DEPS)),)
-include $(INO_DEPS)
endif
ifneq ($(strip $(PDE_DEPS)),)
-include $(PDE_DEPS)
endif
ifneq ($(strip $(S_UPPER_DEPS)),)
-include $(S_UPPER_DEPS)
endif
ifneq ($(strip $(CPP_DEPS)),)
-include $(CPP_DEPS)
endif
ifneq ($(strip $(C_DEPS)),)
-include $(C_DEPS)
endif
endif

-include ..\makefile.defs

# Add inputs and outputs from these tool invocations to the build variables 
AR += \
E:\sloeber\test\Release\arduino.ar \

ELF += \
test.elf \

HEX += \
test.hex \

# All Target
all: test

# Tool invocations
test: $(HEX)
	@echo 'Building target: $@'
	@echo 'Printing size:'
	"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-size" -A "E:\sloeber\test\Release/test.elf"
	@echo 'Finished building target: $@'
	@echo ' '

E:\sloeber\test\Release\arduino.ar: $(AR_OBJ)
	@echo 'Starting archiver'
	"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-ar" cru     E:\sloeber\test\Release/arduino.ar $(AR_OBJ)
	@echo 'Finished building: $@'
	@echo ' '

test.elf: $(LINK_OBJ) $(AR)
	@echo 'Starting combiner'
	"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group   $(LINK_OBJ)   E:\sloeber\test\Release/arduino.ar  -lgcc -lesp32 -lphy -lesp_http_client -lmbedtls -lrtc -lesp_http_server -lbtdm_app -lspiffs -lbootloader_support -lmdns -lnvs_flash -lfatfs -lpp -lnet80211 -ljsmn -lface_detection -llibsodium -lvfs -ldl_lib -llog -lfreertos -lcxx -lsmartconfig_ack -lxtensa-debug-module -lheap -ltcpip_adapter -lmqtt -lulp -lfd -lfb_gfx -lnghttp -lprotocomm -lsmartconfig -lm -lethernet -limage_util -lc_nano -lsoc -ltcp_transport -lc -lmicro-ecc -lface_recognition -ljson -lwpa_supplicant -lmesh -lesp_https_ota -lwpa2 -lexpat -llwip -lwear_levelling -lapp_update -ldriver -lbt -lespnow -lcoap -lasio -lnewlib -lconsole -lapp_trace -lesp32-camera -lhal -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr  -lstdc++ -Wl,--end-group -Wl,-EL -o "E:\sloeber\test\Release/test.elf"
	@echo 'Finished building: $@'
	@echo ' '

test.hex: $(ELF)
	@echo 'Do all objcopy commands'
	${A.RECIPE.OBJCOPY.EEP.PATTERN}
	${A.RECIPE.OBJCOPY.HEX.PATTERN}
	@echo 'Finished building: $@'
	@echo ' '

# Other Targets
clean:
	-$(RM) $(C_UPPER_DEPS)$(PDE_DEPS)$(C_DEPS)$(AR)$(CC_DEPS)$(AR_OBJ)$(C++_DEPS)$(LINK_OBJ)$(CXX_DEPS)$(ASM_DEPS)$(HEX)$(INO_DEPS)$(SIZEDUMMY)$(S_UPPER_DEPS)$(ELF)$(CPP_DEPS)
	-@echo ' '

.PHONY: all clean dependents

-include ..\makefile.targets

```

jantje, 2020-07-31:
If the makefile is ok (and it looks like it is) it must be a make issue.
Do you use the make delivered with Sloeber?

LusterWong, 2020-07-31:
> Do you use the make delivered with Sloeber?

Sorry, I don't quite understand 'make delivered'.

I downloaded here:
[V4.3.3_win64.2020-04-23_11-36-51.zip](http://eclipse.baeyens.it/stable.php?OS=Windows)

The unzip installation path is e:\sloeber.

jantje, 2020-07-31:
Then you should be using the Sloeber delivered make
Can you check project properties->C/C++build "expand Env. Variable refs in Makefile" is on?
Looking at the makefile it seems as if it is not 
The linking command states 
```
@echo 'Starting combiner'
	"E:\software\... $(LINK_OBJ)   E:\sloeber\test\Release/arduino.ar  ... -o "E:\sloeber\test\Release/test.elf"
	@echo 'Finished building: $@'
	@echo ' '
```
Maybe $(LINK_OBJ)  expands to E:\sloeber\test\Release/arduinoar

LusterWong, 2020-07-31:
> Then you should be using the Sloeber delivered make
> Can you check project properties->C/C++build "expand Env. Variable refs in Makefile" is on?
> Looking at the makefile it seems as if it is not
> The [...]

Yes, the options are selected.
[image]
What's the next step?

jantje, 2020-07-31:
>What's the next step? 

run [sloeber install]/arduinoPlugin/tools/make/make in D:\ESP32_Arduino_Sloeber/ble_test/Release
with or without debugging info https://www.gnu.org/software/make/manual/html_node/Options-Summary.html

LusterWong, 2020-08-01:
> > What's the next step?
> 
> run [sloeber install]/arduinoPlugin/tools/make/make in D:\ESP32_Arduino_Sloeber/ble_test/Release
> with or without debugging info https://www.gnu.org/software/make/manual/html_node/Options-Summary.html

Sorry, I'm very unfamiliar with the command line. I tried three ways, but it didn't work...
1. E:\software\Sloeber\arduinoPlugin\tools\make>make D:\ESP32_Arduino_Sloeber\Build_Test\Release
    **make: Nothing to be done for 'D:\ESP32_Arduino_Sloeber\Build_Test\Release**
2. E:\software\Sloeber\arduinoPlugin\tools\make>make all D:\ESP32_Arduino_Sloeber\Build_Test\Release
    **make: *** No rule to make target 'all'.  Stop.**
3. E:\software\Sloeber\arduinoPlugin\tools\make>make -f D:\ESP32_Arduino_Sloeber\Build_Test\Release\makefile
    **make: *** No rule to make target 'ar', needed by 'Build_Test.elf'.  Stop.**

The above commands are search on network, I do not understand the specific meaning, could you give an example for that path, thank you.

LusterWong, 2020-08-01:
I tried the following steps again:
1. Compile the project with an error, as described previously. 
```
'Starting archiver'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-ar" cru     D:\ESP32_Arduino_Sloeber\Build_Test\Release/arduino.ar  .\core\core\libb64\cdecode.c.o .\core\core\libb64\cencode.c.o  .\core\core\Esp.cpp.o .\core\core\FunctionalInterrupt.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\IPAddress.cpp.o .\core\core\IPv6Address.cpp.o .\core\core\MD5Builder.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\StreamString.cpp.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\base64.cpp.o .\core\core\cbuf.cpp.o .\core\core\esp32-hal-adc.c.o .\core\core\esp32-hal-bt.c.o .\core\core\esp32-hal-cpu.c.o .\core\core\esp32-hal-dac.c.o .\core\core\esp32-hal-gpio.c.o .\core\core\esp32-hal-i2c.c.o .\core\core\esp32-hal-ledc.c.o .\core\core\esp32-hal-matrix.c.o .\core\core\esp32-hal-misc.c.o .\core\core\esp32-hal-psram.c.o .\core\core\esp32-hal-rmt.c.o .\core\core\esp32-hal-sigmadelta.c.o .\core\core\esp32-hal-spi.c.o .\core\core\esp32-hal-time.c.o .\core\core\esp32-hal-timer.c.o .\core\core\esp32-hal-touch.c.o .\core\core\esp32-hal-uart.c.o .\core\core\main.cpp.o .\core\core\stdlib_noniso.c.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o 
'Finished building: D:\ESP32_Arduino_Sloeber\Build_Test\Release\arduino.ar'
' '
'Starting combiner'
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group    .\Build_Test.cpp.o  .\libraries\lv_arduino\src\src\lv_widgets\lv_arc.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_bar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btnmatrix.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_calendar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_canvas.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_chart.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_checkbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cont.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cpicker.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_dropdown.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_gauge.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_img.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_imgbtn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_keyboard.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_label.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_led.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_line.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_linemeter.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_list.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_msgbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objmask.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objx_templ.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_page.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_roller.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_slider.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinner.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_switch.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_table.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tabview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_textarea.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tileview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_win.c.o  .\libraries\lv_arduino\src\src\lv_themes\lv_theme.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_empty.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_material.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_mono.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_template.c.o  .\libraries\lv_arduino\src\src\lv_misc\lv_anim.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_area.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_async.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_bidi.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_color.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_debug.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_fs.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_gc.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_ll.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_log.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_math.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_mem.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_printf.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_task.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_templ.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt_ap.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_utils.c.o  .\libraries\lv_arduino\src\src\lv_hal\lv_hal_disp.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_indev.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_tick.c.o  .\libraries\lv_arduino\src\src\lv_gpu\lv_gpu_stm32_dma2d.c.o  .\libraries\lv [... cut ...]
"E:/software/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
```
As shown in you "xtensa-esp32-elf-gcc: error: D:\ESP32_Arduino_Sloeber\uild_Test\Release/arduino.ar: No such file or directory" that "Build_Test" become "uild_Test" string error, an **arduino.ar** is generated.

2. Use cmdline (CMD) enter the directory "D:\ESP32_Arduino_Sloeber\Build_Test\Release" and then call the command : 
    E:\software\Sloeber\arduinoPlugin\tools\make\make all
    
[image]

   The same error occurs after execution:
   
[image]

    Build_Test" become "uild_Test".

LusterWong, 2020-08-01:
It's incredible. 
After I modified the project name from "Build_Test" to "B", I recompiled it and it actually succeeded.

[image]

Maybe the same workspaces project can only name a single letter ? 
I think it seems to be a problem with character processing .

LusterWong, 2020-08-01:
I tried to change the project name and then compiled to get the following results:

**1. project name "_1_B"**
```
"E:\software\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/lib" "-LE:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group    .\Main.cpp.o  .\libraries\lv_arduino\src\src\lv_widgets\lv_arc.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_bar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_btnmatrix.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_calendar.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_canvas.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_chart.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_checkbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cont.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_cpicker.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_dropdown.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_gauge.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_img.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_imgbtn.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_keyboard.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_label.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_led.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_line.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_linemeter.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_list.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_msgbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objmask.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_objx_templ.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_page.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_roller.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_slider.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinbox.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_spinner.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_switch.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_table.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tabview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_textarea.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_tileview.c.o .\libraries\lv_arduino\src\src\lv_widgets\lv_win.c.o  .\libraries\lv_arduino\src\src\lv_themes\lv_theme.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_empty.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_material.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_mono.c.o .\libraries\lv_arduino\src\src\lv_themes\lv_theme_template.c.o  .\libraries\lv_arduino\src\src\lv_misc\lv_anim.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_area.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_async.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_bidi.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_color.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_debug.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_fs.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_gc.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_ll.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_log.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_math.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_mem.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_printf.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_task.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_templ.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_txt_ap.c.o .\libraries\lv_arduino\src\src\lv_misc\lv_utils.c.o  .\libraries\lv_arduino\src\src\lv_hal\lv_hal_disp.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_indev.c.o .\libraries\lv_arduino\src\src\lv_hal\lv_hal_tick.c.o  .\libraries\lv_arduino\src\src\lv_gpu\lv_gpu_stm32_dma2d.c.o  .\libraries\lv_arduino\src\src\lv_font\lv_font.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_dejavu_16_persian_hebrew.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_fmt_txt.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_12_subpx.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_14.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_16.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_18.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_20.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_22.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_24.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_26.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_28_compressed.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_30.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_32.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_34.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_36.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_38.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_40.c.o .\libraries\lv_arduino\src\src\lv_font\lv_font_montserrat_42.c.o .\librari [... cut ...] al -lprotobuf-c -lsdmmc -lcore -lpthread -lcoexist -lfreemodbus -lspi_flash -lesp-tls -lwpa -lwifi_provisioning -lwps -lesp_adc_cal -lesp_event -lopenssl -lesp_ringbuf -lfr  -lstdc++ -Wl,--end-group -Wl,-EL -o "D:\ESP32_Arduino_Sloeber\1_2_3\Release/1_2_3.elf"
xtensa-esp32-elf-gcc: error: D:\ESP32_Arduino_Sloeber\1_2_3\elease/arduino.ar: No such file or directory
makefile:95: recipe for target '1_2_3.elf' failed
make: *** [1_2_3.elf] Error 1
```
"Release" become "elease", the letter R is missing

jantje, 2020-08-01:
The fact this is so reproducible makes me think we are actually not looking at the right thing.
First some feedback

>Sorry, I'm very unfamiliar with the command line. I tried three ways, but it didn't work...
>
>    E:\software\Sloeber\arduinoPlugin\tools\make>make D:\ESP32_Arduino_Sloeber\Build_Test\Release
>    make: Nothing to be done for 'D:\ESP32_Arduino_Sloeber\Build_Test\Release

This is the right command. What is weird is that make tells you all is build fine. Which goes against the fact the build fails.

Anyway.
The combine command is 92332 characters long. arduino.ar is around position 8326. Windows has a maximum command line of 8191.
I'm pretty sure that is the problem you are experiencing.
This makes this issue a duplicate of #1053 
To work around that rename [sloeber install]/arduinoPlugin/tools/make to [sloeber install]/arduinoPlugin/tools/make.org
download https://github.com/Sloeber/arduino-eclipse-plugin/files/2695478/make.zip
extract the make.zip in [sloeber install]/arduinoPlugin/tools
If I remember correctly this package cause some issues with other stuff (like escaping ") so I havn't moved to this make yet.
Actually it is the same make but sh is included overcoming the cmd command line limitation

keep me posted

LusterWong, 2020-08-01:
So amazing!
The compilation succeeded after replacing the file inside the make.zip.
Some issues you mentioned have not been tested yet, I will continue to
test, and If there is any exception, I  will post it.

I can continue my work and sleep happily. Thank you very much for your help
!

wmarkow, 2021-01-03:
Hi all. Now I have this issue reproducible - but only on one of my laptops. The issue is that a letter is swallowed from the path and **make.exe** complains that the path is invalid or it doesn't exist. I'm building some application with ESP32 package that uses NimBle-Arduino library. There is a lot of files to compile but the error comes from combiner. I'm not quite sure how it is related to the limit of cmd.exe command length. I have noticed that some compile commands are longer than 8192 and they succeed. It fails only while linking.

First laptop:
* Windows 10
* sloeber-4.3.3 with internal make
* cmd.exe limit seems to be 8192 because when I paste a long command to it, it gets cut after 8191 character
* the issue is not reproducible

Second laptop:
* Windows 7
* sloeber-4.3.3 with internal make (the same as on first laptop)
* cmd.exe limit seems to be 8192 because when I paste a long command to it, it gets cut after 8191 character
* the issue is reproducible

The **make** seems to be old:
````
C:\prj\dev-test\sources\arduino\arduino-projects\arduino-sandbox\LegoinoTrains\Release>C:\prj\dev-test\tools\sloeber-4.3.3\arduinoPlugin\tools\make\make --version
GNU Make 3.82.90
Built for i686-pc-mingw32
Copyright (C) 1988-2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
````
Interesting. Maybe this **make.exe** works better with Windows 10?

uzi18, 2021-01-03:
https://docs.microsoft.com/en-us/troubleshoot/windows-client/shell-experience/command-line-string-limitation

wmarkow, 2021-01-04:
This article doesn't help much. It mostly says that the characters limit is 2047. Better explanation about limits I found here:
https://mcuoneclipse.com/2015/03/29/solving-the-8192-character-command-line-limit-on-windows/
https://devblogs.microsoft.com/oldnewthing/20031210-00/?p=41553

I tested on Windows 7 the make-4.2.1 from the attached https://github.com/Sloeber/arduino-eclipse-plugin/files/2695478/make.zip It works nice, I mean the issue is solved with this version of make (4.2.1). It also works on Windows 10 (however the issue is not reproducible there even with the older make version). It even works when I just use the **make.exe** without any **sh.exe** etc. Now I think that there was some bug in **make** which is already fixed.
I did more tests and found the first **make** version which fixed the issue:
make version|issue status|download URL|remark|
--------------|-------------|----------------|--------
|4.3.1|fixed|?||
|4.2.1|fixed|?||
|3.99.92|fixed|?||
|3.99.90|fixed|?||
|3.82.90|not fixed| |this make comes from sloeber-4.3.2|

There must be some commit done between **3.82.90** and **3.99.90** and - yes - I already saw in [make repo](https://savannah.gnu.org/git/?group=make) - there was a lot of commits between those two versions.
In my opinion this issue may be not an exact duplicate of #1053 but is automatically fixed by the workaround documented there: which is to use **make-4.2.1**.

jantje, 2021-01-04:
1) The problem of truncating commands is in the windows component cmd.exe.

2) The difference between the 2 make.zip's is that one contains make.exe and sh.exe the other only make.exe.

3) If there is a sh.exe in the folder next to make.exe; make.exe will use sh.exe and not cmd.exe (at least with the supplied make.exe)

As sh.exe does not have the truncation problem I hoped to switch to sh.exe.

However

1) sh.exe behaves very similarly to cmd.exe 

2) but there are differences.

These differences between make.exe and sh.exe cause problems for certain boards.

So the zip with make.exe and sh.exe can be used as a workaround but not as general solution.

LusterWong, 2021-03-03:
> 1. The problem of truncating commands is in the windows component cmd.exe.
> 2. The difference between the 2 make.zip's is that one contains make.exe and sh.exe the other only make.exe.
> 3. If there is [...]

Today, I found that the Arduino of ESP32 updated 1.0.5. I tried not to modify the code compilation, but there was an error:
```
"F:\\software\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
F:/software/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "D:\App\New\partitions.csv" copy /y "D:\App\New\partitions.csv" "D:\App\New\Release\partitions.csv"
cmd /c if not exist "D:\App\New\Release\partitions.csv" copy "F:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.5\tools\partitionsmin_spiffs.csv" "D:\App\New\Release\partitions.csv"
make[1]: *** [makefile:105: pre-build] Error 1
make: *** [makefile:64: all] Error 2
"F:/software/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.
```

The mistake is:
```
"F:\software\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/1.0.5\tools\partitionsmin_spiffs.csv" 
```
The path is wrong. Missing character " \"

My sloeber is 4.3.2, not up to date. I don't know if it's make.exe(4.2.1), but ESP32 Arduino V1.0.4 doesn't have this problem

tyllurius, 2021-04-02:
Great! Updating the make works for me, too! Thank you!!

I can confirm that this issue doesn't appear on Win 10 even with old make.
Another great side effect is that it speeds up the compiling on my Win 7 PC: I had a weird issue that after a while of compiling, half or so, it slowed terribly down, not taking full CPU load anymore, and took 5:30 min. With the the make, it runs the whole compiling process with full load, taking only 1:17 min. No clue why, but it's great ;)

jantje, 2021-04-04:
I tried to run my default.ino on all boards test with the sh version and I get errors with ESP32 in regards to the prebuild commands
I could fix the first prebuild command but I fail to understand what is the problem with the second prebuild command

```

12:29:32 **** Incremental Build of configuration Release for project 00573_esp32cam ****
"e:\\platform\\arduinoPlugin\\tools\\make\\make" -j4 all 
e:/platform/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "L:\DefaultInoOnAllBoards\00573_esp32cam\partitions.csv" copy /y "L:\DefaultInoOnAllBoards\00573_esp32cam\partitions.csv" "L:\DefaultInoOnAllBoards\00573_esp32cam\Release\partitions.csv"
cmd /c if not exist "L:\DefaultInoOnAllBoards\00573_esp32cam\Release\partitions.csv" copy "e:\platform\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6\tools\partitions\huge_app.csv" "L:\DefaultInoOnAllBoards\00573_esp32cam\Release\partitions.csv"
process_begin: CreateProcess(L:\DefaultInoOnAllBoards\00573_esp32cam\Release\make7480-1.bat, L:\DefaultInoOnAllBoards\00573_esp32cam\Release\make7480-1.bat, ...) failed.
make (e=267): The directory name is invalid.

make[1]: [makefile:106: pre-build] Error 267 (ignored)
e:/platform/arduinoPlugin/tools/make/make --no-print-directory main-build
process_begin: CreateProcess(L:\DefaultInoOnAllBoards\00573_esp32cam\Release\make11000-1.bat, L:\DefaultInoOnAllBoards\00573_esp32cam\Release\make11000-1.bat, ...) failed.
make (e=267): The directory name is invalid.

make[1]: *** [subdir.mk:18: sketch.cpp.o] Error 267
make: *** [makefile:65: all] Error 2
"e:/platform/arduinoPlugin/tools/make/make -j4 all" terminated with exit code 2. Build might be incomplete.

12:29:33 Build Failed. 3 errors, 0 warnings. (took 704ms)
```

jantje, 2021-04-04:
I'm reopening this as I want the workaround to become the standard solution.

mikkeaa, 2021-04-20:
Thanks jantje as you pointed me to this solution. Works very good.

jantje, 2021-07-11:
Closing as I'm releasing 4.4

closed 2021-07-11 by jantje

patricklaf, 2025-06-20:
Seems the problem is back.

Eclipse CDT 2025-03 and 2025-06
Sloeber plugin  4.4.3.202402080117
OS: Windows 10   

Build fails at combiner step.

I am using STM32 core 2.10.1, so SrcWrapper library, and lvgl version 9.2.2, both are quite huge in term of number of files.

Partial log shows error in CreateProcess:

```
process_begin: CreateProcess(C:\Eclipse\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\14.2.1-1.1\bin\arm-none-eabi-gcc.exe, C:\Eclipse\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\14.2.1-1.1/bin/arm-none-eabi-gcc -mcpu=cortex-m7 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Og -g -DNDEBUG --specs=nano.specs -u _printf_float -Wl,--defsym=LD_FLASH_OFFSET=0x0 -Wl,--defsym=LD_MAX_SIZE=1048576 -Wl,--defsym=LD_MAX_DATA_SIZE=327680 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--default-script=C:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.10.1\variants\STM32F7xx\F746B(E-G)T_F746N(E-G)H_F750N8H_F756BGT_F756NGH/ldscript.ld -Wl,--script=C:\Eclipse\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.10.1\system/ldscript.ld -Wl,-Map,C:\Documents\Eclipse\Ramure\Ramure\Release/Ramure.map -Wl,--no-warn-rwx-segments -o C:\Documents\Eclipse\Ramure\Ramure\Release/Ramure.elf -LC:\Documents\Eclipse\Ramure\Ramure\Release -Wl,--start-group .\FT5336.cpp.o .\LTDC.cpp.o .\SDRAM.cpp.o .\lv_font_montserrat_fr_14.c.o .\sloeber.ino.cpp.o .\libraries\lvgl\src\widgets\win\lv_win.c.o .\libraries\lvgl\src\widgets\tileview\lv_tileview.c.o .\libraries\lvgl\src\widgets\textarea\lv_textarea.c.o .\libraries\lvgl\src\widgets\tabview\lv_tabview.c.o .\libraries\lvgl\src\widgets\table\lv_table.c.o .\libraries\lvgl\src\widgets\switch\lv_switch.c.o .\libraries\lvgl\src\widgets\spinner\lv_spinner.c.o .\libraries\lvgl\src\widgets\spinbox\lv_spinbox.c.o .\libraries\lvgl\src\widgets\span\lv_span.c.o .\libraries\lvgl\src\widgets\slider\lv_slider.c.o .\libraries\lvgl\src\widgets\scale\lv_scale.c.o .\libraries\lvgl\src\widgets\roller\lv_roller.c.o .\libraries\lvgl\src\widgets\property\lv_dropdown_properties.c.o .\libraries\lvgl\src\widgets\property\lv_image_properties.c.o .\libraries\lvgl\src\widgets\property\lv_keyboard_properties.c.o .\libraries\lvgl\src\widgets\property\lv_label_properties.c.o .\libraries\lvgl\src\widgets\property\lv_obj_properties.c.o .\libraries\lvgl\src\widgets\property\lv_roller_properties.c.o .\libraries\lvgl\src\widgets\property\lv_style_properties.c.o .\libraries\lvgl\src\widgets\property\lv_textarea_properties.c.o .\libraries\lvgl\src\widgets\objx_templ\lv_objx_templ.c.o .\libraries\lvgl\src\widgets\msgbox\lv_msgbox.c.o .\libraries\lvgl\src\widgets\menu\lv_menu.c.o .\libraries\lvgl\src\widgets\lottie\lv_lottie.c.o .\libraries\lvgl\src\widgets\list\lv_list.c.o .\libraries\lvgl\src\widgets\line\lv_line.c.o .\libraries\lvgl\src\widgets\led\lv_led.c.o .\libraries\lvgl\src\widgets\label\lv_label.c.o .\libraries\lvgl\src\widgets\keyboard\lv_keyboard.c.o .\libraries\lvgl\src\widgets\imagebutton\lv_imagebutton.c.o .\libraries\lvgl\src\widgets\image\lv_image.c.o .\libraries\lvgl\src\widgets\dropdown\lv_dropdown.c.o .\libraries\lvgl\src\widgets\checkbox\lv_checkbox.c.o .\libraries\lvgl\src\widgets\chart\lv_chart.c.o .\libraries\lvgl\src\widgets\canvas\lv_canvas.c.o .\libraries\lvgl\src\widgets\calendar\lv_calendar.c.o .\libraries\lvgl\src\widgets\calendar\lv_calendar_chinese.c.o .\libraries\lvgl\src\widgets\calendar\lv_calendar_header_arrow.c.o .\libraries\lvgl\src\widgets\calendar\lv_calendar_header_dropdown.c.o .\libraries\lvgl\src\widgets\buttonmatrix\lv_buttonmatrix.c.o .\libraries\lvgl\src\widgets\button\lv_button.c.o .\libraries\lvgl\src\widgets\bar\lv_bar.c.o .\libraries\lvgl\src\widgets\arc\lv_arc.c.o .\libraries\lvgl\src\widgets\animimage\lv_animimage.c.o .\libraries\lvgl\src\tick\lv_tick.c.o .\libraries\lvgl\src\themes\simple\lv_theme_simple.c.o .\libraries\lvgl\src\themes\mono\lv_theme_mono.c.o .\libraries\lvgl\src\themes\default\lv_theme_default.c.o .\libraries\lvgl\src\themes\lv_theme.c.o .\libraries\lvgl\src\stdlib\rtthread\lv_mem_core_rtthread.c.o .\libraries\lvgl\src\stdlib\rtthread\lv_sprintf_rtthread.c.o .\libraries\lvgl\src\stdlib\rtthread\lv_string_rtthread.c.o .\libraries\lvgl\src\stdlib\micropython\lv_mem_core_micropython.c.o .\libraries\lvgl\src\stdlib\clib\lv_mem_core_clib.c.o .\libraries\lvgl\src\stdlib\clib\lv_sprintf_clib.c.o .\libraries\lvgl\src\stdlib\clib\lv_string_clib.c.o .\libraries\lvgl\src\stdlib\builtin\lv_mem_core_builtin.c.o .\libraries\lvgl\src\stdlib\builtin\lv_sprintf_builtin.c.o .\libraries\lvgl\src\stdlib\builtin\lv_string_builtin.c.o .\libraries\lvgl\src\stdlib\builtin\lv_tlsf.c.o .\libraries\lvgl\src\stdlib\lv_mem.c.o .\libraries\lvgl\src\others\vg_lite_tvg\vg_lite_matrix.c.o .\libraries\lvgl\src\others\vg_lite_tvg\vg_lite_tvg.cpp.o .\libraries\lvgl\src\others\sysmon\lv_sysmon.c.o .\libraries\lvgl\src\others\snapshot\lv_snapshot.c.o .\libraries\lvgl\src\others\observer\lv_observer.c.o .\libraries\lvgl\src\others\monkey\lv_monkey.c.o .\libraries\lvgl\src\others\imgfont\lv_imgfont.c.o .\libraries\lvgl\src\others\ime\lv_ime_pinyin.c.o .\libraries\lvgl\src\others\gridnav\lv_gridnav.c.o .\libraries\lvgl\src\others\fragment\lv_fragment.c.o .\libraries\lvgl\src\others\fragment\lv_fragment_manager.c.o .\libraries\lvgl\src\others\file_explorer\lv_file_explorer.c.o .\libraries\lvgl\src\osal\lv_cmsis_rtos2.c.o .\libra [... cut ...]  .\core\variant\PeripheralPins_DISCO_F746NG.c.o .\core\variant\generic_clock.c.o .\core\variant\variant_DISCO_F746NG.cpp.o .\core\variant\variant_generic.cpp.o C:\Documents\Eclipse\Ramure\Ramure\Release\arduino.ar -lc -Wl,--end-group -lm -lgcc -lstdc++, ...) failed.
make (e=87): Param�tre incorrect.

make[1]: *** [makefile:258: Ramure.elf] Error 87
make: *** [makefile:237: all] Error 2
"C:/Eclipse/arduinoPlugin/tools/make/make -j24 all" terminated with exit code 2. Build might be incomplete.
```

jantje, 2025-06-20:
There are 2 possible explenations if you have this problem.
1) you use the wrong make.exe (old install; a make found on your system ...)
2) The board now has so long command lines even the better make.exe thinks it is to long.

patricklaf, 2025-06-21:
Hello @jantje ,
I think that it is answer 2. You can see in the last log line that the make invoked is the one installed with the plugin.

jantje, 2025-06-21:
>Is it possible to write to a file the arguments and invoke make with https://github.com/file parameter?

Yes but that means a file needs to be created for each source file (they may be different) and I don't know how it will work with files using @file (which they would do if I do not parse these out.

What may help though is placing sloeber as close to the root as possible (or define the SLOEBER_HOME variable as something close to the root like d:\) as Sloeber uses FQN this may take away lots of the command line.

jantje, 2025-06-22:
>I copied the failed command content from the console. I cut the gcc command and save all parameters in a file name `Release\combiner'. I replaced all \ by /.

This worked in your case but may not work in every case. So I do not believe in automating that.

Now Sloeber is closer to the root also putting your workspace closer to the root may help. Now it is at C:\Documents\Eclipse\Ramure

One thing that really bugs me is that Sloeber uses "arduinoPlugin" and Arduino uses "packages" which are way to long.
Shortening these is easy but creates a downwards (and arduino framework) incompatibility which made me hold off but maybe I can create a branch with these changes and you can see whether this helps.

Also
I created V5 (I should release it) which runs by default without make. Maybe it works that way (command line length is variable in windows in all kinds of way and only god knows what works where)

patricklaf, 2025-06-23:
> Now Sloeber is closer to the root also putting your workspace closer to the root may help. Now it is at C:\Documents\Eclipse\Ramure

It's not that important. Workspace path is not used a lot in the problematic command (only twice) and all the object file path are relative to workspace, like `.\libraries\lvgl\src\misc\lv_log.c.o`.

> Also I created V5 (I should release it) which runs by default without make. Maybe it works that way (command line length is variable in windows in all kinds of way and only god knows what works where)

Yes, I know but not tested yet. I use fresh Eclipse-CDT installation and add Sloeber plugin. I am waiting for the V5 plugin availability.

jantje, 2025-06-24:
>I tried latest V5 nightly. Problem is still here:

That is annoying.