Sloeber/arduino-eclipse-plugin #1211: Importing sloeber project is causing links and compile settings to be changed

BriscoeTech, 2020-07-31:
Hello Everyone,

I have run into some interesting bugs when trying to share sloaber projects. When sloeber imports an existing sloeber project from another location for the first time, it is changing the existing links and compiler includes. These bugs are making it very difficult to use git and share projects with a new developer, as the settings are getting mangled when they import it for the first time and they have no idea whats going on since they have never seen the code before.

Sloeber allowing to link to local libraries within the project is wonderfull, and is allowing me to version control my libraries with the projects using them. This bug is breaking that however when it is shared, the import is changing the links to point to the default libraries in the core or Arduino folder, and overriding the explicit inclusion of the local projects link.

The compile includes are also getting changed, it appears there is a logic bug and it does not interpret ${ProjectName} as being the same and is adding duplicate entries to the includes list with explicit project folder name.

**To Reproduce**

1) I have created a github project to recreate this bug. First step is to pull the repository to your workspace. I recommend doing a git clone as opposed to just a copy of the raw files so you can see the settings files getting changed after the import with a dif

https://github.com/BriscoeTech/Sloeber_ImportBugToy

2) import the project to your workspace
 * Right click project explorer -> Import
 * Select Existing Project into Workspace
 * Import the project "Sloeber_ImportBugToy"

3) the .project files and the .cproject files are now modified, and they are changing the settings.

before
[image]

after, you can see it decided to duplicate the includes since the symbol is not recognized properly
[image]

before, note we want to use the spi library version controlled in our local library folder
[image]

after, the link to the local library is overridden by a link to the default adafruit cores library
[image]

files diffed

[image]

[image]

Sloeber 
 - OS: Windows 10
 - Version v4.3

jantje, 2020-07-31:
Thanks for your extensive problem description.
Using version control is a hard topic and is for sure not a trouble free with eclipse/sloeber. I have all my projects on eclipse and can compile them on a linux and windows system. So it does work but .... not always as smooth as I want.
On top of that the "arduino way" (take the latest and greatest) is not a "version control friendly way" (take that version of that file/tool/platform/board....).
Lets go through some of your remarks
>When sloeber imports an existing sloeber project from another location for the first time, it is changing the existing links and compiler includes. 

This is by design because the location of the libraries and sketch code can change. So if Sloeber detects the code has moved or the os has changed Sloeber triggers the code behind project properties->arduino->apply close and arduino->reattach libraries

>Sloeber allowing to link to local libraries within the project is wonderfull, and is allowing me to version control my libraries with the projects using them. This bug is breaking that however when it is shared,

No it is not. But you have to make sure that when you import a sloeber project all the private libraries are available. If you havn't done so, make the private library available and delete the libraries folder in you sloeber project

>, and overriding the explicit inclusion of the local projects link.

I don't understand what you mean by this.

>The compile includes are also getting changed, it appears there is a logic bug and it does not interpret ${ProjectName} as being the same and is adding duplicate entries to the includes list with explicit project folder name.

This is a bug indeed. I think it is a regression somewhere.

>note we want to use the spi library version controlled in our local library folder

Do not try to control the stuff in the libraries or core folders. Sloeber controls these folders and assumes it can be recreated any time any way.
Create another folder (like myLibraries) and link your code there or create project dependencies. There is no special processing of the libraries folder at compile time.

As to the .project and .cproject files. Sloeber is not in control of these. It are eclipse and cdt files. The changing is out of my control (It annoys me from time to time) and they contain hardcode paths and uid. So they are not good candidates for version control.
After I checked in a initial version I ignore them so basically all instances have their own version.

Hope this helps you out

BriscoeTech, 2020-07-31:
Hello jantje,

Thank you so much for the detailed message, this helps alot.

> This is by design because the location of the libraries and sketch code can change. So if Sloeber detects the code has moved or the os has changed Sloeber triggers the code behind project properties->arduino->apply close and arduino->reattach libraries

> Do not try to control the stuff in the libraries or core folders. Sloeber controls these folders and assumes it can be recreated any time any way.
> Create another folder (like myLibraries) and link [...]

That makes sense. I went into the toy and moved the sd library to a local_library folder, and deleted the linked resource and that works great. I am happy with that :-)

> This is a bug indeed. I think it is a regression somewhere.

Ok good to know, I will keep my eyes peeked for a fix.

On the topic of version control and the arduino core, is there any advice you have to version control the core within a sloeber project? I tried copying the Adafruit core to the core folder awhile back and didn't have any success, makes sense now with how sloeber manages the core/libraries folder. Would it be possible to version control a core by putting it in its own folder (like myCore), and then just updating the linked resources to point to that folder?

jantje, 2020-08-01:
>On the topic of version control and the arduino core, is there any advice you have to version control the core within a sloeber project?

This is by default. Sloeber uses a specific version of the platform and you can have multiple versions at the same time each project on a different version of the same board.
You can even have different versions in different configurations.
If you want to cater for json files and download files no longer being available for download you'll have to version control these as well. This is what is sometimes described as "versioning the build environment" which is mostly done by making a virtual machine of the build machine.
Given your questions I would call this overkill for you ;-)

jantje, 2021-07-12:
I just made a video on how Sloeber V4.4.0 has improved version control
https://youtu.be/bqdw1L_8Gjs 
I guess this closes this issue

closed 2021-07-12 by jantje