Sloeber/arduino-eclipse-plugin #1215: Sloeber doesn't resolved definitions in include command

Jan-NiklasB, 2020-08-17:
Hey,
I'm trying to compile the HTTPClient lib on the ESP-32.
But it seems like Sloeber isn't able to resolve the defined path to a header file in a definition:

`/media/jan-niklas/Data/Programme/Installiert/Linux/Sloeber/4.3.3//arduinoPlugin/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls/mbedtls/platform.h:39:10: error: #include expects "FILENAME" or <FILENAME>
 #include MBEDTLS_CONFIG_FILE

                ^`

I don't know if it's a known problem (as I can't find the known bugs section on the side, and it also doesn't opens via the button in Sloeber), but it would be nice if this could be fixed in some way, as it compiles without problems in the example file of the Arduino-IDE.

NG
Jan-Niklas

jantje, 2020-09-26:
> it also doesn't opens via the button in Sloeber

strange. The button should open the browser to https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is:issue+label:%22status:%20known%20limitation%22

As to MBEDTLS_CONFIG_FILE was a known issue that has been fixed.

jantje, 2020-10-12:
Can this issue be closed?

closed 2020-10-23 by jantje