Sloeber/arduino-eclipse-plugin #1217: avrdude error uploading to Arduino Nano - ATMega328P - AVRISP mkII

dapollonio, 2020-08-20:
# Environment
Windows 10 pro version 1903 - build 18362.356 64bit
Sloeber product version V4.3.3_win64.2020-04-23_11-36-51
Board: Elegoo Arduino Nano - ATMega328P - AVRISP mkII

# Problem description:
The compilation of code works fine but upload doesn't work also if the output seems OK.

# Steps to reproduce:
Add AVR Boards 1.8.3 and create a simple hello world project, set platform for Arduino Nano - ATMega328P - AVRISP mkII and finally compile and upload the project.
You'll get the following output:

```
Inizio upload
Caricamento in corso per il progetto "blink" mediante "AVRDUDE"
Reset non richiesto dato che si sta utilizzando un dispositivo programmatore

In avvio: E:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CE:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -patmega328p -cstk500v2 -Pusb -Uflash:w:E:\Arduino\sloeber-workspace\blink\Release/blink.hex:i 
Output:

avrdude done.  Thank you.

Esecuzione del comando "6.3.0-arduino17/bin/avrdude" completata
```

The problem is that the done message is wrong since no code has been really uploaded.
I've so increase the verbose output by adding the following property under the section "Project properties" - "C/C++ Build" - "Build variables":

```A.TOOLS.AVRDUDE.PROGRAM.VERBOSE=-v -v -v -v```

and I found the following output:

```
Inizio upload
Caricamento in corso per il progetto "blink" mediante "AVRDUDE"
Reset non richiesto dato che si sta utilizzando un dispositivo programmatore
![workaround](https://user-images.githubusercontent.com/52105905/90755298-46b1e400-e2db-11ea-8466-dc3ca5d2c411.png)

In avvio: E:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CE:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -v -v -v -v -patmega328p -cstk500v2 -Pusb -Uflash:w:E:\Arduino\sloeber-workspace\blink\Release/blink.hex:i 
Output:

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

Esecuzione del comando "6.3.0-arduino17/bin/avrdude" completata
```

I then compared the avrdude command line with the one of the Arduino IDE and I found that:

1. The USB port is not set
2. The protocol is wrong (it should be ```arduino``` and not ```stk500v2```
3. Speed is not set

# Workaround:
I temporary fixed it with the workaround of adding the two following properties (see screenshot):

A.PROGRAM.EXTRA_PARAMS=-PCOM4 -b115200
A:PROTOCOL=arduino

May be there are better workaround but I'm not so expert to find them.

NOTE: both Arduino IDE and Eclipse C++ with plugin for Arduino work fine.

jantje, 2020-08-20:
Did you use the default upload protocol in project properties->arduino?

dapollonio, 2020-08-20:
I made the try you suggest me a few day ago on a C++ Eclipse + Sloeber plugin but the IDE crashed.
I tried now but a fresh "Sloeber product" installation and it works out of the box.

Thank you very much (I searched a lot on the internet for a solution to this problem without success)

P.S On Arduino IDE and Eclipse C++ with plugin for Arduino I choose AVRISP mkII and it works (it is what the board documentation say to use). It's normal this behaviour?

In any case the avrdude command is equal to the one of Arduino IDE using "AVRISP mkII":
```In avvio: E:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CE:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -patmega328p -carduino -PCOM4 -b115200 -D -Uflash:w:E:\Arduino\sloeber-workspace\blink\Release/blink.hex:i ```

jantje, 2020-08-20:
I don't know the answers to those questions.
But as it works with default with the product I think closing this issue is the only sensible thing to do

closed 2020-09-26 by jantje