Sloeber/arduino-eclipse-plugin #1218: Unable to compile: /bin/avr-size not found

Kashouryo, 2020-08-26:
**Describe the bug**
make can not locate the location of /bin/avr-size

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project
2. restart IDE
3. hit verify and error

**Expected behavior**
compile and upload normally 

**Screenshots**
"/bin/avr-size" is not recognized as an internal or external command, operable program or batch file
makefile:70: recipe for target 'sketch1' failed

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox (why browser?)
 - Version 81.0 b2 (sure...)

jantje, 2020-08-26:
Sounds like a typical configuration problem.
Which board do you use and how did you install it?

Kashouryo, 2020-08-27:
This is the core that I am using https://github.com/dbuezas/lgt8fx
MiniCore and Arduino OEM core works fine

jantje, 2020-09-28:
This seems to be a very general problem.
This is now fixed in master but unfortunately my build server (that pushed changes to the update site) is down

Kashouryo, 2020-09-30:
Thank you for fixing it! I am sorry to hear that your server got attacked(from your website). Maybe consider using a CI or GitHub Action to build your nightly binary?

Kashouryo, 2020-09-30:
With this new release, I am having a new issue:

Starting upload
Uploading project "test" with "AVRDUDE"
no reset because we are using a programmer

Launching:  
Output:
Cannot run program "" (in directory "C:\Users\dell\Documents\sloeber-workspace"): CreateProcess error=87, the parameter is incorrect
failed to upload "L/test/test}/Release/MegaHitsv2.hex"

jantje, 2020-09-30:
Do you build the code yourself?
If not you do not have the fix.
Anyway, for the new problem check your project properties->aduino.

Kashouryo, 2020-09-30:
yes, I just pulled git master branch, which contains the newest code.

jantje, 2020-10-12:
I think this one can be closed now. Please reopen if I think wrong ;-)

closed 2020-10-12 by jantje