Sloeber/arduino-eclipse-plugin #1220: Using Arduino Nano BLE 33

Klauzen, 2020-09-01:
Hello!

I would like to use the Arduino nano BLE 33 in sloeber. Is there any possibility to compile the sketches? The BLE 33 is using a nRF52840, so i tried to download at Preferences > Arduino > package_index.json > arduino nRF52 Boards (1.0.2) and arduino NRF52 Boards (1.0.0). But by creating a new sketch I can only choose the Arduino Primo and Primo Core....?

Could someone help me to run the Arduino Nano BLE 33 on sloeber? That would be really nice!

Thanks!

jantje, 2020-09-01:
If you installed the arduino nRF52 Boards (1.0.2) and arduino NRF52 Boards (1.0.0) platforms and you waited till they are installed (info provided by a dialog box or visible in the bottom right) you should be able to select these boards.txt in the create new sketch wizard on in project properties->arduino

Thou I'm unsure why you (as a beginner) want to install two versions of a platform; it is perfectly possible and fine

Klauzen, 2020-09-02:
Hello jantje, thank you for replying!
I have to confess that I haven't waited till the platforms/boards were installed at the first time (because it took to long). Then I tried to reinstall the platforms/boards. Maybe this could be the problem?
Now I deleted the nRF52 folder from sloeber\arduinoPlugin\packages\arduino\hardware and reinstalled it one more time. But there is only the Arduino Primo and Primo Core boards to choose from, when I create a new sketch...?
How can I install the Arduino Nano BLE 33 board? 

> Thou I'm unsure why you (as a beginner) want to install two versions of a platform; it is perfectly possible and fine
Because there are two type of platforms/boards (**n**RF52 and **N**RF52).

Thanks a lot!

jantje, 2020-09-26:
Have you figured it out yet?
If not please share a image of the Arduino board properties with the board dropdown dropped down.

Klauzen, 2020-09-27:
Hi. No, I didn't figured it out :( In the meanwhile i'm using the Arudino Pro IDE (Beta). What dropdown do you mean? I've attached the Platforms/Boards-preferences, thanks!
[image]

jantje, 2020-09-27:
I meant project properties->arduino

jantje, 2020-10-12:
Is this working yet?

closed 2020-10-23 by jantje