Sloeber/arduino-eclipse-plugin #1222: Multiple build build configurations. Problem with excluded files.

ArminJo, 2020-09-17:
First thank you very much for this wonderful project 🥇 
It saved me much time and I am a [real fan of](https://github.com/ArminJo/ServoEasing#consider-to-use-sloeber-as-ide) Sloeber.

For developing Arduino libraries, I have projects with multiple build configurations.
They differ especially in files excluded from build for the different configurations (to test different examples for the same Arduiino platform).
Compiling different configurations works well and correct 😀 

But the display in the left Project Explorer **always** shows the enabled / disabled state for the initial configuration.
And only files included in the initial configuration can be dropped to the Include Browser view, regardless of the active build configuration (which is consistent with the display in Project Explorer).
And in the editor macros are only evaluated for initial configuration 😟 
So for me some vital functions of Eclipse are missing, if I use multiple build configurations.

Is there any chance to fix this?
Can I support you?

I run it on W10 and the problem exists up to Sloeber version 4.3.3

Thanks in advance
Armin

jantje, 2020-09-17:
Thanks for the nice words

If you want to help.. please do a writeup of the configuration page here http://eclipse.baeyens.it/learn.shtml
This page correspondents with this page in github https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/website/WebContent/fragments/learn/multiple_configurations.md

I assume you make libraries as described here: http://eclipse.baeyens.it/how_to.shtml#/j
If so check weather you have CDT use the active configuration. That works on my system.
[image]

ArminJo, 2020-09-17:
Thanks a lot 👍 
The Windows/Preferences/C/C++/Indexer - Use active build configuration was not set for Sloeber.
In my standard CDT Eclipse instance, it was set. Maybe I enabled it years ago....
So you saved my days 😀 thanks!
If I have a few hours left (the probability for this just rised) I can add a few words and pictures to the empty documentation.
Best regards
Armin

closed 2020-09-17 by ArminJo

ArminJo, 2021-04-15:
> If you want to help.. please do a writeup of the configuration page here http://eclipse.baeyens.it/learn.shtml
> This page correspondents with this page in github https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/website/WebContent/fragments/learn/multiple_configurations.md

Hi, I finally succeded in creating a [tutorial for Sloeber](https://www.dropbox.com/sh/nneah2clfaedf7h/AABzt5ZGFoLzJMtQu-_BeWcga?dl=0), for my Arduino friends from the maker space.
Is word format OK for you?
Is it useful for you, and/or do you wish same chapters to be extracted to your wiki?

Thanks again for this marvelous tool!
Armin