Sloeber/arduino-eclipse-plugin #1225: sloeber.ino.cpp autogenerated file, setting ifdef, and not ifndef, causing classes sub variables not to be found.

MangyDogUK, 2020-10-06:
**Describe the bug**
Just discovered after a day of head scratching, I couldnt figure out why I was getting an error 'Could not be resolved' when trying to grab a value while in a .h file from a class made in the main file.

The project compiled fine in Arduino IDE but in sloeber it failed, and that error was shown. And the class also got the error 'not declared in scope' after a compile attempt. 

Code auto complete was able to see the class and the sub variables of the class. 

The header file was placed under the class creation code.

I've split my project into several headers to try and make it easier to navigate. 

Anyway the main error that was causing the build failure was 
'the name of my class' does not name a type; did you mean 'BitOrder'?
This was under the extern declaration of my class in the auto generated sloeber.ino.cpp file.

I really was scratching my head all day trying to solve this, and wasted a lot of time. But now just gone 12 at night, I found the issue...

I googled around about this and some one had another issue with there own header, the user mixed up ifdef and ifndef on there header file. A different issue to mine, but made me have a look. Glad I did.

I went to check the sloeber file and it indeed was ifdef.... Changing this to ifndef corrected the problem and the project now compiles correctly. 

So the error is in the autogenerated file creating the ifdef statement, when I guess should be setting ifndef.

**To Reproduce**
Steps to reproduce the behavior:
Hard to say the complexity of my project might have had a role in tripping up the  autogenerated file.... but....
Create project, create a simple class with some public variables. Create the object of that class thisclass objectofclass;

Create a .h file with the intension of putting some functions in. Place include bellow the class creation in the main file.

Try to grab the variable from the class object objectofclass.thevariable;

**Expected behavior**
I expect to be able to read the variable and manipulate the class as usual...
But I got the errors as described. 

**Screenshots**
NA

**Desktop (please complete the following information):**
Win10Pro

jantje, 2020-10-07:
I think the answer/work around and root cause of your issue is the same as https://github.com/Sloeber/arduino-eclipse-plugin/issues/775#issuecomment-311961636

MangyDogUK, 2020-10-07:
Ok yeah, Put the class declaration in its own header and that works too and links correctly... even with ifdef in the sloeber file. 

So whats breaking this then? Why is it happening?

jantje, 2020-10-07:
There is no easy way to define classes/structs without implementing them.
Therefore they are incompatible with the Sloeber ino parser which is based on
<define stuff in ino>
<include ino>

jantje, 2020-10-08:
duplicate of #775

closed 2020-10-08 by jantje