Sloeber/arduino-eclipse-plugin #1228: Way to open existing Arduino project without creating sample file

rmspacefish, 2020-10-13:
Hi,

Thanks for creating this great plugin. I have a small improvement suggestion.
Maybe I did not see the option, but is there a way to open an "exisiting" project and ommit the generation of sample files?
I had an existing project inside a folder, I created a project inside that folder. Then I deleted the sample files and everything worked just fine. I think it would be nice if there is an option to just not add any sample files. What do you think about this?

Kind Regards
Robin

jantje, 2020-10-13:
If the sketch name is the [foldername.ino] and you select default ino project no files should be created.

rmspacefish, 2020-10-13:
Ah okay, thanks for the tip.
I think an explicit option still would be nice.

jantje, 2020-10-14:
Feel free to implement one.

rmspacefish, 2020-10-14:
Got following error when trying to build the SDK on Windows 10. Do you have an idea what might cause this?

`
C:\Users\Robin\NoSyncDokumente\arduino-eclipse-plugin>mvn clean verify -PSDK,latest -DskipTests=true
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[ERROR] Internal error: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-12' -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-12'
`

Maybe I will have more luck on Ubuntu.

jantje, 2020-10-14:
It looks like it tries to build with java 12 and it doesn't find one
Note that I havn't build the sdk the last couple of months so it may have been an issue before you applied changes

jantje, 2020-10-14:
FYI the SDK is to build a development environment for sloeber
Sloeber is the development environment for arduino boards

jantje, 2020-10-14:
SDk builds fine here. Did you use maven 3.6.3?

rmspacefish, 2020-10-14:
Yes, I installed the newest version 3.6.3

jantje, 2020-10-14:
I have looked at the changes you have and I don't see any relation to sloeber sdk so I don't think the fact that the sdk build fails on your system is relevant.

jantje, 2020-11-21:
@rmspacefish are you still working on this?

rmspacefish, 2020-11-22:
Hi, I haven't had a chance to look at it again. Can you give a hint on how to "load" the changes after making them? 

Kind Regards
Robin

jantje, 2020-11-22:
>Can you give a hint on how to "load" the changes after making them?

I'm not sure what you mean here. I assume you want to debug them or get them in some production environment.
Debug->simply debug the product in your eclipse environment where you made the changes
spread out 1->do the maven build  (as described here https://github.com/Sloeber/arduino-eclipse-plugin) and put the update site on a server (This will cause conflicts conflict with the sloeber update site so this is not a sustainable solution)
spread out 2-> Do the maven build and spread the product

rmspacefish, 2020-11-22:
I just want to test whether the new feature (was quite small after all) just works. So instead of the default installed Sloeber Plugin, I would just like to use the version of the plugin with my changes. Can this be done by debugging it?

jantje, 2020-11-22:
> Can this be done by debugging it?

yes off course. That is what debugging is for. You can even step through the code to validate that things do what you intended them to do.

rmspacefish, 2020-11-22:
Alright, I will try to test it today or tomorrow.

PR #1229 (2020-11-23): explicit no file option for new sketch
I still need to test this, I have problems building the SDK on Windows 10.
fixes #1228
closed 2020-11-23 by jantje