Sloeber/arduino-eclipse-plugin #1232: #pragma once insertion breaks compilation

patricklaf, 2020-10-19:
I tried to use STM32Ethernet library with official STM32 official core.
It needs STM32Duino LwIP library but the compilation fails with message "MEM_TCP_PCB not declared".

In _lwip/memp.h_, LWIP_MEMPOOL macro is defined:
```cpp
/** Create the list of all memory pools managed by memp. MEMP_MAX represents a NULL pool at the end */
typedef enum {
#define LWIP_MEMPOOL(name,num,size,desc)  MEMP_##name,
#include "lwip/priv/memp_std.h"
  MEMP_MAX
} memp_t;
```
This code will define MEMP_TCP_PCB.
I solved it by commenting the pragma added by Sloeber in _lwip/priv/memp_std.h_:
```cpp
//Added by Sloeber 
//#pragma once
````

I don't know for sure the rules for #pragma once insertion but is it possible to exclude files from this process? I mean, if I attach this library to another project, will Sloeber insert it again?

jantje, 2020-10-19:
>I don't know for sure the rules for #pragma once insertion 

The pragma once is added to all header files at "install into sloeber time". 

>is it possible to exclude files from this process? 

No, but you can turn of the adding of pragma once in the preferences->arduino. Though I do not advice so as this is the first time I hear of a problem and I know it fixes many problems related to ino files.

>I mean, if I attach this library to another project, will Sloeber insert it again?

No. The pragma once is not added during "add library to a project time"

Background
Because of how Sloeber handles ino files (creating the sloeber.ino.cpp that includes headers used in ino files, function declarations and includes ino files) all headers used in ino files are included at least 2 times.

The standard include guards or pragma once are commonly accepted ways to avoid problems with double inclusion of header files. These solutions are so common that most advanced ide have them in their header templates.

Because arduino ide handles ino files differently arduino ide does not include headers at least 2 times and arduino IDE does not have header templates.
Therefore there are a lot of headers in arduino world that fail when included twice (as Sloeber does when uising ino files).
Therefore Sloeber adds pragma once to all header files when added to the Sloeber ecosystem.
I opted for pragma once because 
1) I think it is a better solution than include guards
2) testing showed that all arduino world compilers support pragma once
3) you can have multiple pragma once in the same file without problems
4) pragma once is compatible with include guards
5) pragma once does not clutter the define space

jantje, 2020-10-19:
Workaround: remove the lines added by sloeber

patricklaf, 2020-10-19:
Thanks @jantje.
I close the issue.

closed 2020-10-19 by patricklaf

ArminJo, 2022-07-09:
Hi Jaintje,
just for info. I also ran into problems with pragma once. It seems that the (unmaintained?) Eclipse indexer is misleaded by this directive. It happens with most of my library examples which includes the library sources as .hpp files instead of compiling them separately as *.cpp files. I then get tons of errors displayed by the indexer when looking in the *.hpp files, but compiling runs without any errors.

jantje, 2022-07-10:
The eclipse indexer has some "special behaviours" and is known to be confused when files are linked into multiple projects (like sloeber does)
FI if you have multiple projects using the same library: opening a header/source files of that library is confusing to the indexer as it always opens the same file and from that file it can not derive the project/platform as multiple projects link to the this file. Therefore the include path is not uniquely defined and sometimes the wrong include path is used with lots of indexer errors as a consequence.
In my experience this problem can be solved by closing the file and opening it from your "active project"(a concept eclipse unfortunately does not want to implement)
When building (which is the original issue) this is not a problem because then the project is known.

ArminJo, 2022-07-10:
Thanks for the explanation. My indexer errors were all gone after I removed all `pragma once` directives from all imported files only leaving the macro standard include guards.

jantje, 2022-07-10:
lets hope for you they stay gone