Sloeber/arduino-eclipse-plugin #1233: Automatic port detection as like in VSCode / PlatformIO

scottmberry, 2020-10-21:
In relation to issue #255
https://github.com/Sloeber/arduino-eclipse-plugin/issues/255

I've read the discussion on the above issue. For me, I develop on multiple host OS's with various types of Arduino even for the same code base. I often use a VM to develop and, at least with my Windows VM, the COM port assigned to the redirected USB port changes frequently. Even the same device, connected to the same host USB port can give the guest a brand new, never used COM port. This alone makes me need to go to the Properties each time I start a programming session because I can't even be sure the port has stayed the same from the last time I edited this same code, on the same guest, without even rebooting.

VSCode with the PlatformIO extension uses an "automatic" port detection. I am not sure their methods for this, but a simple "is there only one port?" detection could easily resolve my issue and I am guessing many others who have this kind of situation. On top of that, it would be doubly beneficial to have quick access to the Port, Board and Processor as "quick" settings somewhere in the main IDE window (or maybe a popup when a build/upload is initiated). Integrated with the "would you like to build?" popup seems like an easy way to add this if the main Eclipse window isn't possible. 

I love the plugin and I think this would make it even better. Thanks!

jantje, 2020-10-23:
>. Even the same device, connected to the same host USB port can give the guest a brand new, never used COM port. 

I had that problem to when I used to use vm's. As far as I can remember I fixed it in the vm software.

>This alone makes me need to go to the Properties each time I start a programming session

Not that it is different from the user perspective but to be correct is should be "at uploading time".

 >VSCode with the PlatformIO extension uses an "automatic" port detection. 

There is a possibility to detect what type of board is connected at which com port. However standard java com port support does not allow to do so (well it was in java 8 not sure about the newer versions)
Arduino IDE worked around this. I spend a lot of time trying to figure out how this worked but I never understood how. As nothing really changed since then, I do not feel like trying again. Note that as far as I understood the code is different for windows and linux.

So in other words. I like the idea, I tried to do this, I don't know how.

> "is there only one port?"

This may be feasible

>it would be doubly beneficial to have quick access to the Port, Board and Processor 

Port I can see a benefit. Board and processor, I don't think so.
That is because changing the (platform) board or processor; changes the command line and as such changes CDT input that changes the indexer that changes the ino processing, includes, defines, gui .....
This is why changing the board/processor requires a full build (in Sloeber and arduino IDE)

Having a port combo box has a lot of other implications related to 
1) eclipse having multiple project at the same time
2) detecting com port changes that can happen any time (for instance during upload)

This is basically a adventure I do not want to venture into because I fear lots of stuff that is working may break.
However feel free to make your own plugin. It should be possible based on the sloeber api and I'm willing to help out.

> popup seems like an easy way to add this if the main Eclipse window isn't possible.

This means that you are or a very experienced eclipse programmer or you do not have a clue how complex anything gui related is in eclipse. I hope it is the first, I fear it is the second.

mr-miky, 2022-02-01:
Use symlink with arduino sub serial number as filter...

jantje, 2022-02-01:
> Use symlink with arduino USB serial number as filter...

I think this is only usable on linux based oses, so not for windows

uzi18, 2022-02-02:
> Use symlink with arduino USB serial number as filter...

it is ok only for original arduino boards

But ... there is space to improve here.
@jantje  Solution should be simple as add to the list all serials (also symlinks) and store one selected by user.
https://stackoverflow.com/a/6768690

mr-miky, 2022-02-02:
I took a quick look at the source of the arduino ide and in a specific directory for windows there is listcomports.exe Maybe you can associate com and vid:pid

https://github.com/lithiumhead/ListCOMPorts

mr-miky, 2022-02-02:
@uzi18 
>  it is ok only for original arduino boards

Not necessarily, I use it mainly for esp32 and other stm32 of my construction.

Right at the end of the link you posted there is something interesting that could be inspiring

https://github.com/RishiGupta12/SerialPundit

jantje, 2022-02-02:
>@jantje Solution should be simple as add to the list all serials (also symlinks) and store one selected by user.
https://stackoverflow.com/a/6768690

I don't understand what you are trying to say and if it is that simple, feel free to contribute the solution. 😒 

>https://github.com/lithiumhead/ListCOMPorts

is a windows only tool

>https://github.com/RishiGupta12/SerialPundit

this seems interesting at first sight

As to how arduino IDE does it.
Arduino IDE uses a program written in golang to list the ports https://github.com/arduino/board-discovery . Unfortunately I do not know how to talk to golang from java.
This would be the best route as it make sloeber 100% guaranteed arduino  ide compatible (for the com port listing/detection)
It is also most robust for the future.

I however do not think the issue is about that. The issue is about "if no board is selected and only 1 is available use the one available"

mr-miky, 2022-02-02:
>Arduino IDE uses a program written in golang to list the ports https://github.com/arduino/board-discovery . Unfortunately I do not know how to talk to golang from java.

I don't know if it can help

https://stackoverflow.com/questions/49986729/how-can-i-call-a-go-function-from-java-using-the-java-native-interface#49991098