Sloeber/arduino-eclipse-plugin #1234: Debugging Arduino code in Sloeber - recipe for target 'sloeber.ino.cpp' error

ponmathanaro, 2020-10-22:
i am getting started with Arduino code debugging using Sloeber. I have followed all steps mentioned in the following video link till 11 min 25 sec - https://www.youtube.com/watch?v=ZN-rLSyhBN8&t=644s . In project properties i have selected Platform folder as "......hardware/jantje/avr_local". once clicked Apply, i have started seeing error in the project. And console log below. Attached setup screenshots and reference files.

**Console Log:**
08:03:26 **** Incremental Build of configuration localDebug for project blinkNew **** "C:\Sloeber\arduinoPlugin\tools\make\make" all 'Building file: ..\sloeber.ino.cpp' 'Starting C++ compile' "/bin/g++" -c -g -O0 -fpermissive -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR_LOCAL -DLOCAL_RUN -I"C:\Sloeber_Libraries\hardware\jantje\avr_local\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.2\variants\mega" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp" -o "sloeber.ino.cpp.o" -Dsrandom=srand -Drandom=rand The system cannot find the path specified. subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed make: *** [sloeber.ino.cpp.o] Error 1 "C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

08:03:27 Build Failed. 2 errors, 0 warnings. (took 357ms)

jantje, 2020-10-22:
I see you missed the first pinned comment under the video

closed 2020-10-23 by jantje

ponmathanaro, 2020-10-24:
Just Deleted Hardware library added through Git. Because Jantje Hardware already preloaded in Sloeber 4.3.3. it's resolved the issue.