Sloeber/arduino-eclipse-plugin #1235: Can't install the software - dead link

fneves78, 2020-10-23:
When trying to install the software on my Debian laptop and running it for the first time, there is an error :

Failed to configer Sloeber
Failed to install [DEPRECATED] Arduino mbed-enabled Boards
Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".
Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404

[image]

My internet connection has no problem.
Can you help ?

jantje, 2020-10-23:
duplicate of #1230

closed 2020-10-23 by jantje

amerging, 2020-12-14:
Is anyone resolved this issue as we are looping with duplicate - post , not solution to this ..