Sloeber/arduino-eclipse-plugin #1237: Sloeber - Arduino code Debug terminated immediately

ponmathanaro, 2020-10-24:
Hi ,
    Trying to Debug Arduino code using Sloeber with the help of https://www.youtube.com/watch?v=ZN-rLSyhBN8&t=644s . All works well till Debug Demo. When i click "Debug" in Debug Configuration, control navigating to Debug perspective and terminated immediately.  Debug console logs as follows:

GNU gdb (GDB) 7.6.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "mingw32".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
[New Thread 5148.0x2a4]
[New Thread 5148.0x39b0]
[New Thread 5148.0x19f8]
Thread ID 0 not known.
Thread ID 0 not known.
Thread ID 0 not known.

jantje, 2020-10-24:
There is not much I can say based on the info you provided

ponmathanaro, 2020-10-24:
Could you please tell me, what are the details needed?

ponmathanaro, 2020-10-24:
[Sloeber_debug issue_2.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/5433354/Sloeber_debug.issue_2.docx)

Attached complete Sloeber configuration.

jantje, 2020-10-24:
The document provides way more info to investigate :-) thanks for that
I see you got confused between installing from git and installing with the boards manager.
I guess this resulted in mixed versions (fi 32 bit versus 64 bit) which causes your issue
I would propose to undo the git install and only use jantje/hardware boards install

In the commands below setting the path is wrong (missing ;) but also unnecessary as sloeber -on windows- overwrites the path setting to a fixed path (the standard windows fresh install path).
Using the sloeber_path_extension is fine but is not necessary when using the boards manager and may cause conflict (read untested)
my advice : start sloeber directly without the sloeber.cmd
[image]

Below you can see you use the boardsmanager installed platform version 1.0.2 and not the github installed platform
[image]

Here remove the c:\unitTest\git\hardware\jantje as you are not using it and it may lead to confusion
[image]

Here I would remove versions 1.0.0 and 1.0.1 as you use 1.0.2
[image]

please try again with these changes

ponmathanaro, 2020-10-25:
@jantje -
Did below two steps. Now Debugger working fine.  Thank You very much for your Support and wonderful debugging feature for Arduino.
1. Configured gdb in debugger 
2. set MinGW bin path in Environment variable

**Set Eclipse for using gdb:**
https://www.thomasvitale.com/how-to-setup-gdb-and-eclipse-to-debug-c-files-on-macos-sierra/#:~:text=Set%20Eclipse%20for%20using%20gdb,-If%20you%20want&text=Select%20a%20launch%20configuration%20from,the%20full%20path%20of%20your%20.   ---> Go to section "Set Eclipse for using gdb"

**Configure MinGW bin path in Environment variable:**
https://stackoverflow.com/questions/52746760/netbeans-debugger-exited-with-code-0xc0000135

closed 2020-10-25 by ponmathanaro