Sloeber/arduino-eclipse-plugin #1238: Problem with product installation

mclenaghan, 2020-10-25:
Windows 10 64 bit, fresh installation from product downloaded today.
I'm getting an error:

Failed to configer Sloeber
Failed to install [DEPRECATED] Arduino mbed-enabled Boards
Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".
Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404

I haven't found this mentioned in the notes - although I could have missed it!

jantje, 2020-10-25:
duplicate of #1235

closed 2020-10-25 by jantje

mclenaghan, 2020-10-25:
My apologies. There was a filter to only show open issues – so I missed the other one.

 

Ian

 

Closed #1238 <https://github.com/Sloeber/arduino-eclipse-plugin/issues/1238> .