Sloeber/arduino-eclipse-plugin #1240: Failed to configure Sloeber error 404 on ArduinoCore-mbed-nonexistant.tar.bz2

MotoBarsteward, 2020-10-28:
**Describe the bug**
While trying to do an install of arduino-eclipse-plugin from the Eclipse market place, on restarting the Eclipse GUI, I get the error message: -
Failed to configer Sloeber
Failed to install [DEPRECATED] Arduino mbed-enabled Boards
Failed to download "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2".
Failed to download url http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2 error code is: 404

If I enter the url into my browser, I get a web page with the words "download.arduino.cc" and no file is downloaded 

**To Reproduce**
This was a fresh install of Eclipse 2020-09 C/C++ and a fresh install of the plugin

**Expected behaviour**
I was expecting it to work!

**Screenshots**
Not really required

**Desktop (please complete the following information):**
 Windows 10 2004 build

**Additional context**
Nothing

rapola, 2020-10-28:
Hi,
the same problem as well. Further configuration of the ide is corrupted, you can't hit the checkboxes for downloading any board-files...
[image]

MotoBarsteward, 2020-10-28:
This is a duplicate of #1196. Read speedfixer's comment towards the bottom. His workaround worked for me. Clearer instructions are "Click Arduino in menu bar and select preferences. In the Preferences window, under Arduino, click Platforms and Boards. Drill down through arduino, Then Arduino AVR boards and then select the version" "Apply and Close". Wait for the activity in the bottom right-hand corner to die down. Close and re-open Eclipse and your problem has gone away.

closed 2020-10-28 by MotoBarsteward

jantje, 2020-10-28:
@MotoBarsteward 
Correct. Only the last action: "Close and re-open Eclipse" should not be nessesary