Sloeber/arduino-eclipse-plugin #1241: Linux Eclipse 2020-09: Compilers not found

rmerriam, 2020-10-28:
Ubuntu Mate 20.04
Eclipse 2020-09
Plugin via Marketplace (4.3.3)

I have two projects with different processors. They build  and upload with the Arduino IDE but do not with Eclipse. Tools are under the Arduino directory. The Eclipse tool paths are incorrect.

The Arduduino IDE:
```
/mnt/devel/arduino/hardware/tools/avr/bin/avr-g++
```

In Eclipse the compilers, and `avrdude`  are not found. The checkbox "Select the toolchain like the ide" is checked. 
```
Starting C++ compile
/bin/sh: 1: /bin/avr-gcc-ar: not found
make: *** [makefile:73: /mnt/devel/ws/test/Release/arduino.ar] Error 127
```
The compiler and `avrdude` paths are incorrect in the makefile because environment variables are not set. 

`A.RUNTIME.TOOLS.AVR-GCC.PATH` is missing.
`A.TOOLS.AVRDUDE.RUNTIME.TOOLS.AVRDUDE.PATH` is missing.

Workaround is to set those environment variables using the C++ preferences: `Window : Preferences :  C/C++ : Build : Environment`

jantje, 2020-10-28:
You probably used a private hardware board and not a Sloeber boards manager installed board

rmerriam, 2020-10-29:
Not sure what you mean by a `private` board. They are Sparkfun Redboard  Uno clone and ESP32 Thing. 

BTW I used to write for Hackaday and have used your plugin for quite awhile. Returning to some embedded work now.

I stripped Eclipse and Arduino IDE off my system which meant digging into some niches. Reinstalled both. The IDE worked fine for the Uno. It worked okay on the new Eclipse install but when I first selected boards in the main preferences got this error:
```
Failed to install Arduino avr Boards (local debug)
No valid system found for "MinGW".
```
Using your package for the Uno I can't upload a sketch. Is there a version of your board definitions that isn't setup for `debug/test`? 

**Debug generates:**
```
io.sloeber.arduino.core
Error
Wed Oct 28 23:27:43 CDT 2020
A.TOOLS.AVRDUDE.UPLOAD.PATTERN : not found in the platform.txt file
```
**Test generates:**
```
"g++" -w -Os  -Wl,--gc-sections  -o "/mnt/devel/ws/j_blink/Release/j_blink.exe"    ./sloeber.ino.cpp.o    /mnt/devel/ws/j_blink/Release/arduino.ar  "-L/mnt/devel/ws/j_blink/Release" -lm -lpthread
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x24): undefined reference to `main'
```

jantje, 2020-10-29:
It seems as if you are trying to use jantjes hardware for local debugging but you do not want to do local debugging.

can you share what you see in
project properties->arduino
[image]

windows->preferences->arduino->boards and platforms
[image]

rmerriam, 2020-10-30:
Are your packages only to allow the debugging and unittest capabilities? If so then I'm fine since I don't need to use them.

jantje, 2020-10-30:
> 
> 
> Are your packages only to allow the debugging and unittest capabilities? If so then I'm fine since I don't need to use them.

yes

jantje, 2020-11-08:
usage error as the wrong package was being used.

closed 2020-11-08 by jantje