Sloeber/arduino-eclipse-plugin #1242: Adding Libraries for Other Hardware

rmerriam, 2020-10-29:
Linux Eclipse 2020-09 

I added the ESP32 package from https://dl.espressif.com/dl/package_esp32_index.json. This is for Sparkfun ESP32 Thing. 

Can compile and upload `blink`. 

None of the examples or libraries from this package are available. They are in the plugin directory. How do I get them into the plugin?

jantje, 2020-10-29:
So I added the json
[image]

Then installed platform  version 1.0.4
[image]

waited for it to install 

I tried creating a new project based on the BasicOTA example
[image]
[image]

And indeed the example is not there
[image]

But it is available in the add libraries
[image]

and he does pick the correct one
[image]

So as a workaround you can create the default ino and copy and past the example code over the default ino content.
This worked fine for me with basicOTA
[image]

jantje, 2020-10-29:
I tried again now finding the library and all is well here on my system

rmerriam, 2020-10-30:
Okay, got it to work. Wasn't aware that examples would show up under the libraries. Thanks.

Also found you can check using `Project | Properties | Resource | Linked Resource | Linked Resource `. That lists all the libraries and their directories.

closed 2020-10-30 by rmerriam

jantje, 2020-10-30:
>On mine it did load the BasicOTA.ino file into the project. But it also loaded the other .ino files from other projects. I'll test that more for another issue.

When you create a sketch based on examples you can select multiple examples. Sloeber remembers your previous selections and add the new ones. At the bottom you can see how many have been selected.
[image]

The reason you can select multiple is for if you want a project using spi, wire, servo's .... you can have all the code in one go. Though it won't compile.