Sloeber/arduino-eclipse-plugin #1247: Sloeber Plugin cannot download files from downloads.Arduino.cc in Eclipse

grijspetje, 2020-11-03:
I installed the plugin but it cannot finish the installation due to an error.

See errorlog.

What am I doing wrong?

[Sloeber.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/5483088/Sloeber.log)

closed 2020-11-07 by jantje

w-wallner, 2020-11-07:
I experience this error on Kubuntu 20.10 with Eclipse 2020-09.
My Eclipse log file is attached.
I also see unrelated Eclipse log entries on my system, e.g. from PyDev, so I have attached a second copy of the log file showing only the Sloeber-related entries (file with the _short suffix).

[Sloeber_Kubuntu_20.10_Eclipse_2020-09_short.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5505363/Sloeber_Kubuntu_20.10_Eclipse_2020-09_short.txt)
[Sloeber_Kubuntu_20.10_Eclipse_2020-09.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/5505364/Sloeber_Kubuntu_20.10_Eclipse_2020-09.txt)

jantje, 2020-11-08:
duplicate of #1245