Sloeber/arduino-eclipse-plugin #1248: Cannot select Arduino boards in preferences (4.3.3)

grijspetje, 2020-11-06:
Hi all,

I searched for a solution but couldn't found any. I followed the installation procedure on the site and tried the suggestion as mentioned in issue 1196:

"Read speedfixer's comment towards the bottom. His workaround worked for me. Clearer instructions are "Click Arduino in menu bar and select preferences. In the Preferences window, under Arduino, click Platforms and Boards. Drill down through arduino, Then Arduino AVR boards and then select the version" "Apply and Close". Wait for the activity in the bottom right-hand corner to die down. Close and re-open Eclipse and your problem has gone away."

My preference pane shows all the boards but I cannot select them. So the initial problem I started with (cannot down load boards from dowloads.arduino.cc ...) still remains. Please help me out!

Thanks,

Kind regards,

Peter

jantje, 2020-11-06:
please sharesome screenshots
the boardsmanager installed boards for instance and the new project wizard board selection

grijspetje, 2020-11-07:
Here are three screen shots:

The first I see is the "Problem occurred after startup" and the corresponding error log screenshot "Error log after startup" 

And then the screenshot of the window in which I cannot make any selections: "No selections Arduino Preferences"

Thanks for the help

grijspetje, 2020-11-07:
[image]
[image]
[image]

jantje, 2020-11-07:
you did not do the workaround yet.
You have to click on one of the trangles to see the available versions and then select one of the versions
[image]

jantje, 2020-11-08:
duplicate of #1245

closed 2020-11-08 by jantje