Sloeber/arduino-eclipse-plugin #1249: Question: Where Do External Libraries Go?

rmerriam, 2020-11-10:
I have an existing project. The project has a number of source and header files but they aren't configured as Arduino libraries. Tried putting them in a private library folder but the plug-in doesn't find them. I can put them in with the main source file and it works. 

Is that the way to do it? Or is there a standard way for the plugin to find these non-Arduino files?

jantje, 2020-11-10:
The private libraries is one way to go.
You can have multiple private libraries so they can be spread over your system.
You can point to the library itself or to the parent folder. Sloeber is searching for .h files to identify the libraries

I for instance have one pointing to my git repository https://github.com/jantje/ArduinoLibraries which allows me to use all the libraries in the repository using 1 single private library.

Another way is to link/copy the folders by drag and drop from the system explorer into your project. Do not use the Sloeber created library or core folder to drop as these are "maintained" by sloeber.

rmerriam, 2020-11-11:
I'm going to using Sloeber a lot with different boards and libraries so need to understand how all this works to make things go smoothly. 

Should I see the private library path

rmerriam, 2020-11-11:
Don't know how that previous comment got posted. Starting over...

I'm going to using Sloeber a lot with different boards and libraries so need to understand how all this works to make things go smoothly.

Still totally baffled by setting up libraries. Here's what I have which should work, I believe. 

Trying to work with the NEO GPS library. First screen shot shows it selected as a library.

[image]

This screen shot shows it exists in the Eclipse plugin subdirectory. Added by using the 3rd party URL.

[image]

When I 'verify' this is the error.

[image]

--------
Still really unclear how to add private libraries to a project.

jantje, 2020-11-12:
I do not really understand what you want to do so lets start from the beginning.
Arduino grouped boards in a platform. 
These platforms are described in a json file and third parties can provide platforms described in json files. You can use the boards manager to install platforms (I know it is confusing)
Similarly arduino created a json file to describe libraries. You install these libraries with the library manager.
Arduino IDE also supports a location on disk where it looks for (what I call  private libraries)
Last time I checked Arduino IDE supports multiple platform json files but only one library json file and only one private library location.
Sloeber supports multiple platform json files and multiple library json files and multiple private library locations.

Focussing on libraries
1) Each platform comes with libraries I call these hardware libraries
2) There are private libraries that are not installed by the library manager
3) There are libraries managed by the library manager
The above order is the order Sloeber uses to search for libraries. So if a library exists with the same name in hardware libraries, private libraries and libraries the hardware library will be used.

From the first image I can see the library manager installed NeoGPS.
This library manager managed NeoGPS library was added to the project (implied by your second image). But not correctly (I assume not by Sloeber) because the command line shows that the NeoGPS library folder is not added to the include path.
Here is an example of one of my projects and you can see the libraries added to the include path
[image]

Libraries and version management.
Arduino IDE only allows for one version of a platform. Sloeber allows any number of versions of the platform. As such Sloeber allows different versions of hardware libraries.
Both Sloeber and Arduino IDE only allow for 1 version of a library managed library.
So if you have 2 projects in the same eclipse install using the same library but needing a different version ... you will have to manage the the library of one project yourself.

Managing libraries yourself
You can manage your libraries all by yourself. After all you are in CDT world when you are in Sloeber and Sloeber translates Arduino world in CDT world.
So it can be done but .... there are hurdles to take. The only reason I think it might be interesting is "forcing to use old libraries case described above"

rmerriam, 2020-11-12:
The NEO was added by Sloeber. I just deselected the library, deleted the directory in the 2nd image, and selected the library. It was installed in the same location. 

I created an example from the library and it compiles. Created another empty project with an INO file. Added the three headers needed by the project I'm trying to get working:
```
#include <EEPROM.h>
#include <SoftwareSerial.h>
#include "NMEAGPS.h" //NeoGps by SlashDevin"
```
The the other two were added but not the NMEAGPS.h. 

Just created another example project. It works. Move my main file into that project and removed the existing main. 

Created a project with INO main. Cut and paste my source into file. The library isn't added. 

Basically creating a project not from an example doesn't add the library. Creating an example does. What could cause that?

jantje, 2020-11-12:
>The NEO was added by Sloeber

Sloeber will install libraries based on library name if you activate it (I think it is on by default, I have it turned off)
[image]

>Created a project with INO main. Cut and paste my source into file. The library isn't added.

When creating a new project based on a example Sloeber knows the library the example is coming from and will add that library at project creation.
When you create a new project and the "Automatically import libraries based on include" is on and you add the code Sloeber **guesses** the library that is needed. This is done after the code has been saved and after the indexer indexed the code.
I say **guess** because Sloeber is more strict in finding libraries than the Arduino IDE. If the Library name matches the include directive (-.h) Sloeber should find it. In your case these do not match.
You can manually add the library using menu item  "arduino" and  "add library to the selected project"

rmerriam, 2020-11-12:
That explains what is happening. Thank you.

closed 2020-11-12 by rmerriam